espressif/esp-idf #16018: install new component should have a search function

emaayan, 2025-05-23:
### Is your feature request related to a problem?

currently when wanting to install a componenet directly from esp plugin you get just a dump of all the components 

### Describe the solution you'd like.

when selecting install new component , the screen should include a search bar that will narrow down by typing the string.

brianignacio5, 2025-05-27:
Please try the `ESP-IDF: Show ESP Component Registry` command.

emaayan, 2025-06-01:
> Please try the `ESP-IDF: Show ESP Component Registry` command.

is it in the plugin?

emaayan, 2025-06-30:
> For the Espressif IDE, please review [Install ESP-IDF Components](https://docs.espressif.com/projects/espressif-ide/en/latest/additionalfeatures/install-esp-components.html)

i have reviewed it, but it can't search for component by name (or by anything)

alirana01, 2025-06-30:
@emaayan the component feature in IDE is outdated we are working on creating a new and more robust UI for that. For now you can try to visit https://components.espressif.com/ and search components there. You can use the esp-idf console inside the ide to instsall the components as mentioned in the readme for each component.