espressif/esp-idf #16019: [esp_http_client]: Ability to check if client needs polled

bryghtlabs-richard, 2025-05-23:
### Is your feature request related to a problem?

I have several http clients handling streams, uploads, and downloads in parallel all managed by one thread that handles requests, responses, and polling of the clients. It's tricky to tell if the client should be polled by perform or fetch_headers in the idle loop.

### Describe the solution you'd like.

I'd like a function like `esp_http_client_want_polled()` that makes it easy to tell if an HTTP client needs to be polled, or if it is idle (before request or after response).

This is particularly helpful when using async clients or clients with timeouts, and calling `esp_http_client_fetch_headers()` and `esp_http_client_perform()` in an idle loop.

`esp_http_client_want_polled()` is useful in the idle loop to tell if a client should be polled or not. This is needed since we don't want to call `esp_http_client_perform()` when the client is idle - that will start a new request.

### Describe alternatives you've considered.

I've started tracking esp_err_t for each esp_http_client handle I have, but it's messy and error-prone since each step `esp_http_client_open()`,`esp_http_client_write()`,`esp_http_client_fetch_headers()` has different result codes.

bryghtlabs-richard, 2025-05-28:
An example of why this is needed:

Consider [this API](https://lichess.org/api#tag/Bot/operation/apiStreamEvent):

- When authentication succeeds, it returns a chunked ND-JSON stream. In this case, I should call `esp_http_client_fetch_headers()` to get the next chunk/line of JSON.
- When authentication fails, it returns an HTTP401 response with a plain JSON body explaining the failure. In this case, the response is a single plain response, so `esp_http_client_fetch_headers()` won't fetch the body - already fetched the headers, no more work is needed. In this case, `esp_http_client_perform()` should be called to retrieve the body and trigger the remaining event handlers.

But we must not call `esp_http_client_perform()` if the client doesn't need polled - that will re-trigger the currently configured request(minus any POST data).

This `esp_http_client_want_polled()` simplifies the above case, like this:

```
void idle_func(void){
    if(esp_http_client_want_polled(http_client)){
        esp_http_client_perform(http_client);
    }
}
```

In the case of a working stream, we can poll to fetch another line.
In the case of a failed response, we can read out the response body.

bryghtlabs-richard, 2025-06-03:
Hi @nileshkale123,

> I’d appreciate it if you could help clarify the exact expectations for the esp_http_client_want_polled() API. Specifically, what should be the intended behavior of this API?

I would like esp_http_client_want_polled() to evaluate a client and tell me when there is an outstanding request(whose complete response is not yet received). We could do true/false, or esp_err_t/ESP_ERR_HTTP_EAGAIN (if we want more flexibility for returning other errors).

> It would be helpful if you could provide a sample or dummy algorithm for the API to better understand the intended behavior.

Sure! I think it's just a map of `esp_http_client_state_t` to `esp_err_t`. With this internal client state:

```
typedef enum {
    HTTP_STATE_UNINIT = 0,
    HTTP_STATE_INIT,
    HTTP_STATE_CONNECTED,
    HTTP_STATE_REQ_COMPLETE_HEADER,
    HTTP_STATE_REQ_COMPLETE_DATA,
    HTTP_STATE_RES_COMPLETE_HEADER,
    HTTP_STATE_RES_ON_DATA_START,
    HTTP_STATE_RES_COMPLETE_DATA,
    HTTP_STATE_CLOSE
} esp_http_state_t;
```

 I expect it'll be something like this(not tested, or even compiled):

```
esp_err_t esp_http_client_want_polled(esp_http_client_handle_t client)
{
    if (client == NULL) {
        ESP_LOGD(TAG, "Client handle is NULL");
        return ESP_ERR_INVALID_ARG;
    }
    switch (client->state){
        case HTTP_STATE_UNINIT:
            return ESP_ERR_INVALID_STATE;
        case HTTP_STATE_INIT:
        case HTTP_STATE_CONNECTED:
        case HTTP_STATE_REQ_COMPLETE_HEADER:
        case HTTP_STATE_CLOSE:
            //ESP_OK means client is ok, and does not need polled
            return ESP_OK;
        case HTTP_STATE_REQ_COMPLETE_DATA://Need to perform() to fetch headers
        case HTTP_STATE_RES_COMPLETE_HEADER://Need to perform() to fetch data
        case HTTP_STATE_RES_ON_DATA_START://Need to perform() to fetch data
            return ESP_ERR_HTTP_EAGAIN;
        case HTTP_STATE_RES_COMPLETE_DATA:
            //If you call perform() here, it would repeat the request. So  the currently the client does not need polled.
            return ESP_OK;
    }
```

An alternative approach would be moving `esp_http_state_t` to the header and adding a getter function to fetch it for a client - then I could examine the client state directly.

nileshkale123, 2025-06-04:
Please find the patch and let me know any changes required in this. 

[GH16019_patch.txt](https://github.com/user-attachments/files/20586371/GH16019_patch.txt)

bryghtlabs-richard, 2025-06-04:
Thanks - it'll take me a bit to test this.

bryghtlabs-richard, 2025-06-05:
It seems things are not so simple. I've updated the code-block above with my findings. I thought this could be a simple evaluation of the HTTP parser state, but a few states aren't enough to understand if we should request polling or not.

nileshkale123, 2025-06-24:
Hello @bryghtlabs-richard ,

Would it meet your requirements if we made the client state public and provided an API to retrieve it? This way, the user can handle further logic on the application side as needed.

nileshkale123, 2025-07-15:
Hello @bryghtlabs-richard ,

We have raised a combined MR for this issue along with [16070](https://github.com/espressif/esp-idf/issues/16070).
Key changes we have made:
1. Made client state public and added API to get client current state.
2. Added state HTTP_STATE_CONNECTING as per your suggestion and updated the flow for API esp_http_client_perform().
3. Updated the API esp_http_client_close() to close the connection if `state >= HTTP_STATE_CONNECTING`.

I am attaching the patch. Please let me know if you have any suggestions.

[patch_16019.txt](https://github.com/user-attachments/files/21233088/patch_16019.txt)

NOTE: To determine whether the chunk is complete or not, or if we received complete data, you can refer API `esp_http_client_is_complete_data_received()` to poll in case the client is working in async mode.

nileshkale123, 2025-07-18:
Hello @bryghtlabs-richard,

I wanted to share some results from my recent experiments regarding chunked transfer encoding.

Test Setup
- Server: Custom chunk-encoded server (see server.py attached below).
- Client: Modified async HTTP client (esp_http_client_example.c) with updates in the http_download_chunk() function.

Observations
- A single call to esp_http_client_perform() successfully downloaded the entire file.
- I added logging for client->is_chunk_complete. Its value remained 0 during all intermediate recursive calls, and turned 1 only on the last chunk.
- This suggests the flag is set upon receiving the final chunk — which follows the chunked encoding format 0\r\n\r\n.

Findings
- The server correctly terminates the response with a single final chunk (0\r\n\r\n).
- As per HTTP/1.1 standards, a chunked response must contain only one final chunk; multiple “final” chunks in the same response are not valid.

Conclusion
- Given the behavior, esp_http_client_perform() continues reading until the final chunk is received.
- Therefore, the following loop should suffice to handle chunked transfers:
 `while (1) {
    err = esp_http_client_perform(client);
    if (err != ESP_ERR_HTTP_EAGAIN) {
        break;
    }
}`

[server.txt](https://github.com/user-attachments/files/21310711/server.txt)
[client.txt](https://github.com/user-attachments/files/21310710/client.txt)

**Question**
- In light of the above, could you clarify why we need to explicitly use the HTTP_STATE_RES_COMPLETE_DATA state in the state machine?
- Is there a specific edge case or architectural reason for depending on that state, even though esp_http_client_perform() already loops until the full response is received?
- Is  your posted another query [16205](https://github.com/espressif/esp-idf/issues/16205) points to the same question?

bryghtlabs-richard, 2025-07-21:
Hi @nileshkale123 , I wanted to double-check your results.

1. You create an HTTP(not HTTPS) server, but enable `is_async` on the client. This [is an unsupported configuration](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_http_client.html#_CPPv4N24esp_http_client_config_t8is_asyncE). I'm not sure what it's supposed to do, but it's not behaving as expected.
2. "A single call to esp_http_client_perform() successfully downloaded the entire file." - With `is_async=true`, `esp_http_client_perform()` should return temporary codes like EAGAIN/EWOULDBLOCK/EINPROGRESS, requiring multiple calls to successfully download the entire file.

> In light of the above, could you clarify why we need to explicitly use the HTTP_STATE_RES_COMPLETE_DATA state in the state machine?
> Is your posted another query https://github.com/espressif/esp-idf/issues/16205 points to the same question?

Yes. I mostly mean that this solution works for exposing the ability to check if client needs polled to the caller, but we'll still need to fix `HTTP_STATE_RES_COMPLETE_DATA`. This state is necessary for detecting if the client is idle. Knowing the client is idle is helpful when we have a request-queue, and I want to submit another request, but only once the previous response has been fully received(state == HTTP_STATE_RES_COMPLETE_DATA).

bryghtlabs-richard, 2025-07-22:
> We hope this enhancement addresses your requirement effectively.

It is necessary, but I'm afraid it's not sufficient. For each state, the caller needs to be able to detect if a request has been started and not finished. By overriding HTTP_STATE_RES_COMPLETE_DATA to HTTP_STATE_CONNECTED, that information is lost. Instead use them like this:

* `HTTP_STATE_CONNECTED` - the underlying connection to support a request has been made, and the pending request has not yet been fully sent.
* `HTTP_STATE_RES_COMPLETE_DATA` - the response is complete, and another has not been started. In this state, it's safe to prepare a new request, or calling `esp_http_client_perform()` in this state should repeat the previous request.

nileshkale123, 2025-07-23:
Hello @bryghtlabs-richard ,

If we change the state to HTTP_STATE_RES_COMPLETE_DATA at the end of a successful response retrieval, it would disrupt the expected flow for keep-alive support. Specifically, the client would reset to the initial state and attempt to establish a new connection for each subsequent request, rather than reusing the existing one.

Since esp_http_client_perform() returns ESP_OK only after the response has been fully and successfully retrieved, this return value can be reliably used to confirm completion before starting the next request.

Additionally, we have raised an internal fix to ensure that any issues encountered during response retrieval are properly captured and the appropriate event is dispatched. With this fix in place, ESP_OK will be returned only in the case of successful response completion.