espressif/esp-idf #16026: ESP32-S3 与电脑连接到同一个热点，电脑PING通一次ESP32-S3后无法再次PING通，过段时间恢复

FanhuaCloud, 2025-05-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

ESP32-S3 与电脑连接到同一个热点，电脑PING通一次ESP32-S3后无法再次PING通，过段时间恢复

### What is the actual behavior?

正常应该可以PING通且可以正常通信

### Steps to reproduce.

1. 运行例程: examples\wifi\getting_started\station
2. 手机创建热点
3. ESP32-S3与电脑连接到热点
4. 电脑PING ESP32-S3芯片
5. 可以PING通一次，后面就一直不能通上，过几分钟后恢复正常

### Debug Logs.

```plain

```

zhangyanjiaoesp, 2025-06-05:
@FanhuaCloud 
能提供一下 wifi 抓包吗？
请提供一下用来做热点的手机型号和版本，本地也会尝试复现看看。

FanhuaCloud, 2025-06-05:
手机是mate60 pro，4.2系统，抓包等会我传一下
[image]

FanhuaCloud, 2025-06-05:
[image]

貌似传不了pcap文件，之前是因为跑UDP server，发现ESP端只能send成功一次，后面运行send函数就无法发出去了，上图是跑udp_server例程的抓包，只能回显一次后面就收不到回显了
刚刚重新试了一下，发不出去的时候ping应该还是通上的

zhangyanjiaoesp, 2025-06-09:
@FanhuaCloud 
pcap 文件尝试压缩一下或者把后缀修改一下，应该能上传成功。所以现在是能ping 通但是发包有问题？

> 之前是因为跑UDP server，发现ESP端只能send成功一次，后面运行send函数就无法发出去了

ESP32-S3 做 UDPserver ？send 函数发送失败，返回了什么错误？

Alvin1Zhang, 2025-06-30:
@FanhuaCloud 您好，麻烦问下可以提供更多更新么？谢谢。

closed 2025-08-04 by FanhuaCloud