espressif/esp-idf #16027: How do I eliminate the error report of I2C? （我该如何消除I2C的错误）

culionbear, 2025-05-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.3

### Espressif SoC revision.

ESP32S3-BOX-3（目前安装了Micropython，之前的信息查不到了）

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-BOX-3

### Power Supply used.

USB

### What is the expected behavior?

I write a code about Microphone to speaker。
我写了一个简单的麦克风声音传递给喇叭的简易代码

### What is the actual behavior?

record error “I2C transaction unexpected nack detected”
烧录到板子后会提示“I2C transaction unexpected nack detected”

### Steps to reproduce.

```cpp
/*
* 麦克风输入到扬声器输出示例
* 适用于ESP-S3-BOX-3开发板
* 功能：采集16kHz、单声道、16位PCM音频数据并实时播放
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/i2s.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "esp_peripherals.h"
#include "periph_sdcard.h"
#include "periph_button.h"
#include "periph_touch.h"
#include "board.h"
#include "es8311.h"
#include "i2s_stream.h"
#include "algorithm_stream.h"
static const char *TAG = "MIC_SPEAKER_EXAMPLE";
[... 638 lines cut ...]
I (4288) MIC_SPEAKER_EXAMPLE: 注册音频元素到管道完成
I (4295) AUDIO_PIPELINE: link el->rb, el:0x3c0512c4, tag:mic, rb:0x3c051a94
I (4303) AUDIO_THREAD: The mic task allocate stack on internal memory
I (4310) AUDIO_ELEMENT: [mic-0x3c0512c4] Element task created
I (4316) AUDIO_THREAD: The speaker task allocate stack on internal memory
I (4323) AUDIO_ELEMENT: [speaker-0x3c05175c] Element task created
I (4330) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:17074580 Bytes, Inter:351739 Bytes, Dram:351739 Bytes, Dram largest free:258048Bytes
I (4345) AUDIO_ELEMENT: [mic] AEL_MSG_CMD_RESUME,state:1
I (4351) AUDIO_ELEMENT: [speaker] AEL_MSG_CMD_RESUME,state:1
I (4357) AUDIO_PIPELINE: Pipeline started
I (4362) MIC_SPEAKER_EXAMPLE: 事件信息：
I (4367) MIC_SPEAKER_EXAMPLE: source_type: 131072
I (4372) MIC_SPEAKER_EXAMPLE: source: 0x3c0512c4
I (4378) MIC_SPEAKER_EXAMPLE: cmd: 8
I (4382) MIC_SPEAKER_EXAMPLE: data: 0xc
I (4387) MIC_SPEAKER_EXAMPLE: data_len: 4
I (4392) MIC_SPEAKER_EXAMPLE: 音频元素 mic 正在运行
I (4398) MIC_SPEAKER_EXAMPLE: 事件信息：
I (4403) MIC_SPEAKER_EXAMPLE: source_type: 131072
I (4409) MIC_SPEAKER_EXAMPLE: source: 0x3c05175c
I (4414) MIC_SPEAKER_EXAMPLE: cmd: 8
I (4419) MIC_SPEAKER_EXAMPLE: data: 0xc
I (4423) MIC_SPEAKER_EXAMPLE: data_len: 4
I (4428) MIC_SPEAKER_EXAMPLE: 音频元素 speaker 正在运行
```

mythbuster5, 2025-05-26:
It already told you that `I2C transaction unexpected nack detected`. Check hardware connection or check address is correct or not please

culionbear, 2025-05-27:
> 它已经告诉你了`I2C transaction unexpected nack detected`。请检查硬件连接或检查地址是否正确

connection is true，because i used box-3 demo and not error

culionbear, 2025-05-27:
> `(int)msg.data == AEL_STATUS_STATE_RUNNING` is comparison valid? (pointer and AEL_STATUS_STATE_RUNNING)

i think this is true(or not?)，because i2c error is not in here.
我认为这个应该是正确的，因为i2c error在init的时候就产生了

mythbuster5, 2025-05-27:
If your issue is related to box-3, I suggest you ask question here https://github.com/espressif/esp-iot-solution

mythbuster5, 2025-06-03:
Any update on this issue? If no, this issue will be concluded soon until new clue has been found.