espressif/esp-idf #16028: Running examples based on NimBle on ESP32-c2 cannot be detected by mobile phones

HonestQiao, 2025-05-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.1

### Espressif SoC revision.

ESP32-C2 (revision v1.2)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C2 | Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_get_started/nimble/NimBLE_Beacon

After compiling this example and burning it to the board, it was able to run successfully.
I would like to use nRF Connect for Mobile on my phone to search for this device.

### What is the actual behavior?

After compiling this example and burning it to the board, it was able to run successfully.
But on my phone, I couldn't find the device with nRF Connect for Mobile.

### Steps to reproduce.

1. cp -a ~/esp/esp-idf/examples/bluetooth/ble_get_started/nimble/NimBLE_Beacon ./
2. cd NimBLE_Beacon
3. idf.py set-target esp32c2
4. idf.py menuconfig build flash monitor

### Debug Logs.

```plain
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x15a4
load:0x403acb70,len:0xbb0
load:0x403aeb70,len:0x29fc
entry 0x403acb7a
I (21) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (21) boot: compile time May 26 2025 18:54:58
I (21) boot: chip revision: v1.2
I (21) boot: efuse block revision: v0.1
I (25) boot.esp32c2: MMU Page Size  : 32K
I (28) boot.esp32c2: SPI Speed      : 60MHz
I (32) boot.esp32c2: SPI Mode       : DIO
I (36) boot.esp32c2: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 00100000
I (73) boot: End of partition table
I (76) esp_image: segment 0: paddr=00010020 vaddr=3c048020 size=11c5ch ( 72796) map
I (100) esp_image: segment 1: paddr=00021c84 vaddr=3fcb1000 size=00178h (   376) load
I (100) esp_image: segment 2: paddr=00021e04 vaddr=3fcb23f8 size=02074h (  8308) load
I (106) esp_image: segment 3: paddr=00023e80 vaddr=40380000 size=04198h ( 16792) load
I (116) esp_image: segment 4: paddr=00028020 vaddr=42000020 size=4704ch (290892) map
I (185) esp_image: segment 5: paddr=0006f074 vaddr=40384198 size=073d8h ( 29656) load
I (193) esp_image: segment 6: paddr=00076454 vaddr=4038b570 size=05a90h ( 23184) load
I (203) boot: Loaded app from partition at offset 0x10000
I (204) boot: Disabling RNG early entropy source...
I (214) cpu_start: Unicore app
I (222) cpu_start: Pro cpu start user code
I (222) cpu_start: cpu freq: 120000000 Hz
I (222) app_init: Application information:
I (222) app_init: Project name:     nimble_beacon
I (226) app_init: App version:      1
I (230) app_init: Compile time:     May 26 2025 18:54:24
I (235) app_init: ELF file SHA256:  444cc5e0c...
I (239) app_init: ESP-IDF:          v5.4.1
I (243) efuse_init: Min chip rev:     v1.0
I (247) efuse_init: Max chip rev:     v2.99
I (251) efuse_init: Chip rev:         v1.2
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (261) heap_init: At 3FCB5440 len 00027730 (157 KiB): RAM
I (266) heap_init: At 3FCDCB70 len 0000294C (10 KiB): RAM
I (272) spi_flash: detected chip: gd
I (275) spi_flash: flash io: dio
W (277) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (290) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (296) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (302) coexist: coex firmware version: e727207
I (307) coexist: coexist rom version de8c800
I (311) main_task: Started on CPU0
I (311) main_task: Calling app_main()
I (331) BLE_INIT: Using main XTAL as clock source
I (331) BLE_INIT: ble controller commit:[5922fa3]
I (331) BLE_INIT: ble rom commit:[3314f9d]
I (331) BLE_INIT: Bluetooth MAC: a0:b7:65:80:bc:c6
I (331) phy_init: phy_version 360,c034f7c,Feb  6 2025,15:34:12
I (371) phy: libbtbb version: d3face4, Feb  6 2025, 15:34:25
I (371) NimBLE_Beacon: nimble host task has been started!
I (381) NimBLE: GAP procedure initiated: stop advertising.

I (381) NimBLE_Beacon: device address: C6:BC:80:65:B7:A0
I (381) NimBLE: GAP procedure initiated: advertise;
I (391) NimBLE: disc_mode=2
I (391) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (401) NimBLE:

I (401) NimBLE_Beacon: advertising started!
I (401) main_task: Returned from app_main()
```

### More Information.

**can be detected**
If I use an esp32-c3 board or an esp32-c6 board.
```
idf.py set-target esp32c3 | esp32c6
idf.py menuconfig build flash monitor
```
The output log is basically the same as above, with no significant difference.
I can use nRF Connect for Mobile to search for this device, which name is **NimBLE_Beacon**.

**can't be detected**
I have tried the example named NimBLE_GATT_Server, and the result is the same: the device cannot be detected in esp32-c2.

**can be detected**
I have also tried the examples named Bluedroid_Beacon or Bluedroid_GATT_Server, the device can be detected in esp32-c2.

I have used two phones from different manufacturers to install nRF Connect for Mobile for testing, and the results are the same.

What happened or was there anything I didn't notice?

Alvin1Zhang, 2025-05-27:
Thanks for reporting.

strange969, 2025-05-27:
Hi @HonestQiao ,

I ran the Nimble_Beacon example on the ESP32-C2 chip, and I was able to see the device in the nRF Connect app. I also checked with a different mobile phone, and the device was visible there as well. I have attached a screenshot for your reference.

Could you please try running the bleprph example on the ESP32-C2 chip from your side and confirm whether the device is being detected?

Thank you!

[image]

[image]

HonestQiao, 2025-05-30:
**esp32c2:** 

[sdkconfig.esp32c2.txt](https://github.com/user-attachments/files/20514780/sdkconfig.esp32c2.txt)

[image]

nRF Connec: not in list

**esp32c3:**

[sdkconfig.esp32c3.txt](https://github.com/user-attachments/files/20514863/sdkconfig.esp32c3.txt)

[image]

nRF Connec: "nimble-bleprph" in list

[image]

HonestQiao, 2025-05-30:
Patch apply:
[image]

enable BT_HCI_LOG_DEBUG_EN:

[image]

log:
[run.log.txt](https://github.com/user-attachments/files/20524945/run.log.txt)

HonestQiao, 2025-05-30:
log for esp32c3:
[esp32c3_run.log.txt](https://github.com/user-attachments/files/20525242/esp32c3_run.log.txt)

nRF Connec: 
"nimble-bleprph" in list

ZhouXiao616, 2025-06-04:
Hi @HonestQiao 
I cannot reproduce the issue either. Could I ask for the mobile phone model that you are using?

ZhouXiao616, 2025-06-09:
@HonestQiao Thanks for the info, we will try to figure it out.

ZhouXiao616, 2025-09-01:
Hi @HonestQiao , does the issue still exist?

closed 2025-11-19 by Alvin1Zhang