espressif/esp-idf #16029: Error: app partition is too small for binary

GuiguixMtp, 2025-05-26:
Hello!

I want to advertise the esp32C5 devkit in BLE, based on the BLE and WIFI examples framework. IDF version 5.4. I've started to program the devkit advertisement but when compiling, I get an error message, Error: app partition is too small for binary :

`FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size C:/Espressif/frameworks/esp-idf-v5.4/examples/test_guillaume/build/esp-idf/esptool_py/CMakeFiles/app_check_size 
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.4\examples\test_guillaume\build\esp-idf\esptool_py && C:\Users\gdri\.espressif\python_env\idf5.5_py3.11_env\Scripts\python.exe C:/Users/gdri/esp/master/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Espressif/frameworks/esp-idf-v5.4/examples/test_guillaume/build/partition_table/partition-table.bin C:/Espressif/frameworks/esp-idf-v5.4/examples/test_guillaume/build/test_guillaume.bin"
Error: app partition is too small for binary test_guillaume.bin size 0x170650:
  - Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0x70650)
ninja: build stopped: subcommand failed.`

In the “partitions tables” settings I don't know if I can manually increase it to 2MB or if I need to manage the problem differently.

Thank you in advance for your answers.

GuiguixMtp, 2025-05-27:
I have 4MB in my esp32C5, I think is good

nopnop2002, 2025-05-27:
Yes.
You can create a 2M factory partition.

nopnop2002, 2025-05-27:
Use Custom partition table CSV.

adokitkat, 2025-07-08:
I am closing this issue. Feel free to reopen if needed or create a new issue.