espressif/esp-idf #16031: Thread Commissioning Support for Thread-Enabled Espressif Devices

kunj2308, 2025-05-27:
### Is your feature request related to a problem?

Currently, there is no official support or streamlined method in the ESP-IDF SDK for commissioning Thread-enabled Espressif devices using standard Thread commissioning protocols such as QR code-based onboarding, Bluetooth LE-based commissioning, or mobile app integrations (e.g., Google Home, Apple Home, or OpenThread Commissioner apps). This makes it difficult for developers and integrators to securely and easily onboard Thread devices to an existing Thread mesh network.

### Describe the solution you'd like.

We would like to see support for Thread commissioning capabilities within ESP-IDF for Espressif chips with Thread support (e.g., ESP32-H2), including:

Integration with Thread Commissioner role to onboard new devices.

Support for BLE-based commissioning, including integration with standard mobile apps (e.g., Google Home App, Apple Home App, or a dedicated OpenThread Commissioner mobile app).

Support for QR code onboarding (like Matter QR codes or similar).

Sample applications showing how to commission a Thread device into an existing mesh network using a mobile device.

CLI tools or APIs to trigger the commissioning process via UART or other interfaces for testing and prototyping.

### Describe alternatives you've considered.

Manual commissioning via CLI or hardcoded network credentials – this is not practical or secure for production deployments.

Using third-party Thread Commissioner implementations – these are often complex, not optimized for Espressif hardware, or lacking documentation.

Developing a custom mobile app – time-consuming and non-standard for many developers.

### Additional context.

Adding native support for Thread commissioning would significantly improve the development experience and help align Espressif's Thread offerings with current industry practices and ecosystems. This is especially important with growing adoption of Matter and other protocols relying on Thread networks.

We are actively working on solutions with Espressif hardware, and having this feature in ESP-IDF would be a major enabler for production-ready Thread-based solutions.

kunj2308, 2025-05-29:
Hi @chshu,

Thank you for the detailed response and the helpful links.

I’ve reviewed the [CLI Commissioning Guide](https://github.com/openthread/openthread/blob/main/src/cli/README_COMMISSIONING.md) as referenced. I understand that this method uses the Command Line Interface (CLI) on the Thread client device for commissioning, but I’m specifically looking for a solution where the commissioning process can be done wirelessly and without direct CLI access to the Thread client device.

My goal is to enable seamless commissioning—ideally via BLE and QR code scanning—without requiring manual CLI intervention on the device being onboarded. This would be critical for end-users and production-level deployments where physical access to the device or CLI interfaces is impractical.

Could you please clarify:

Is there any ongoing or planned support in ESP-IDF or associated SDKs (e.g., esp-thread, esp-matter, esp-rainmaker) for enabling wireless commissioning of Thread-only devices, without CLI interaction on the client side?

Is there a way to integrate OpenThread Commissioner functionality directly into an ESP32-H2 (or other Espressif device) to handle the entire commissioning process over BLE or Wi-Fi from another device acting as a commissioner?

Are there any reference examples or ongoing work for achieving a mobile-app-style or headless wireless commissioning workflow for Thread?

Looking forward to your insights on this.

Best regards,
@kunj2308

chshu, 2025-06-18:
@kunj2308 @adrien3d 

Both Matter and Rainmaker support commissioning without requiring CLI access. Could you please evaluate the links provided:  https://github.com/espressif/esp-idf/issues/16031#issuecomment-2914650415.

The quickest way to try it out:
- Run the esp-matter [light example](https://github.com/espressif/esp-matter/tree/main/examples/light) on H2 or C6
- Use Apple Home App (iOS 18+) to commission the device by scanning this QR code:

[image]