espressif/esp-idf #16032: How to control Wifi buffers in internal memory (CON-1651)

jonsmirl, 2025-05-27:
I have every option in wifi I can see set to use SPIRAM including dynamic buffers. But WIFI is still using 44,264 of internal RAM.  Anything I can do about this?

[sdkconfig.defaults.txt](https://github.com/user-attachments/files/19851725/sdkconfig.defaults.txt)

[image]

zhangyanjiaoesp, 2025-06-09:
@jonsmirl 
Do you set the `ESP_WIFI_TX_BUFFER` to static?

[image]

jonsmirl, 2025-06-09:
[sdkconfig.txt](https://github.com/user-attachments/files/20655227/sdkconfig.txt)

I have them set dynamic. I attached my sdkconfig.

zhangyanjiaoesp, 2025-06-11:
@jonsmirl How much memory do you want to reduce? You could try decreasing `CONFIG_ESP_WIFI_STATIC_TX_BUFFER_NUM` a bit.

closed 2025-09-19 by Sherry616