espressif/esp-idf #16039: PCNT - set counter value

DangisSl, 2025-05-28:
### Is your feature request related to a problem?

In earlier versions (ESP-IDF v4...) it was possible to set a PCNT unit to a specific value by setting a register value directly. One only needed to know the unit number to do that.

Since ESP-IDF v5... PCNT unit numbers are managed automatically and not directly accessible. Thus there is no longer a way to set a counter unit to a specific value.

### Describe the solution you'd like.

Would it be possible to have a simple function, which takes as arguments a unit handle and a int16_t value, and sets that counter unit's count to the given 16 bit value?

### Describe alternatives you've considered.

I've considered guessing the unit numbers. This would work if numbers are assigned sequentially to new units and I have complete control of the units being created and deleted. This is a fragile, unreliable and not scalable approach.

Reverting back to older versions of ESP-IDF, but there are many advantages to V5 and I don't see this as a long-term sustainable solution.

Not setting the counter values (i.e. starting from 0 which is the default behavior). Handling the "offset" everywhere else in the code, e.g. adding a value to the count every time it is retrieved. This would work. But results in messy implementation. It would be far more elegant and practical to be able to set counter values to any value one needs.

Kainarx, 2025-05-29:
Hi, which register did you set? According to the  [TMR](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#pcnt#Register%20Summary), the count register of pcnt is read-only and we have no way to set it manually

DangisSl, 2025-06-01:
Good point. So it turns out there never was a way to set counter value explicitly. Does that mean it is not possible to implement a function to set counter value explicitly? One could always feed artificial pulses to the counter until it gets to the desired value, but if the required number is large, this could take an unreasonable amount of time.

suda-morris, 2025-06-04:
We would consider implementing this hardware feature in the future chip, i.e. allow loading a predefined counter value asynchronously.

DangisSl, 2025-06-04:
Kainarx - software counter is always an option. In the end I opted for pulsing a GPIO to set the counter value. Not elegant, but after some evaluation the delay will be tolerable for my use case. Thank you for your insights.

suda-morris - thank you, that is good news. I don't know if it is just me, but I think this functionality would be pretty useful to have. For example when you want to apply an offset to a "home" position to compensate for a known systemic error (e.g. manufacturing or assembly error of a motor drive system with encoder).