espressif/esp-idf #16046: On monitor output comes currupted.( flash encrypted + secure boot v2 enabled )

Vaibhav0216, 2025-05-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have enabled secure boot v2 and falsh encryption on esp32 wroom. I got output on the monitor like:

```
Leaving...
Staying in bootloader.
RY=C:/Users/Admin/Desktop/hello_world/build -P C:/Users/Admin/.espressif/frameworks/esp-idf-v5.3/components/esptool_py/run_serial_tool.cmake"
Executing action: monitor
Running idf_monitor in directory C:\Users\Admin\Desktop\hello_world
Executing "C:\Users\Admin\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\Users\Admin\.espressif\frameworks\esp-idf-v5.3\tools/idf_monitor.py -p COM10 -b 460800 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 301 C:\Users\Admin\Desktop\hello_world\build\hello_world.elf --encrypted --force-color -m 'C:\Users\Admin\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe' 'C:\Users\Admin\.espressif\frameworks\esp-idf-v5.3\tools\idf.py' '-p' 'COM10' '-b' '460800'"...
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM10 instead...
--- esp-idf-monitor 1.5.0 on \\.\COM10 460800
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
x<�x�x��x��x�x<☼x⌂�x����x�������x⌂�x���x⌂����������x⌂���������x��xx��x��x��x�x����x<��x�x<��x�x<☼x���x<��x��x�x<��x<☼x��x<☼x�x<☼x▲�x��x��x��x�x����x<�����x���x<x<☼x�x�x<�x<x��x<x<☼x<�x⌂�x<x��x<�x�x<

```
I am confused is it successful or not?

My espefuse summary:

```
Security fuses:
UART_DOWNLOAD_DIS (BLOCK0)                         Disable UART download mode. Valid for ESP32 V3 and = False R/W (0b0)
                                                    newer; only
ABS_DONE_0 (BLOCK0)                                Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0)                                Secure boot V2 is enabled for bootloader image     = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0)                        Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0)                        Disable flash decryption in UART bootloader        = True R/W (0b1)
KEY_STATUS (BLOCK0)                                Usage of efuse block 3 (reserved)                  = False R/W (0b0)
SECURE_VERSION (BLOCK3)                            Secure version for anti-rollback                   = 0 R/W (0x00000000)
BLOCK1 (BLOCK1)                                    Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2)                                    Security boot key
   = cd 1d a0 a7 8f fc c5 89 dd 9f ac 5d e3 b9 a6 36 00 19 8b 55 26 5b 3a 17 dc 3b c2 9e 60 10 9a c9 R/-
BLOCK3 (BLOCK3)                                    Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
Flash fuses:
FLASH_CRYPT_CNT (BLOCK0)                           Flash encryption is enabled if this field has an o = 1 R/W (0b0000001)
                                                   dd number of bits set
FLASH_CRYPT_CONFIG (BLOCK0)                        Flash encryption config (key tweak bits)           = 15 R/W (0xf)
```
I am following commad to flash new code:
**flash_secure.ps1**

```
# Set your COM port and baud rate
$Port = "COM10"
$Baud = 460800

# Set file paths
$Bootloader = "build\bootloader\bootloader.bin"
$PartitionTable = "build\partition_table\partition-table.bin"
$AppBinary = "build\mqtt_ssl.bin"
$AppSigned = "app-signed.bin"
$SigningKey = "secure_boot_signing_key.pem"

# Sign the application binary
Write-Host "🔐 Signing application with Secure Boot V2 key..."
espsecure.py sign_data --key $SigningKey --version 2 --output $AppSigned $AppBinary

if ($LASTEXITCODE -ne 0) {
    Write-Host "❌ Signing failed!" -ForegroundColor Red
    exit 1
}

# Flash all components encrypted
Write-Host ""
Write-Host "Flashing bootloader, partition table, and signed app (encrypted)"

idf.py -p $Port -b $Baud encrypted-flash monitor
```

Vaibhav0216, 2025-06-09:
@AdityaHPatwardhan , It not works.
Esp idf does not provide proper documentation of flash encrypt and secure boot. There are different documentations. 
In git repo. readme file different commands. 

Can you please tell me how can I apply this functionalities?

AdityaHPatwardhan, 2025-07-22:
@Vaibhav0216 

Does this issue still affect your case?
If yes, could you please provide more details? (sdkconfig file)
Thanks