espressif/esp-idf #16054: ESP32 Ethernet ENC28j60 error with esp_netif_attach

yarncode, 2025-05-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-devkit

### Power Supply used.

USB

### What is the expected behavior?

I want module en28j60 nice work with esp32 

### What is the actual behavior?

It alway this trace error went run func esp_netif_attach

### Steps to reproduce.

I use example of lib ethernet_init here: https://components.espressif.com/components/espressif/ethernet_init/versions/0.3.3

### Debug Logs.

```plain
I (31) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (31) boot: compile time May 31 2025 22:22:08
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 003b6000
I (99) boot:  4 ota_1            OTA app          00 11 003d0000 003b6000
I (107) boot:  5 storage          Unknown data     01 83 00786000 0000a000
I (115) boot:  6 coredump         Unknown data     01 03 00790000 00020000
I (122) boot: End of partition table
I (126) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a0a0h (172192) map
I (194) esp_image: segment 1: paddr=0003a0c8 vaddr=3ff80000 size=00004h (     4) load
I (194) esp_image: segment 2: paddr=0003a0d4 vaddr=3ffb0000 size=040f8h ( 16632) load
[... 171 lines cut ...]
  ESP_ERROR_CHECK(esp_netif_attach(this->net, esp_eth_new_netif_glue(this->eth_handles[0])));

  /* phase: Register event handler for IP_EVENT_ETH_GOT_IP */
  ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &Ethernet::ip_event_handler, this));
  ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &Ethernet::eth_event_handler, this));

  /* phase: start ethernet */
  ESP_ERROR_CHECK(esp_eth_start(this->eth_handles[0]));

  eth_dev_info_t info = ethernet_init_get_dev_info(&this->eth_handles[0]);
  if (info.type == ETH_DEV_TYPE_INTERNAL_ETH)
  {
    ESP_LOGI(TAG, "Device Name: %s", info.name);
    ESP_LOGI(TAG, "Device type: ETH_DEV_TYPE_INTERNAL_ETH(%d)", info.type);
    ESP_LOGI(TAG, "Pins: mdc: %d, mdio: %d", info.pin.eth_internal_mdc, info.pin.eth_internal_mdio);
  }
  else if (info.type == ETH_DEV_TYPE_SPI)
  {
    ESP_LOGI(TAG, "Device Name: %s", info.name);
    ESP_LOGI(TAG, "Device type: ETH_DEV_TYPE_SPI(%d)", info.type);
    ESP_LOGI(TAG, "Pins: cs: %d, intr: %d", info.pin.eth_spi_cs, info.pin.eth_spi_int);
  }
}

void Ethernet::onReceive(Services s, int event, void *data) {}

kostaond, 2025-06-02:
The pointer to Ethernet handles array is initialized by `ethernet_init_all` and the address is stored to local variable `esp_eth_handle_t *eth_handles;`
```cpp
while (ethernet_init_all(&eth_handles, &eth_port_cnt) != ESP_OK)
```
However, you then use `this->eth_handles[0]` for the subsequent calls and I don't see `this->eth_handles` to be assigned anywhere.

closed 2025-06-02 by yarncode

yarncode, 2025-06-02:
> The pointer to Ethernet handles array is initialized by `ethernet_init_all` and the address is stored to local variable `esp_eth_handle_t *eth_handles;`
> 
> while (ethernet_init_all(&eth_handles, &eth_port_cnt) != ESP_OK)
> However, you then use `this->eth_handles[0]` for the subsequent calls and I don't see `this->eth_handles` to be assigned anywhere.

thanks for help me <3

kostaond, 2025-06-03:
> u right, my bad, i so stupid, this the basic error, ahhhhhhhhhhh @.@

No problem, it happens... I struggled with a stupid mistake last week too 😄 Just one suggestion, if you have any problems with hardware drivers in the future, please consider trying official examples to reproduce the issue at first. It's a great help for everybody:
* it may help you to narrow down the issue to your code
* component maintainers know them very well and it's much easier for us to find potential problem in the driver itself