espressif/esp-idf #16055: 在Vscode下使用ESP-IDF出现“拒绝访问”/"Access denied" when using ESP-IDF under Vscode

Drzad1024, 2025-06-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

当我编译时，经常运行到一半出现“拒绝访问”，然后等待几秒再次点击编译，就可以继续进行下去。

  
 *  正在执行任务: C:\Users\Administrator\.espressif\tools\ninja\1.12.1\ninja.EXE  

[25/702] Linking C static library esp-idf\console\libconsole.a
FAILED: esp-idf/console/libconsole.a 
C:\Windows\system32\cmd.exe /C "cd . && C:\Users\Administrator\.espressif\tools\cmake\3.30.2\bin\cmake.exe -E rm -f esp-idf\console\libconsole.a && C:\Users\Administrator\.espressif\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32s3-elf-ar.exe qc esp-idf\console\libconsole.a  esp-idf/console/CMakeFiles/__idf_console.dir/commands.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/esp_console_common.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/split_argv.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/linenoise/linenoise.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/esp_console_repl_chip.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_cmd.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_date.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_dbl.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_dstr.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_end.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_file.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_hashtable.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_int.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_lit.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_rem.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_rex.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_str.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_utils.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/argtable3.c.obj && C:\Users\Administrator\.espressif\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32s3-elf-ranlib.exe esp-idf\console\libconsole.a && cd ."
拒绝访问。
[42/702] Linking C static library esp-idf\protobuf-c\libprotobuf-c.a
ninja: build stopped: subcommand failed.

 *  终端进程“C:\Users\Administrator\.espressif\tools\ninja\1.12.1\ninja.EXE”已终止，退出代码: 1。 

 *  正在执行任务: C:\Users\Administrator\.espressif\tools\ninja\1.12.1\ninja.EXE  

[648/661] Linking C static library esp-idf\wifi_provisioning\libwifi_provisioning.a

Drzad1024, 2025-06-02:
我感觉是电脑的问题，我用的是Windows 10 专业工作站版，版本号为22H2，我重启电脑后关闭所有其他软件，只运行vscode依然存在该问题。我更换使用另外一台Windows11笔记本电脑就正常。

larryli, 2025-06-04:
多半是杀毒软件
`idf-env antivirus exclusion add --all`
https://github.com/espressif/idf-env?tab=readme-ov-file#working-with-antivirus