espressif/esp-idf #16069: Esp-Now and BLE

DartDragunov, 2025-06-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### Espressif SoC revision.

ESP32-D0WD (revision v1.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect to receive packages from ESP-NOW advertising approximately every 300ms, but changing the delay doesn’t seem to have any effect.

### What is the actual behavior?

Instead, I receive a random number of packets — sometimes 3, other times 70 — with no clear pattern. It randomly stops for unpredictable amounts of time: sometimes for 3 seconds with absolutely no packets, other times for up to 20 seconds. Even when it doesn't completely stop, the packet intervals are inconsistent — for example, I might receive 3 packets in a row, then 5 more with 3-second gaps. It's not a fixed pattern; it's purely random.

As for the error data in my debug logs, that’s simply because I’m not following the default ESP-NOW example packet format. When I do receive a packet, it contains the correct data that I’m sending. I was initially using my own custom application and encountered this problem. To troubleshoot, I tested it using the official ESP-NOW example. However, after adding BLE, the same issue occurred. Even after stripping BLE down to just starting a service — without any characteristics — the problem persists and the behavior remains the same.

### Steps to reproduce.

Just building the ESP-NOW example and adding esp_ble_gatts_app_register — even with only a service created and no characteristics — is enough to trigger this behavior.

### Debug Logs.

```plain
I (379249) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (379249) espnow_example: Receive error data from: cc:7b:5c:c7:4c:4c
I (379549) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (379849) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (379849) espnow_example: Receive error data from: cc:7b:5c:c7:4c:4c
I (380149) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (380449) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (380449) espnow_example: Receive error data from: cc:7b:5c:c7:4c:4c
I (380749) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (380749) espnow_example: Receive error data from: cc:7b:5c:c7:4c:4c
I (381049) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (381349) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (381649) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (381949) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (382249) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (382549) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (382849) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (383149) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (383149) espnow_example: Receive error data from: cc:7b:5c:c7:4c:4c
I (383449) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (383749) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (384049) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (384349) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (384649) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (384949) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (385249) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (385549) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (385849) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (386149) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (386449) espnow_example: send data to ff:ff:ff:ff:ff:ff
```

### More Information.

I also tested with other custom boards, as well as some TTGOs I had lying around.

zhangyanjiaoesp, 2025-06-09:
@DartDragunov 

> Just building the ESP-NOW example and adding esp_ble_gatts_app_register — even with only a service created and no characteristics — is enough to trigger this behavior.

Can you provide your test code? As far as I know, simply calling the `esp_ble_gatts_app_register()` API should not affect Wi-Fi packet transmission or reception.

DartDragunov, 2025-06-12:
SDKCONFIG for ESP_NOW_CONTROLLER
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.4.1 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_RNG_SUPPORTED=y
CONFIG_SOC_LIGHT_SLEEP_SUPPORTED=y
CONFIG_SOC_DEEP_SLEEP_SUPPORTED=y
CONFIG_SOC_LP_PERIPH_SHARE_INTERRUPT=y
CONFIG_SOC_PM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_DIGI_MONITOR_NUM=0
[... 4697 lines cut ...]
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

DartDragunov, 2025-06-12:
Just run this for 2–3 minutes and you should see the behavior I described. Sometimes the controller receives messages instantly, but then suddenly stops. Other times, it starts by receiving nothing at all, then suddenly receives a burst of messages and begins failing consistently.

DartDragunov, 2025-06-12:
For example, in the log below, the node and controller was broadcasting constantly:
```
I (95539) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (95639) espnow_example: send data to ff:ff:ff:ff:ff:ff
...
I (96939) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (97039) espnow_example: Receive 970th broadcast data from: fc:b4:67:88:38:5c, len: 10
I (97139) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (97139) espnow_example: Receive 971th broadcast data from: fc:b4:67:88:38:5c, len: 10
I (97239) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (97339) espnow_example: send data to ff:ff:ff:ff:ff:ff
...
```
Despite broadcasting continuously, only packets 970 and 971 were received, and then it stopped receiving again.

DartDragunov, 2025-06-12:
@zhangyanjiaoesp I've added a #define INTRODUCE_BUG 1 to toggle the bug-related part of the code. Sorry for the trouble — I know this isn't easy to reproduce consistently. I suspect it's a coexistence issue between Bluetooth and Wi-Fi.

According to the documentation, ESP-NOW should work fine alongside BLE GATTS. However, in my case, it doesn’t seem to behave properly once a BLE service is started. What’s odd is that you don’t even need to connect a client — just advertising seems enough to break ESP-NOW reception.

I haven’t tested this further because I can’t give up on the GATT server — it's essential to my project. So unfortunately, I had no choice but to switch to BLE Mesh to keep moving forward. Still, I really hope this can be fixed someday, since ESP-NOW would be a much better fit for my use case due to the lower latency between messages.

zhangyanjiaoesp, 2025-06-13:
@DartDragunov 
thanks for providing the code and sdkconfig, we will test it and reply you ASAP.

DartDragunov, 2025-06-14:
Thank you so much for the support. I thought it was something related to coexistence, but I couldn’t figure out why. That cleared up a lot of my doubts. Thanks again — I’ll go ahead and close the issue now.

closed 2025-06-14 by DartDragunov