espressif/esp-idf #16072: USB Host: Ability to free individual endpoints without closing the device

SneezingCactus, 2025-06-05:
### Is your feature request related to a problem?

I'm working on an application which inquired

SneezingCactus, 2025-06-05:
(Sorry, I misclicked and created the issue before writing it out. I will complete the issue in a bit.)

SneezingCactus, 2025-06-05:
I've also tried hardcoding the number of available channels at line 149 of usb_dwc_hal.c to see if the number being set to 7 was simply some sort of config error during manufacturing, with no avail. The second device would connect "successfully" but `hid_class_request_set_report` would be completely ignored so the controller was never initialized (I'm assuming this is because its assigned OUT endpoint doesn't really exist).

SneezingCactus, 2025-06-05:
Just tested w/ the hardcoding, and yep! It works perfectly. Seems like my ESP32-S3 does have 8 endpoints after all.

That said, before closing the issue, I must ask: is there a cleaner way of overriding chan_num_total (or any sort of read only hardware config for that matter) without having to fork ESP-IDF? Maybe by redirecting `usb_dwc_ll_ghwcfg_get_channel_num` to my own function somehow?

roma-jam, 2025-06-05:
Hi @SneezingCactus, 

yes, this is the incorrect channel calculation.

It was fixed in the master already: https://github.com/espressif/esp-idf/commit/d53b0a91113d01d2ce8c99f8283fec70f706ac4d

And will be available in releases v5.4.2 and v5.3.4. 

So, either try to use the latest master or wait for the next release. 

I'll close the issue as it was already solved. 
Thanks for reporting the problem!