espressif/esp-idf #16081: Weird SPI timing issue when using IOMUX instead of GPIO Matrix

acha666, 2025-06-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-3644-g28ac0243bb

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

When driving a MAX31865 over SPI on the ESP32, communication works only if I explicitly set `SPICOMMON_BUSFLAG_GPIO_PINS` .
With my custom pin mapping the driver defaults to IOMUX routing; this introduces timing errors - the bus appears to run in Mode 0 (CPHA = 0) instead of the requested Mode 1, or the clock edges are shifted.

My board lacks test-point pads, so I cannot capture waveforms yet. I can try to reproduce the problem on an dev board if needed.

### What is the actual behavior?

With IOMUX routing enabled, the ESP32 seemingly uses Mode 0 or exhibits distorted timing. Forcing SPICOMMON_BUSFLAG_GPIO_PINS (GPIO matrix) restores proper Mode 1 operation.

### Steps to reproduce.

Minimal code: 

``` c
#include <driver/spi_master.h>
#include <esp_log.h>

void app_main(void)
{
    spi_bus_config_t spi_cfg = {
        .mosi_io_num = 7,
        .miso_io_num = 2,
        .sclk_io_num = 6,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 64,
        .flags = SPICOMMON_BUSFLAG_IOMUX_PINS,
        // .flags = SPICOMMON_BUSFLAG_GPIO_PINS,
    };

    spi_bus_initialize(SPI2_HOST, &spi_cfg, SPI_DMA_DISABLED);

    spi_device_handle_t spi_dev;
    spi_device_interface_config_t dev_cfg = {
        .mode = 1,
        .clock_speed_hz = 1000,
        .spics_io_num = 1,
        .queue_size = 1,
    };
    spi_bus_add_device(SPI2_HOST, &dev_cfg, &spi_dev);

    static uint8_t rxb[20] = {}, txb[20] = {};

    spi_transaction_t t = {
        .length = 2 * 8,
        .rxlength = 2 * 8,
        .tx_buffer = txb,
        .rx_buffer = rxb,
    };
    txb[0] = 0x80;
    txb[1] = 0x00;
    spi_device_polling_transmit(spi_dev, &t);

    txb[0] = txb[1] = 0x00;
    spi_device_polling_transmit(spi_dev, &t);
    ESP_LOGI("Test", "%02X %02X", rxb[0], rxb[1]);

    txb[0] = 0x80;
    txb[1] = 0x90;
    spi_device_polling_transmit(spi_dev, &t);

    txb[0] = txb[1] = 0x00;
    spi_device_polling_transmit(spi_dev, &t);
    ESP_LOGI("Test", "%02X %02X", rxb[0], rxb[1]);
}
```

Schematic: [link](https://github.com/acha666/Fan_Controller/releases/download/ci-build-0004/Schematic-f60cdc6.pdf)

Full hardware design: [link](https://github.com/acha666/Fan_Controller)https://github.com/[acha666/Fan_Controller](https://github.com/acha666/Fan_Controller)

### Debug Logs.

```plain

```

### Diagnostic report archive.

<details>

<summary>IOMUX Log</summary>

I (683) main_task: Calling app_main()
D (693) spi: SPI2 use iomux pins.                                                                                          ***Using IOMUX***
V (693) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (693) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (703) intr_alloc: Connected src 72 to int 11 (cpu 0)
V (713) bus_lock: device registered on bus 1 slot 0.
D (713) spi_hal: eff: 78, limit: 80000k(/0), 0 dummy, -1 delay
D (723) spi_master: SPI2: New device added to CS0, effective clock: 78125 Hz
V bus_lock: dev 0 acquired.
V (733) spi_master: polling trans
V bus_lock: SPI dev changed from -1 to 0
V (733) spi_master: polling trans done
V (743) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (743) spi_master: polling trans
V (753) spi_master: polling trans done
V (753) bus_lock: dev 0 released.
I (753) Test: 00 90                                                             ***0x90 is register value since the last run, should be 0x00***
V bus_lock: dev 0 acquired.
V (763) spi_master: polling trans
V (763) spi_master: polling trans done
V (763) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (773) spi_master: polling trans
V (773) spi_master: polling trans done
V (773) bus_lock: dev 0 released.
I (783) Test: 00 90
I (783) main_task: Returned from app_main()

</details>

<details>

<summary>GPIO Matrix Log</summary>

I (683) main_task: Calling app_main()
D (693) spi: SPI2 use gpio matrix.                                                                              ***NOT Using IOMUX***
V (693) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (693) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (703) intr_alloc: Connected src 72 to int 11 (cpu 0)
V (713) bus_lock: device registered on bus 1 slot 0.
D (713) spi_hal: eff: 78, limit: 80000k(/0), 0 dummy, -1 delay
D (723) spi_master: SPI2: New device added to CS0, effective clock: 78125 Hz
V bus_lock: dev 0 acquired.
V (733) spi_master: polling trans
V bus_lock: SPI dev changed from -1 to 0
V (733) spi_master: polling trans done
V (743) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (743) spi_master: polling trans
V (753) spi_master: polling trans done
V (753) bus_lock: dev 0 released.
I (753) Test: 00 00
V bus_lock: dev 0 acquired.
V (763) spi_master: polling trans
V (763) spi_master: polling trans done
V (763) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (773) spi_master: polling trans
V (773) spi_master: polling trans done
V (773) bus_lock: dev 0 released.
I (783) Test: 00 90

</details>

suda-morris, 2025-06-10:
according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux The IOMUX uses the GPIO16 as the CS signal.

acha666, 2025-06-10:
> according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux The IOMUX uses the GPIO16 as the CS signal.

Could you explain a little more about how this affect SPI timing?

Thanks

wanckl, 2025-06-12:
@acha666  Not really understand in your log `Test: 00 90 ----------------------- 0x90 is register value since the last run, should be 0x00`

And checked that the output is right `mode 1`, So your problem should be RX direction. Maybe in gpio_matrix, the data have some delay compare to the IO_MUX. So need a wave form to see dose the slave `MAX31865` output data earlier then the standard edge.

But for C6, using either gpio_matrix or iomux don't have performance different under 40M, you can still conitnue using gpio_matrix for your aplication.

closed 2025-06-30 by Alvin1Zhang