espressif/esp-idf #16084: How to Enable Flash Encryption After Secure Boot V2 is Enabled?

Vaibhav0216, 2025-06-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all,

I’ve successfully enabled Secure Boot V2 on my ESP32 device using ESP-IDF, and everything is working fine. Now, I want to enable Flash Encryption as the next step in securing my firmware.

I’m referring to the official ESP-IDF documentation:
[https://docs.espressif.com/projects/esp ... externally](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/security-features-enablement-workflows.html#enable-flash-encryption-externally)

According to this, the first step to enable flash encryption externally is:
```esptool.py --port PORT erase_flash```

I am following the documentation, but if there is a way to enable Flash Encryption using menuconfig after Secure Boot V2 has already been enabled, I’m open to that approach as well.

This command erases the entire flash, including the bootloader region (< 0x8000). However, from my understanding, once Secure Boot V2 is enabled and applied, the bootloader cannot be rewritten (as Secure Boot expects a verified bootloader with a valid signature). This raises some concerns:

1. If I erase the flash, including the bootloader, how can I safely re-flash it again, given that Secure Boot V2 is already enforced?

2. Is there a safe workflow for enabling Flash Encryption after Secure Boot V2 is enabled, without violating the secure boot policy or bricking the device?

3. What are the specific challenges or caveats I should be aware of when enabling Flash Encryption after Secure Boot V2?

My goal is to enable both Secure Boot V2 and Flash Encryption, but I’m doing them step-by-step. Any guidance or an updated recommended workflow for this scenario would be appreciated.

Thanks!

Vaibhav0216, 2025-06-11:
Yes. I understood this after brick my 2 devices.

Vaibhav0216, 2025-06-11:
Hi @AdityaHPatwardhan ,
We tried Qemu but we face problem .  Suppose we are trying on the real hardware then for flash encrypted or signed binary we use command like below:
`esptool.py --chip esp32 -b 460800 --before default_reset --after no_reset write_flash 0x1000 bootloader-signed.bin 0xd000 build/partition_table/partition-table.bin 0x20000 secure-boot-signed.bin`

But in qemu esptool.py not working how can we achieve it ?

AdityaHPatwardhan, 2025-07-08:
Hi @Vaibhav0216,
For secure boot and qemu related instructions, you can follow [security_features_app](https://github.com/espressif/esp-idf/tree/master/examples/security/security_features_app#enable-security-features-with-help-of-qemu)