espressif/esp-idf #16091: ESP32-S3 crashes with “Failed to reset co-processor” when running Thread BR with ESP32-H2 RCP

kunjsn123, 2025-06-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-849-g28980b62c9

### Espressif SoC revision.

Ubuntu 22.04 LTS

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 DevKitC  ESP32-H2 DevKitC (used for RCP)

### Power Supply used.

USB

### What is the expected behavior?

ESP32-S3 (running thread_br from esp-rainmaker) should initialize and communicate with ESP32-H2 (running ot_rcp) via UART, acting as a Thread Border Router.

### What is the actual behavior?

The ESP32-S3 reboots repeatedly after printing:

W(5155) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout  
E(5155) OPENTHREAD:[C] P-SpinelDrive-: Failed to reset co-processor!

Then a fatal assert is triggered and the device resets. This prevents the Thread BR from starting correctly.

### Steps to reproduce.

Clone esp-rainmaker and esp-idf (v5.4.1)

Build and flash ot_rcp example to ESP32-H2 (examples/openthread/ot_rcp)

Build and flash thread_br to ESP32-S3 (esp-rainmaker/examples/thread_br)

Connect ESP32-S3 and ESP32-H2 via UART (TX/RX/GND)

Power both devices

Observe ESP32-S3 crash and reset loop in terminal

### Debug Logs.

```plain
W(5155) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout  
E(5155) OPENTHREAD:[C] P-SpinelDrive-: Failed to reset co-processor!

assert failed: 0x42070fc6
Backtrace: 0x40375505:0x3fccbe30 0x40380f71:0x3fccbe50 0x4038846a:0x3fccbe70
           0x42070fc6:0x3fccbeb0 0x42070ff2:0x3fccbed0 0x420909c9:0x3fccbef0
           0x42090d19:0x3fccbf10 0x42071069:0x3fccbf30 0x42070c81:0x3fccbf80
           0x4207006a:0x3fccbfa0 0x4200f9fb:0x3fccbfc0

Reset reason: RTC_SW_CPU_RST
```

### More Information.

UART connection between ESP32-S3 and ESP32-H2 has been double-checked (photo attached in comments).

Both devices flash successfully, and UART TX/RX are connected per the documentation.

Unsure if the issue is due to UART config mismatch or firmware incompatibility.

Any guidance on menuconfig UART pin setup, baud rate, or version compatibility would be appreciated.

kunjsn123, 2025-06-11:
[https://dl.espressif.com/dl/schematics/esp_thread_br_zigbee_gw_schematiccs_v1.2.pdf](url)
i am replicating this schematic using two independent developement boards ESP32-S3 DevKitC ESP32-H2 DevKitC (used for RCP)

zwx1995esp, 2025-06-11:
Hi, please refer to this [FAQ](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#host-soc-can-t-communicate-with-rcp-properly), GPIO17 and GPIO18 on the ESP32-S3 have a different driver curren, GPIO4 and GPIO5 are recommended for use as the UART RX/TX GPIOs.

macgaiser, 2026-09-05:
I'm seeing the exact same issue with a different two-chip combination: ESP32-C6 (host, running the ot_br example) + ESP32-H2-DevKitM-1 (RCP, running the ot_rcp example), both on ESP-IDF v5.4.2.

Wait for response timeout followed by Failed to reset co-processor! and a fatal assert occurs consistently, always around 2.5s and 4.5s after boot, regardless of:

Which chip acts as RCP (I also reproduced this with a second ESP32-C6 as RCP)
Physical wiring (verified continuity, tried both GPIO16/17 and GPIO4/5 pin pairs)
Whether WiFi connection is running concurrently or the RCP init is isolated via a vTaskDelay() before WiFi starts (still fails even with zero concurrent WiFi activity)
Fresh flash / full power cycle of both chips

This suggests the root cause is in the Spinel driver initialization itself (esp_openthread_radio_spinel.cpp / spinel_driver.cpp), not board-specific hardware or wiring. Happy to provide full serial logs if useful.

gytxxsy, 2026-09-07:
> I'm seeing the exact same issue with a different two-chip combination: ESP32-C6 (host, running the ot_br example) + ESP32-H2-DevKitM-1 (RCP, running the ot_rcp example), both on ESP-IDF v5.4.2. [...]

Hi @macgaiser 

I tested ESP32-C6 (ot_br) + ESP32-H2 (ot_rcp) based on IDF v5.4.2, but I was unable to reproduce the issue you encountered. Could you please provide the following information?

1. The complete startup log.
2. Your sdkconfig files for both ot_br and ot_rcp.
3. A diagram showing the physical wiring between the two development boards. Alternatively, you can first refer to my wiring diagram:

| ot_br (ESP32-C6) | ot_rcp (ESP32-H2) |
|---|---|
| Pin 4 (UART RX) | TX (the pin is labeled "TX" on the board) |
| Pin 5 (UART TX) | RX (the pin is labeled "RX" on the board) |
| GND | GND |

Based on my previous testing, when the device fails to start properly, it is most often related to the wiring or the GPIO pins being used. Such issues are usually reflected in the log as follows:
```Bash
I (908) main_task: Calling app_ma)�I (908) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (918) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (928) OPENTHREAD: spinel UART interface initialization completed
I (918) main_task: Returned from app_main()
W(2938) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
W(4948) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
E(4948) OPENTHREAD:[C] P-SpinelDrive-: Failed to reset co-processor!

assert failed: void ot_spinel_coprocessor_reset_failure_callback(void*) esp_openthread_radio_spinel.cpp:82 (s_coprocessor_reset_failure_callback)
--- Stack dump detected
Core  0 register dump:

```