espressif/esp-idf #16094: Add bradge mode in softap_sta example

kaidegit, 2025-06-11:
### Describe the solution you'd like.

The example only shows bradge wifi using nat mode, not bridge mode. How can I use bridge mode to let the device connect to source wifi and the esp wifi has the same gateway and segment？

MaxwellAlan, 2025-08-07:
Hi @kaidegit 

Did you mean forward Wi-Fi pkts in OSI layer 2(MAC) ?

kaidegit, 2025-08-07:
>Did you mean forward Wi-Fi pkts in OSI layer 2(MAC) ?

Yes

tobinq, 2025-08-31:
>> Did you mean forward Wi-Fi pkts in OSI layer 2(MAC) ?

> Yes

as of my understanding, the wifi protocol does not allow an AP "client" to use frames with other than its own mac address (as afaik the mac is not only publicly visible but also bound to stay the same for the clients authentication to not break).
Thus layer 2 bridging - that is IF i am correct about this - cannot be done when connecting as a regular client to a regular AP at least not without some tunneling protocol (tap, l2tp, ..).

There are other modes (maybe wds p2p etc) that IMHO could do this but the softap_sta example seems to use just the regular AP / Client protocol.

However i stumbled over this issue when looking for how to just turn off NAT while still just forwarding layer 3 Packets which is sufficient for my own purposes (and i got it to work).

Do you consider it to be an option to foward Layer 3 instead of 2 ?