espressif/esp-idf #16096: ESP32-C6 WiFi: Bad broadcast response in power save mode (MIN_MODEM)

rojer, 2025-06-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-527-gadcbdd7da4

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1 v1.2

### Power Supply used.

USB

### What is the expected behavior?

Device's response to broadcast (ARP) requests should not be significantly impacted in MIN_MODEM PS mode.

### What is the actual behavior?

Compared to WIFI_PS_NONE, broadcast packet loss of up to 75% is observed in an uncongested network.
This is unacceptable as it impacts device's reachability  and responsiveness.

### Steps to reproduce.

Using the standard [power_save](https://github.com/espressif/esp-idf/tree/master/examples/wifi/power_save) example and `arping` to generate broadcast requests, observe device's behavior in NONE and MIN_MODEM ps modes. Make sure CPU light sleep is also enabled (`pm_config.light_sleep_enable = true`).

1. With PS disabled (`esp_wifi_set_ps()`), device responds to 100% arping requests.

2. With PS enabled (`esp_wifi_set_ps(WIFI_PS_MIN_MODEM)`), high packet loss is observed - 76% in this case.
```

### Debug Logs.

```plain
1. `WIFI_PS_NONE`: [Console log](https://github.com/user-attachments/files/20689591/cons_1.txt), [arping output](https://github.com/user-attachments/files/20689592/arping_1.txt)
2. `WIFI_PS_MIN_MODEM`: [Console log](https://github.com/user-attachments/files/20689590/cons_2.txt), [arping output](https://github.com/user-attachments/files/20689589/arping_2.txt)
```

rojer, 2025-06-16:
@beibeisi right, this may be a function of the AP used and maybe some other local environment conditions. what i can say is that signal is strong and noise level is acceptable: i.e., the spectrum is not overly busy. as you can see, with PM disabled the device has no problem responding to 100% of requests.
AP being used is a [TP-Link Archer C7 v5](https://www.tp-link.com/en/home-networking/wifi-router/archer-c7/), running [OpenWRT 24.10.1](https://openwrt.org/toh/tp-link/archer_c7).
how can i further assist you in debugging?

beibeisi, 2025-06-16:
hi @rojer , Could you also share the sdkconfig file you used?

rojer, 2025-06-16:
sure. nothing interesting there.

[sdkconfig.txt](https://github.com/user-attachments/files/20762020/sdkconfig.txt)

beibeisi, 2025-06-23:
hi @rojer 
I completely agree with you — ESP should indeed be compatible with all kinds of APs. Testing with a different AP is just to further verify whether the issue might be related to the AP. Could you please provide a packet capture file when the issue is reproduced? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

SupraSummus, 2026-10-05:
Possibly the same root cause, seen here with power save **off**.

**Setup:** ESP32-C6FH4 (QFN32) v0.2 (Seeed XIAO), arduino-esp32 3.3.11 (IDF v5.5), `WIFI_PS_NONE`.
AP: MikroTik wAP ax (Qualcomm, RouterOS 7.16.2, `wifi-qcom`), 2.4 GHz, DTIM 1, RSSI −78 dBm.

**Symptom:** after a DTIM beacon the AP sends buffered group frames back to back at 1 Mb/s DSSS, ~10 µs apart.
The C6 often misses a frame that follows one with More Data = 1.
Promiscuous mode (with `FCSFAIL`) shows a gap in the AP's sequence numbers and no failed frame:

```
air 59893987 group from this-sta len 380 seq 2420 rssi -78 format 0 rate 0 more 1   <- AP's copy of our DHCP DISCOVER
                                                                                     <- seq 2421 (the OFFER) never seen
air 60098811 group from this-sta len 80 seq 2422 rssi -78 format 0 rate 0 more 0
```

In that run the frame after More Data = 1 was missing 46 times in 47, after More Data = 0 never in 108.
It is intermittent: other runs lose fewer, some none.
A DHCP server's broadcast reply lands right after the AP's copy of the client's own broadcast,
so DHCP with the broadcast flag fails: 25 DISCOVERs → 3 OFFERs, 0 of 3 ACKs.

**Not the AP:** a Pico 2 W (CYW43439) associated to the same AP at the same time received all 21 replies; the C6 got 4.
(The C6 in that test ran the same Wi-Fi libraries under another OS; the sketch below loses the same way.)
No change with WPA2 vs WPA3, 11ax on/off, or promiscuous on/off.

In MIN_MODEM the station only listens to these post-DTIM bursts, which may explain the loss reported here.

Sketch, README and log: https://github.com/SupraSummus/rvuos/tree/76042020e85cd13fe1c20b6ec4318abb50e1b5bb/repro/esp32c6-group-loss