espressif/esp-idf #16129: ESP32-S3 examples\system\light_sleep fails with timeout > 500us

astrogene1000, 2025-06-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.1-dirty

### Espressif SoC revision.

ESP-ROM:esp32s3-20210327

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Wemos S3 Mini, other variants without USB uart chip

### Power Supply used.

USB

### What is the expected behavior?

Be able to set timer timeout > 500us

### What is the actual behavior?

Using timer timeout of 500us the wakeup fires as expected

Using timeout of 1000us the USB Uart initializes but timer wakeup never fires
--- Error: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
--- Waiting for the device to reconnect..

Using timeout value as default in the example of 2seconds
---Usb uart never initializes
--- Error: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
--- Waiting for the device to reconnect..
And Windows Dev manager reporting under USB controllers
Unknown USB device (Device Descriptor Request failed)

### Steps to reproduce.

1. idf.py set-target esp32s3
2. idf.py build
3. idf.py -p COMX flash monitor
4. Push reset on the S3 device without USB uart chip
...

### Debug Logs.

```plain

```

### More Information.

On S3 based devices with USB uart external chip works as expected

astrogene1000, 2025-06-14:
I first ran into this issue using examples for Arduino ESP32/S3 so switched to Espressif IDF and see the same thing, 500us and timer wakeup fires, 1000us and more things just seem to break.

BitsForPeople, 2025-06-15:
Note that it's (probably) not the example/ESP code "failing", but only the USB connection to the PC when the USB peripheral is powered down for sleep. "Error: ClearCommError failed ..." is a message generated on the PC side/python script, not on the ESP.

esp-wzh, 2025-06-16:
@BitsForPeople  You are right. The USJ (USB-Serial-JTAG) module will stop working in the lightsleep/deepsleep state due to the lack of necessary clocks (for power saving), so it will be disconnected from the USB Host (your PC). Therefore, if the chip goes to sleep, you cannot use this peripheral to print or debug code.

astrogene1000, 2025-06-18:
@esp-wzh @BitsForPeople 
Ok, thank you, understood. The example should have a disclaimer as to this though, if there is no auxiliary USB interface chip for at least S2's and S3's this 'example' will not have any serial messages displayed.

| Supported Targets | ESP32 | ESP32-C2 | ESP32-C3 | ESP32-C5 | ESP32-C6 | ESP32-C61 | ESP32-H2 | ESP32-P4 | ESP32-S2 | ESP32-S3 |
 **--->>** The example also **_prints time spent_** in light sleep mode to illustrate that timekeeping continues while the chip is in light sleep.

There is a disclaimer about a 'boot' button so not directly requiring an Espressif 'development board'.

 **-->>** This example can be used with any _**ESP32 development board**_. Most boards have a button attached to GPIO0, often labelled `BOOT`. If the board does not have such button, an external button can be connected, along with a 10k pull-up resistor, and a 100nF capacitor to ground for debouncing.

Thanks again!
(Hmm, not sure but my response from yesterday went 'poof' and not seen today)