espressif/esp-idf #16132: Example esp_http_client postman-echo example fails

gojimmypi, 2025-06-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm working on adding `esp-tls` support for wolfSSL as noted in https://github.com/wolfSSL/wolfssl/pull/7936 and PlatformIO Certificate Bundle support in https://github.com/wolfSSL/wolfssl/pull/8847

Today while testing, I encountered a problem where the [esp_http_client](https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_http_client) is failing to validate with the [https://postman-echo.com/post](https://github.com/espressif/esp-idf/blob/adcbdd7da4fca12cd48ca5338e3fa0e8cf6695ce/examples/protocols/esp_http_client/main/esp_http_client_example.c#L641) URL using the [postman_root_cert.pem](https://github.com/espressif/esp-idf/blob/adcbdd7da4fca12cd48ca5338e3fa0e8cf6695ce/examples/protocols/esp_http_client/main/postman_root_cert.pem#L1) certificate that is [baked in](https://github.com/espressif/esp-idf/blob/adcbdd7da4fca12cd48ca5338e3fa0e8cf6695ce/examples/protocols/esp_http_client/main/esp_http_client_example.c#L48) at build time.

The problem seems to be that the `postman-echo.com` domain is now using a Let's Encrypt Root certificate:

[image]

Whereas the `postman_root_cert.pem` certificate [file](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/postman_root_cert.pem) is currently an Amazon Root CA:

```text
$ openssl crl2pkcs7 -nocrl -certfile postman_root_cert.pem | openssl pkcs7 -print_certs -noout
subject=C = US, O = Amazon, CN = Amazon Root CA 1
issuer=C = US, O = Amazon, CN = Amazon Root CA 1
```

There's a  ISRG Root X1 PEM Let's Encrypt Certificate here:

https://letsencrypt.org/certs/isrgrootx1.pem.txt

When I download that cert and save it as `postman_root_cert.pem`, the example is successful for the https://postman-echo.com/post test.

See also:

```text
$  openssl crl2pkcs7 -nocrl -certfile postman_root_cert.pem | openssl pkcs7 -print_certs -noout
subject=C = US, O = Internet Security Research Group, CN = ISRG Root X1
issuer=C = US, O = Internet Security Research Group, CN = ISRG Root X1
```

I'm happy to create a PR to fix this if someone else can take a look and confirm.

I should also point out that the `howsmyssl` site does not fully support TLS 1.3 as noted in https://github.com/espressif/esp-idf/issues/14496 (see also https://github.com/jmhodges/howsmyssl/issues/716).

The tests using `howsmyssl` in the same `esp_http_client_example` will have unreliable and incorrect results for TLS1.3.

gojimmypi, 2025-06-15:
Hi @BitsForPeople - good point. I should have added that using the Certificate Bundle, (in a different example), I did not encounter the problem.  

The issue is only with the `https_async()` that assigns [cert_pem = postman_root_cert_pem_start](https://github.com/espressif/esp-idf/blob/adcbdd7da4fca12cd48ca5338e3fa0e8cf6695ce/examples/protocols/esp_http_client/main/esp_http_client_example.c#L644) using the local cert file like this:

```
extern const char postman_root_cert_pem_start[] asm("_binary_postman_root_cert_pem_start");
extern const char postman_root_cert_pem_end[]   asm("_binary_postman_root_cert_pem_end");
```

Here's the one in the Cert Bundle, good until June 4, 2035:

```
$ openssl x509 -in ISRG_root_X1.pem -text -noout
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            82:10:cf:b0:d2:40:e3:59:44:63:e0:bb:63:82:8b:00
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = US, O = Internet Security Research Group, CN = ISRG Root X1
        Validity
            Not Before: Jun  4 11:04:38 2015 GMT
            Not After : Jun  4 11:04:38 2035 GMT
        Subject: C = US, O = Internet Security Research Group, CN = ISRG Root X1
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                Public-Key: (4096 bit)
                Modulus:
                    00:ad:e8:24:73:f4:14:37:f3:9b:9e:2b:57:28:1c:
   [ .. snip .. ]

                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Key Usage: critical
                Certificate Sign, CRL Sign
            X509v3 Basic Constraints: critical
                CA:TRUE
            X509v3 Subject Key Identifier:
                79:B4:59:E6:7B:B6:E5:E4:01:73:80:08:88:C8:1A:58:F6:E9:9B:6E
    Signature Algorithm: sha256WithRSAEncryption
    Signature Value:
   [ .. snip .. ]
```

mahavirj, 2025-06-16:
@gojimmypi @BitsForPeople We have an internal fix for this issue to update the example certificate. It will be merged soon. Thanks for notifying!