espressif/esp-idf #16137: LP Core crashes when enabling interrupts after a chip reset without power drop

ImproperCatGirl, 2025-06-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1-361-g72de7111cc

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

both Custom Board and ESP32-DevKitC-1

### Power Supply used.

External 3.3V

### What is the expected behavior?

The LP core should run reliably, execute my code before, and after the initialization of interrupts, as long as  loaded binary by the HP core, turned on by the HP core,  and its power supply is on.

### What is the actual behavior?

Under certain conditions, it crashes at ulp_lp_core_intr_enable (), specifically, RV_SET_CSR(mstatus, MSTATUS_MIE); . After this, it entered a panic handler.

The "certain condition" is: when the power to the chip is not cut and remains at a stable 3.3V, reset the chip, then start the HP core, The HP core loads LP core's binary and start the LP CPU.

### Steps to reproduce.

1. Modify the official LP core GPIO interrupt program, add some dummy calculation just before the "ulp_lp_core_intr_enable();" in the LP core's code. Remove the deep sleep function in the HP core's code, since without it, debugging is impossible
2. Power on the ESP32-C6
3. try to use OpenOCD + GDB to debug it. Include "mon reset halt" at the top of the gdbinit file, then upon success (your code is executed properly after enabling the interrupt), stop OpenOCD and GDB.
4. do not turn off the power supply, try again, start OpenOCD and GDB with that gdbinit file with "mon reset halt", when hit the ulp_lp_core_intr_enable();, the LP core enters a panic handler.

You can automate this by writing a script: turn the power on, do the 3rd step, and then do the 4th step, if GDB says something like panic handler, stop and save the log. If the 4th step did not fail, turn off the power supply, wait for a few seconds, then turn it on again, start at the 2nd step. 

### Debug Logs.

```plain
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-linux-gnu --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/meow/Projects/experiments/lp_test/interrupt/gpio_intr_pulse_counter/build/lp_core_pulse_counter.elf...
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
0x408038f8 in esp_cpu_wait_for_intr ()
    at /home/meow/esp/v5.4/esp-idf/components/esp_hw_support/cpu.c:64
64	}
JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
[esp32c6.hp.cpu0] Reset cause (24) - (JTAG CPU reset)
add symbol table from file "/home/meow/Projects/experiments/lp_test/interrupt/gpio_intr_pulse_counter/build/esp-idf/main/ulp_main/ulp_main.elf"
Hardware assisted breakpoint 1 at 0x50001264: file /home/meow/Projects/experiments/lp_test/interrupt/gpio_intr_pulse_counter/main/ulp/main.c, line 51.
Breakpoint 2 at 0x50001bca: file /home/meow/esp/v5.4/esp-idf/components/ulp/lp_core/lp_core/lp_core_interrupt.c, line 30.
[Switching to Thread 2]

Thread 2 "esp32c6.lp.cpu" hit Temporary breakpoint 1, main ()
    at /home/meow/Projects/experiments/lp_test/interrupt/gpio_intr_pulse_counter/main/ulp/main.c:51
51	    lp_core_printf("LP Core pulse counter started\n");

Thread 2 "esp32c6.lp.cpu" hit Breakpoint 2, ulp_lp_core_intr_enable ()
    at /home/meow/esp/v5.4/esp-idf/components/ulp/lp_core/lp_core/lp_core_interrupt.c:30
30	    RV_SET_CSR(mstatus, MSTATUS_MIE);
_vector_table ()
    at /home/meow/esp/v5.4/esp-idf/components/ulp/lp_core/lp_core/port/esp32c6/vector_table.S:16
16		j _panic_handler
_panic_handler ()
    at /home/meow/esp/v5.4/esp-idf/components/ulp/lp_core/lp_core/vector.S:101
101	    save_general_regs RV_STK_FRMSZ
102	    save_mepc
104	    addi  t0, sp, RV_STK_FRMSZ /* restore sp with the value when trap happened */
107	    sw    t0, RV_STK_SP(sp)
108	    csrr  t0, mstatus
(gdb) Quit
Open On-Chip Debugger v0.12.0-esp32-20250226 (2025-02-26-14:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Hardware thread awareness created
Info : Hardware thread awareness created
WARNING: ESP flash support is disabled!
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (58:8C:81:40:E8:D4)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6.hp.cpu0] datacount=2 progbufsize=16
Info : [esp32c6.hp.cpu0] Core 0 could not be made part of halt group 1.
Info : [esp32c6.hp.cpu0] Examined RISC-V core; found 2 harts
Info : [esp32c6.hp.cpu0]  XLEN=32, misa=0x40903105
Info : [esp32c6.hp.cpu0] Examination succeed
Info : [esp32c6.lp.cpu] datacount=2 progbufsize=16
Info : [esp32c6.lp.cpu] Core 1 could not be made part of halt group 1.
Info : [esp32c6.lp.cpu] Examined RISC-V core; found 2 harts
Info : [esp32c6.lp.cpu]  XLEN=32, misa=0x40001105
Info : [esp32c6.lp.cpu] Examination succeed
Info : [esp32c6.hp.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c6.hp.cpu0] Target halted, PC=0x408038F8, debug_reason=00000008
Info : New GDB Connection: 1, Target esp32c6.hp.cpu0, state: halted
Error: get register list failed
Error: Unable to Generate Target Description
undefined debug reason 8 (UNDEFINED) - target needs reset
Info : JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6.hp.cpu0] Reset cause (24) - (JTAG CPU reset)
Info : [esp32c6.hp.cpu0] Found 4 triggers
Info : [esp32c6.lp.cpu] Found 2 triggers
Info : [esp32c6.hp.cpu0] Target halted, PC=0x408038F8, debug_reason=00000000
Info : [esp32c6.hp.cpu0] Target halted, PC=0x408038F8, debug_reason=00000000
Info : [esp32c6.hp.cpu0] Target halted, PC=0x408038F8, debug_reason=00000000
shutdown command invoked
```

### More Information.

note: without the dummy calculation, the program runs fine, without entering the panic handler. The dummy: "volatile double foo = 0.0;
volatile char fooo[100] = {0};
"(global variables) , then do "lp_core_printf("LP Core pulse counter started\n");
    
    for (int i = 0; i < 10000; i ++)
    {
        foo += 1;
        foo *= 111.23456788;
        fooo[(i%100)] = 'a';
    }
    ulp_lp_core_intr_enable();"

KonstantinKondrashov, 2025-06-18:
Hi @ImproperCatGirl!
> Modify the official LP core GPIO interrupt program

Could you point out which [one of the examples](https://github.com/espressif/esp-idf/tree/v5.4.1/examples/system/ulp/lp_core)is tested? 

FYI, `lp_core_printf` does not support float and double prints. Maybe this issue just a consequence of it.

KonstantinKondrashov, 2025-06-18:
Thanks for the clarification. I ran this example with your changes using v5.4.1. I do not observe such behavior. I suspect that a stack overflow happens. Can you try to increase the `CONFIG_ULP_COPROC_RESERVE_MEM`?

(FYI, LP core does not have a hardware FPU. The double precision floating point code will pull soft-float math routines, consuming hundreds to over 1 KB).

ImproperCatGirl, 2025-06-18:
I increased the reserved memory option's value to near maximum (16300), and replaced the potentially problematic float point dummy with a one second delay function, it still happens, and even became less erratic, I can observe one such event within 2 power cycles. 
My code on the ESP32-C6 now: 
```
lp_core_printf("LP Core pulse counter started\n");
    /*for (int i = 0; i < 10000; i ++)
    {
        foo_result = foo;
        foo_result *= 111.23456788;
        fooo[(i%100)] = 'a';
    }*/
    //lp_core_printf("%f", foo_result);

    ulp_lp_core_delay_us(1000000);

    ulp_lp_core_intr_enable();
    ulp_lp_core_gpio_intr_enable(CONFIG_EXAMPLE_PULSE_COUNT_PIN, LP_IO_INTR_POSEDGE);
    test_task();
```

If it does not violate rules, I can provide that automated testing script (written in python, power supply is a nordic PPK2).

KonstantinKondrashov, 2025-06-19:
Hi @ImproperCatGirl!
I still can not reproduce it. Maybe I'm missing something, correct me. I do the following:
- idf version `v5.4.1`
- modify the example [gpio_intr_pulse_counter](https://github.com/espressif/esp-idf/blob/v5.4.1/examples/system/ulp/lp_core/gpio_intr_pulse_counter/) according to [the latest comment](https://github.com/espressif/esp-idf/issues/16137#issuecomment-2984707547)
- commented `esp_deep_sleep_start();` from lp_core_pulse_counter_example_main.c
- sdkconfigs:
```
CONFIG_IDF_TARGET="esp32c6"
CONFIG_ULP_COPROC_ENABLED=y
CONFIG_ULP_COPROC_TYPE_LP_CORE=y
CONFIG_ULP_COPROC_RESERVE_MEM=8128
# Set log level to Warning to produce clean output
CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
CONFIG_BOOTLOADER_LOG_LEVEL=2
CONFIG_LOG_DEFAULT_LEVEL_WARN=y
CONFIG_LOG_DEFAULT_LEVEL=2
CONFIG_EXAMPLE_PULSE_COUNT_PIN=6
CONFIG_ULP_COPROC_ENABLED=y
CONFIG_ULP_COPROC_RESERVE_MEM=10000
```
- `.gdbinit` file
```
target remote :3333
monitor reset halt
continue
```
- LP-UART, TX-pin = GPIO5 is connected to a separate USB-UART, so I can see the output from `lp_core_printf()`. `Pulse count: 80, wake-up main CPU`
- Use the Onboard USB-JTAG port for debugging

Steps to reproduce it:
1. idf.py flash monitor
2. Connect OpenOCD with the following config
[image]
3. Connect GDB `riscv32-esp-elf-gdb -x .gdbinit`. I do not see that it is stacked at some points or a panic handler is called.
4. Add some pulses for GPIO6
5. Repeat from 2 to 4 a few times.

Seems I do the same as you described. Can you try the latest release version `v5.4.1`, I see that your idf version has a different tag version.

(You can enable this CONFIG_ULP_PANIC_OUTPUT_ENABLE=y to see the panic output at the LP-UART.)

ImproperCatGirl, 2025-06-19:
Upon further investigation, this issue ONLY happen if you exit gdb by killing it (pkill), not gracefully press ctrl+c, quit, y, then start the next session. 

The process is: 
Power ON, 
OpenOCD start, 
GDB start, 
pkill gdb, 
Send ctrl+c to OpenOCD, 
Start the next OpenOCD and then gdb session, and the problem will appear. 

Now, obviously, one would not expect the core to continue working properly after being forcefully detached off a debugger like GDB, but the issue is, the "mon reset halt" command can not resolve it, only a hard reset can.  (The EN pin or 3V3 disconnection)

Such corrupted state is the soft reset function's reason to exist : when something in the application goes wrong and corrupted something (SRAM, registers, peripherals, ..... ), bring the chip from a corrupt state to a known state and start from scratch. It failed to do this.

I used git to track all changed that have been made : 
```
git diff --cached --diff-filter=M $(git rev-list --max-parents=0 HEAD)
diff --git a/main/lp_core_pulse_counter_example_main.c b/main/lp_core_pulse_counter_example_main.c
index 65bef47..fa0fbf5 100644
--- a/main/lp_core_pulse_counter_example_main.c
+++ b/main/lp_core_pulse_counter_example_main.c
@@ -63,7 +63,7 @@ void app_main(void)
     /* Small delay to ensure the messages are printed */
     ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup());
 
-    esp_deep_sleep_start();
+    //esp_deep_sleep_start();
 }
 
 static void init_ulp_program(void)
diff --git a/main/ulp/main.c b/main/ulp/main.c
index 6a45c04..17caae8 100644
--- a/main/ulp/main.c
+++ b/main/ulp/main.c
@@ -21,6 +21,11 @@
 uint32_t pulse_count;
 static uint32_t last_trigger_time_cycles;
 
+
+volatile double foo = 0.0;
+volatile double foo_result = 0.0;
+volatile char fooo[100] = {0};
+
 void LP_CORE_ISR_ATTR ulp_lp_core_lp_io_intr_handler(void)
 {
     ulp_lp_core_gpio_clear_intr_status();
diff --git a/main/lp_core_pulse_counter_example_main.c b/main/lp_core_pulse_counter_example_main.c
index 65bef47..fa0fbf5 100644
--- a/main/lp_core_pulse_counter_example_main.c
+++ b/main/lp_core_pulse_counter_example_main.c
@@ -63,7 +63,7 @@ void app_main(void)
     /* Small delay to ensure the messages are printed */
     ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup());
 
-    esp_deep_sleep_start();
+    //esp_deep_sleep_start();
 }
 
 static void init_ulp_program(void)
diff --git a/main/ulp/main.c b/main/ulp/main.c
index 6a45c04..17caae8 100644
--- a/main/ulp/main.c
+++ b/main/ulp/main.c
@@ -21,6 +21,11 @@
 uint32_t pulse_count;
 static uint32_t last_trigger_time_cycles;
 
+
+volatile double foo = 0.0;
+volatile double foo_result = 0.0;
[... 113 lines cut ...]
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0
                                                                                                              0

                                                                                                              d

```

KonstantinKondrashov, 2025-06-23:
Hi @ImproperCatGirl!
Finally, I reproduced it! Could you test this fix? Does it help?

```diff
diff --git a/components/ulp/lp_core/lp_core/lp_core_interrupt.c b/components/ulp/lp_core/lp_core/lp_core_interrupt.c
index 66bcb86b013..fc823d21bad 100644
--- a/components/ulp/lp_core/lp_core/lp_core_interrupt.c
+++ b/components/ulp/lp_core/lp_core/lp_core_interrupt.c
@@ -26,6 +26,13 @@

 void ulp_lp_core_intr_enable(void)
 {
+
+    // Clear pending interrupts first
+    __asm__ volatile ("csrw mip, zero");
+
+    // Disable interrupt sources first (clean state)
+    __asm__ volatile ("csrw mie, zero");
+
     /* Enable interrupt globally */
     RV_SET_CSR(mstatus, MSTATUS_MIE);
     RV_SET_CSR(mie, MIE_ALL_INTS_MASK);
```

ImproperCatGirl, 2025-06-23:
This indeed fixed the issue that pkill gdb shows. Unfortunately, while this problem is fixed, the interrupt enabling function is still broken. If you introduce something that actually do request the lp core's peripheral to work on, like, send garbage to the i2c bus to simulate a failed I2C peripheral chip in the field application. (not waiting forever, I set the timeout to 5000 cpu cycles), it gets stuck mid-way:

```
git diff main/ulp/main.c
diff --git a/main/ulp/main.c b/main/ulp/main.c
index 17caae8..5617200 100644
--- a/main/ulp/main.c
+++ b/main/ulp/main.c
@@ -9,6 +9,8 @@
 #include "ulp_lp_core.h"
 #include "ulp_lp_core_utils.h"
 #include "ulp_lp_core_gpio.h"
+
+#include "ulp_lp_core_i2c.h"
 #include "ulp_lp_core_interrupts.h"
 #include "ulp_lp_core_print.h"
 #include "riscv/csr.h"
@@ -42,20 +44,23 @@ void LP_CORE_ISR_ATTR ulp_lp_core_lp_io_intr_handler(void)
     }
 
 }
-
 void test_task()
 {
     lp_core_printf("test_after_intr_enable");
 }
 
-
+void test_i2c()
+{
+    uint8_t garbage [3] = {0x1, 0x2, 0x3, };
+    lp_core_i2c_master_write_to_device(LP_I2C_NUM_0, 0x28, garbage, sizeof(garbage), 5000);
+}
 
 int main (void)
 {
     lp_core_printf("LP Core pulse counter started\n");
     
     ulp_lp_core_delay_us(1000000);
-
+    test_i2c();
     ulp_lp_core_intr_enable();
     
     ulp_lp_core_gpio_intr_enable(CONFIG_EXAMPLE_PULSE_COUNT_PIN, LP_IO_INTR_POSEDGE);

```
```
git diff main/lp_core_pulse_counter_example_main.c
diff --git a/main/lp_core_pulse_counter_example_main.c b/main/lp_core_pulse_counter_example_main.c
index fa0fbf5..6bedbc0 100644
--- a/main/lp_core_pulse_counter_example_main.c
+++ b/main/lp_core_pulse_counter_example_main.c
@@ -21,6 +21,9 @@
 #include "freertos/task.h"
 #include "ulp_lp_core.h"
 #include "ulp_main.h"
+
+#include "lp_core_i2c.h"
+
 #include "lp_core_uart.h"
 
 extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
@@ -29,6 +32,28 @@ extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");
 
 static void init_ulp_program(void);
 
+static void lp_i2c_init(void)
+{
+    esp_err_t ret = ESP_OK;
+
[... 158 lines cut ...]
ulp_lp_core_stop_lp_core () at /home/meow/esp/v5.4/esp-idf/components/ulp/lp_core/lp_core/lp_core_utils.c:150
150         lp_core_ll_set_wakeup_source(0);
(gdb) s
lp_core_ll_set_wakeup_source (flags=0)
    at /home/meow/esp/v5.4/esp-idf/components/hal/esp32c6/include/hal/lp_core_ll.h:110
110         PMU.lp_ext.pwr1.wakeup_en = flags;
(gdb) s
ulp_lp_core_stop_lp_core () at /home/meow/esp/v5.4/esp-idf/components/ulp/lp_core/lp_core/lp_core_utils.c:151
151         lp_core_ll_request_sleep();
(gdb) s
lp_core_ll_request_sleep () at /home/meow/esp/v5.4/esp-idf/components/hal/esp32c6/include/hal/lp_core_ll.h:126
126         PMU.lp_ext.pwr1.sleep_req = 1;
(gdb) s
[esp32c6.lp.cpu] Timed out after 5s waiting for busy to go low (abstractcs=0x10001002). Increase the timeout with riscv set_command_timeout_sec.
[esp32c6.lp.cpu] Timed out after 5s waiting for busy to go low (abstractcs=0x10001102). Increase the timeout with riscv set_command_timeout_sec.
[esp32c6.lp.cpu] Unexpected read of fp via program buffer.
[esp32c6.lp.cpu] Unable to enable triggers.
^CThe target is not running when halt was requested, stopping GDB.

Thread 2 "esp32c6.lp.cpu" received signal SIGINT, Interrupt.
0x40803900 in ?? ()
(gdb) quit

```
If this deviates too much, feel free to close this issue and then instruct me to fill another issue more relevant to the i2c driver, or if I'm using it in the wrong way, fell free to point it out.

closed 2025-06-25 by KonstantinKondrashov