espressif/esp-idf #16139: ESP32-C5  dm9051问题

znicky, 2025-06-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-527-gadcbdd7da4

### Espressif SoC revision.

ESP32-C5

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C5-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

ethernet/basic 示例程序在C5上运行

### What is the actual behavior?

在ESP32-C5上ehternet/basic运行有问题

### Steps to reproduce.

ethernet/basic

### Debug Logs.

```plain
I (2225) eth_example: Ethernet Started
I (2225) eth_example: Ethernet Link Up
I (2225) eth_example: Ethernet HW Addr 32:ed:a0:fe:ff:e4
I (2225) main_task: Returned from app_main()
E (5255) dm9051.mac: dm9051_frame_to_rx_buffer(792): invalid frame length, reset rx fifo pointer
E (5255) dm9051.mac: emac_dm9051_receive(817): moving data to internal rx_buffer failed
E (5255) dm9051.mac: frame read from module failed
E (5265) dm9051.mac: dm9051_frame_to_rx_buffer(792): invalid frame length, reset rx fifo pointer
E (5275) dm9051.mac: emac_dm9051_receive(817): moving data to internal rx_buffer failed
E (5275) dm9051.mac: frame read from module failed
E (10025) dm9051.mac: dm9051_frame_to_rx_buffer(792): invalid frame length, reset rx fifo pointer
E (10025) dm9051.mac: emac_dm9051_receive(817): moving data to internal rx_buffer failed
E (10025) dm9051.mac: frame read from module failed
E (10205) dm9051.mac: dm9051_frame_to_rx_buffer(792): invalid frame length, reset rx fifo pointer
E (10205) dm9051.mac: emac_dm9051_receive(817): moving data to internal rx_buffer failed
E (10215) dm9051.mac: frame read from module failed
E (10255) dm9051.mac: dm9051_frame_to_rx_buffer(792): invalid frame length, reset rx fifo pointer
E (10255) dm9051.mac: emac_dm9051_receive(817): moving data to internal rx_buffer failed
E (10265) dm9051.mac: frame read from module failed
E (11265) dm9051.mac: dm9051_frame_to_rx_buffer(792): invalid frame length, reset rx fifo pointer
E (11265) dm9051.mac: emac_dm9051_receive(817): moving data to internal rx_buffer failed
E (11265) dm9051.mac: frame read from module failed
```

znicky, 2025-06-16:
> 你的定制开发板是用 DM9051 驱动的吗？还是你用 SPI 以太网模块通过线连接到 ESP32C5？SPI 数据完整性可能损坏了……

使用的是DM9051外接模块，同个模块用ESP32C3的开发板测试没有问题

znicky, 2025-06-17:
> What is your sdkconfig?

ethernet/basic 示例 set-target esp32c5 默认生成的sdkconfig

kostaond, 2025-06-17:
OK, thanks. I can confirm that it really doesn't work. I'll have look on it in more details. Thanks you for reporting the issue!

kostaond, 2025-08-04:
Yes, this is problem with C5 SPI driver on the latest version of the C5 silicon. It's been fixed by a48afb4711c17f7badd80174fa2d48001d951a66

huhuhu111222, 2025-08-13:
esp32-c5使用w5500模块的问题解决了吗，我使用了esp-idf的master和v5.5两个版本的/esp-idf/examples/ethernet/basic都出现了：E (2353) w5500.mac: received frame was truncated这种问题

huhuhu111222, 2025-08-13:
Have you tried using the dm9051 module? Can it be used?

Alvin1Zhang, 2025-12-10:
Thanks for reporting, will close due to inactivity and can't be reproduced, feel free to reopen.