espressif/esp-idf #16142: Can P4 be added as supported chip ?

savenlid, 2025-06-16:
### Describe the solution you'd like.

I would like to see the P4 supported here.

igrr, 2025-06-16:
@savenlid It [already is](https://github.com/espressif/esp-idf?tab=readme-ov-file#esp-idf-release-and-soc-compatibility), could you please clarify your request? Are you missing support for P4 in some specific part of ESP-IDF?

savenlid, 2025-06-16:
Hi

In many places P4 is not included.
Example:
https://github.com/espressif/esp-idf/tree/adcbdd7da4fca12cd48ca5338e3fa0e8cf6695ce/examples/peripherals/usb/device/tusb_ncm

Specifically for USB i try to understand how the code would select USB port between the 3 avilable port of the P4.
For S2, S3 there is only 1 to choose from.
I admit I am in early stage of looking into RNDIS.
I want to access internal web-server in P4, not provide net-work access/routing to other interfaces ( ethernet-wifi)
So am also in the process of understanding static routing directives for netif.
Not covered by the normal examples that focus on being a "router".

Thanks
/NIcholas

igrr, 2025-06-16:
Thank you for the clarification, I have updated the issue title to indicate the example your are interested in.

tore-espressif, 2025-06-17:
Hello @savenlid ,

the `tusb_ncm` example shows how you can expose wifi network interface over USB. Since P4 does not have integrated WiFi, adding P4 support is not straighforward. We have plans to extend this example to also expose wired ethernet network connection, or P4 + WiFi companion chip (typically ESP32-C6), however I can't commit to any specific date as of now.

> select USB port between the 3 avilable port of the P4

Only 2 out of the 3 USB ports are available for custom USB functions. One is Full Speed (12Mbit/s) the other one is High Speed (480 Mbit/s). You can select the port in menuconfig `(Top) → Component config → TinyUSB Stack → USB Peripheral `

> I want to access internal web-server in P4

Typically, you'd want to access it through network interface (WiFi/wired ethernet) and only then extend with USB. Are you trying to expose the webserver via USB only?

savenlid, 2025-06-18:
I did fix that problem.
I now try to integrate the RNDIS interaction with netif from this examples/network/sta2eth example into my own project.

It goes fairly well, but...

i was missing some libs in linking.
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__tinyusb/libespressif__tinyusb.a(rndis_reports.c.obj): in function `rndis_query':

I did add to my example:
  ethernet_init:
    path: ${IDF_PATH}/examples/ethernet/basic/components/ethernet_init

2 problems:
#1, I dont want example code in my real-app
#2, idf.py complained about my project eth kconfigds beging in conflict with examples kconfig.
I removed it, I dont want to stray into example code directories and they kconfigs.

I still have the link-errors.
I try to solve them, being a seasoned coder, i lack esp32 enviroment skills, its very clever this environment, but also a bit complicated to understand, i am in a steep learning curve. :-)

I try to fix the below errors now, any hints are deeply appriciated.

FAILED: blink.elf
C:\Windows\system32\cmd.exe /C "cd . && C:\Espressif\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -nostartfiles -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32P4=0 -Wl,--Map=C:/workspace/batteries-chargers/esp32/ch1_p4/build/blink.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32p4.peripherals.ld -T esp32p4.rom.ld -T esp32p4.rom.api.ld -T esp32p4.rom.rvfp.ld -T esp32p4.rom.wdt.ld -T esp32p4.rom.systimer.ld -T esp32p4.rom.version.ld -T esp32p4.rom.newlib.ld -T memory.ld -T sections.ld -Wl,--enable-non-contiguous-regions @CMakeFiles\blink.elf.rsp -o blink.elf && cd ."
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__tinyusb/libespressif__tinyusb.a(ecm_rndis_device.c.obj): in function `handle_incoming_packet':
C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/src/class/net/ecm_rndis_device.c:338:(.text.handle_incoming_packet+0x76): undefined reference to `tud_network_recv_cb'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__tinyusb/libespressif__tinyusb.a(ecm_rndis_device.c.obj): in function `netd_open':
C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/src/class/net/ecm_rndis_device.c:182:(.text.netd_open+0x206): undefined reference to `tud_network_init_cb'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__tinyusb/libespressif__tinyusb.a(ecm_rndis_device.c.obj): in function `netd_control_xfer_cb':
C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/src/class/net/ecm_rndis_device.c:256:(.text.netd_control_xfer_cb+0xfc): undefined reference to `tud_network_init_cb'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__tinyusb/libespressif__tinyusb.a(rndis_reports.c.obj): in function `rndis_query':
C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/lib/networking/rndis_reports.c:117:(.text.rndis_query+0x216): undefined reference to `tud_network_mac_address'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/lib/networking/rndis_reports.c:117:(.text.rndis_query+0x21a): undefined reference to `tud_network_mac_address'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/lib/networking/rndis_reports.c:143:(.text.rndis_query+0x228): undefined reference to `tud_network_mac_address'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/workspace/batteries-chargers/esp32/ch1_p4/managed_components/espressif__tinyusb/lib/networking/rndis_reports.c:120:(.text.rndis_query+0x22c): undefined reference to `tud_network_mac_address'

roma-jam, 2025-06-18:
@savenlid 

`tud_network_init_cb` is the callback from the tinyusb, which should be declared in your application, but is is optional to use it. Is was removed in the last changes in the upstream: https://github.com/hathach/tinyusb/commit/b6ce41188e6780436dc9a9a0ed1232631753e3bb

`tud_network_mac_address` is the mac address buffer, you should have in your application: [example](https://github.com/hathach/tinyusb/blob/master/examples/device/net_lwip_webserver/src/main.c#L75)

savenlid, 2025-06-18:
Yes, it became clear after a while.

Callbacks etc.

I will read and merge and interpolate with the example.

Down to 1 link error now.

Thanks for all help.