espressif/esp-idf #16146: i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

moonlight729, 2025-06-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF：5.4.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-Touch-AMOLED-1.8(微雪开发板)

### Power Supply used.

USB

### What is the expected behavior?

我在乐鑫和火山的RTC Demo上增加了触摸屏应用个，正常可以i2c通信成功

### What is the actual behavior?

我在乐鑫和火山的RTC Demo上增加了触摸屏应用个，同样用到i2c通信。
提示驱动有冲突，RTC Demo应该使用了乐鑫封装的.a库，使用ES7210进行音频编解码。
目前触摸屏也用到i2c通信，怀疑.a库版本还是基于IDF4.x。
我现在工程统一是idf5.4

### Steps to reproduce.

增加触摸屏应用后，编译通过，直接运行。
触摸型使用的是SH8601

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x170c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xec4
load:0x403cb700,len:0x318c
entry 0x403c8950
I (26) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (26) boot: compile time Jun 16 2025 16:57:34
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (33) qio_mode: Enabling QIO for flash chip WinBond
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 ota_data         OTA data         01 00 00010000 00002000
I (82) boot:  3 factory          factory app      00 00 00020000 00300000
I (89) boot:  4 ota_0            OTA app          00 10 00320000 00300000
I (95) boot:  5 ota_1            OTA app          00 11 00620000 00300000
I (102) boot:  6 model            Unknown data     01 82 00920000 00410000
I (109) boot:  7 spiffs_data      Unknown data     01 82 00d30000 00010000
I (115) boot: End of partition table
I (119) boot: Defaulting to factory image
I (122) esp_image: segment 0: paddr=00020020 vaddr=3c1e0020 size=904b8h (591032) map
I (217) esp_image: segment 1: paddr=000b04e0 vaddr=3fca7900 size=08aa8h ( 35496) load
I (224) esp_image: segment 2: paddr=000b8f90 vaddr=40378000 size=07088h ( 28808) load
I (230) esp_image: segment 3: paddr=000c0020 vaddr=42000020 size=1dc504h (1950980) map
I (520) esp_image: segment 4: paddr=0029c52c vaddr=4037f088 size=187b8h (100280) load
I (539) esp_image: segment 5: paddr=002b4cec vaddr=600fe100 size=0001ch (    28) load
I (552) boot: Loaded app from partition at offset 0x20000
I (552) boot: Disabling RNG early entropy source...
I (563) octal_psram: vendor id    : 0x0d (AP)
I (563) octal_psram: dev id       : 0x02 (generation 3)
I (563) octal_psram: density      : 0x03 (64 Mbit)
I (565) octal_psram: good-die     : 0x01 (Pass)
I (569) octal_psram: Latency      : 0x01 (Fixed)
I (574) octal_psram: VCC          : 0x01 (3V)
I (578) octal_psram: SRF          : 0x01 (Fast Refresh)
I (583) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (587) octal_psram: BurstLen     : 0x01 (32 Byte)
I (592) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (597) octal_psram: DriveStrength: 0x00 (1/1)
I (602) MSPI Timing: PSRAM timing tuning index: 5
I (606) esp_psram: Found 8MB PSRAM device
I (609) esp_psram: Speed: 80MHz
I (612) cpu_start: Multicore app
I (901) esp_psram: SPI SRAM memory test OK
I (909) cpu_start: Pro cpu start user code
I (909) cpu_start: cpu freq: 240000000 Hz
I (909) app_init: Application information:
I (910) app_init: Project name:     robot_q
I (913) app_init: App version:      v1.0-dirty
I (917) app_init: Compile time:     Jun 16 2025 16:57:23
I (922) app_init: ELF file SHA256:  ed3b0ed2d...
I (927) app_init: ESP-IDF:          v5.4.1-dirty
I (931) efuse_init: Min chip rev:     v0.0
I (935) efuse_init: Max chip rev:     v0.99
I (939) efuse_init: Chip rev:         v0.2
I (943) heap_init: Initializing. RAM available for dynamic allocation:
I (949) heap_init: At 3FCC8240 len 000214D0 (133 KiB): RAM
I (954) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (959) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (965) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (972) spi_flash: detected chip: winbond
I (975) spi_flash: flash io: qio
E (978) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x42128838 on core 0
--- 0x42128838: check_i2c_driver_conflict at /home/xingzb/esp/esp-adf/esp-idf/components/driver/i2c/i2c.c:1723

Backtrace: 0x4037a225:0x3fceb1c0 0x40389869:0x3fceb1e0 0x40392232:0x3fceb200 0x42128838:0x3fceb270 0x42003e96:0x3fceb290 0x40379d29:0x3fceb2c0 0x403ccbb8:0x3fceb340 0x403ccff5:0x3fceb380 0x403c89b9:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
--- 0x4037a225: panic_abort at /home/xingzb/esp/esp-adf/esp-idf/components/esp_system/panic.c:454
--- 0x40389869: esp_system_abort at /home/xingzb/esp/esp-adf/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x40392232: abort at /home/xingzb/esp/esp-adf/esp-idf/components/newlib/abort.c:38
--- 0x42128838: check_i2c_driver_conflict at /home/xingzb/esp/esp-adf/esp-idf/components/driver/i2c/i2c.c:1723
--- 0x42003e96: do_global_ctors at /home/xingzb/esp/esp-adf/esp-idf/components/esp_system/startup.c:104
---  (inlined by) start_cpu0_default at /home/xingzb/esp/esp-adf/esp-idf/components/esp_system/startup.c:206
--- 0x40379d29: call_start_cpu0 at /home/xingzb/esp/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:849 (discriminator 1)
--- 0x40045c01: ets_run_flash_bootloader in ROM
--- 0x40043ab6: main in ROM
--- 0x40034c45: .stack_ok in ROM
```

mythbuster5, 2025-06-17:
This is not an esp-idf issue. Please find which `.a` used legacy driver and raise a question there.

moonlight729, 2025-06-17:
目前看./components/clouds/volc_engine_rtc/libs/esp32s3/libVolcEngineRTCLite.a
这个.a 应该是使用了新版本驱动，但是触摸屏相关驱动，tca9554 使用的是旧版本驱动

moonlight729, 2025-06-17:
`void app_main(void)
{
    esp_log_level_set("lcd_panel.io.i2c", ESP_LOG_NONE);
    esp_log_level_set("FT5x06", ESP_LOG_NONE);
    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    // // 扩展IO芯片，基于I2C通信
    ESP_LOGI(TAG, "Initialize I2C bus");
    const i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = EXAMPLE_PIN_NUM_TOUCH_SDA,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = EXAMPLE_PIN_NUM_TOUCH_SCL,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 200 * 1000,
    };
    ESP_ERROR_CHECK(i2c_param_config(TOUCH_HOST, &i2c_conf));
    ESP_ERROR_CHECK(i2c_driver_install(TOUCH_HOST, i2c_conf.mode, 0, 0, 0));

    esp_io_expander_handle_t io_expander = NULL;
    esp_io_expander_new_i2c_tca9554(TOUCH_HOST, ESP_IO_EXPANDER_I2C_TCA9554_ADDRESS_000, &io_expander);

    ESP_ERROR_CHECK(esp_io_expander_new_i2c_tca9554(TOUCH_HOST, ESP_IO_EXPANDER_I2C_TCA9554_ADDRESS_000, &io_expander));

    esp_io_expander_set_dir(io_expander, IO_EXPANDER_PIN_NUM_0 | IO_EXPANDER_PIN_NUM_1 | IO_EXPANDER_PIN_NUM_2, IO_EXPANDER_OUTPUT);
    esp_io_expander_set_level(io_expander, IO_EXPANDER_PIN_NUM_0, 0);
    esp_io_expander_set_level(io_expander, IO_EXPANDER_PIN_NUM_1, 0);
    esp_io_expander_set_level(io_expander, IO_EXPANDER_PIN_NUM_2, 0);
    vTaskDelay(pdMS_TO_TICKS(200));
    esp_io_expander_set_level(io_expander, IO_EXPANDER_PIN_NUM_0, 1);
    esp_io_expander_set_level(io_expander, IO_EXPANDER_PIN_NUM_1, 1);
    esp_io_expander_set_level(io_expander, IO_EXPANDER_PIN_NUM_2, 1);

#if EXAMPLE_PIN_NUM_BK_LIGHT >= 0
    ESP_LOGI(TAG, "Turn off LCD backlight");
    gpio_config_t bk_gpio_config = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
#endif

    // LCD采用QSPI总线，传输速度相比标准SPI更快
    ESP_LOGI(TAG, "Initialize SPI bus");
    // QSPI 硬件总线配置
    const spi_bus_config_t buscfg = SH8601_PANEL_BUS_QSPI_CONFIG(EXAMPLE_PIN_NUM_LCD_PCLK,
                                                                 EXAMPLE_PIN_NUM_LCD_DATA0,
                                                                 EXAMPLE_PIN_NUM_LCD_DATA1,
                                                                 EXAMPLE_PIN_NUM_LCD_DATA2,
                                                                 EXAMPLE_PIN_NUM_LCD_DATA3,
                                                                 EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES * LCD_BIT_PER_PIXEL / 8);
    ESP_ERROR_CHECK(spi_bus_initialize(LCD_HOST, &buscfg, SPI_DMA_CH_AUTO));

    ESP_LOGI(TAG, "Install panel IO");
    esp_lcd_panel_io_handle_t io_handle = NULL;
    // QSPI 软件配置
    const esp_lcd_panel_io_spi_config_t io_config = SH8601_PANEL_IO_QSPI_CONFIG(EXAMPLE_PIN_NUM_LCD_CS,
                                                                                example_notify_lvgl_flush_ready,
                                                                                &disp_drv);
    // 初始化配置                                                                                
    sh8601_vendor_config_t vendor_config = {
        .init_cmds = lcd_init_cmds,
        .init_cmds_size = sizeof(lcd_init_cmds) / sizeof(lcd_init_cmds[0]),
        .flags = {
            .use_qspi_interface = 1,
        },
    };
    // Attach the LCD to the SPI bus
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)LCD_HOST, &io_config, &io_handle));

[... 86 lines cut ...]
#endif

    lvgl_mux = xSemaphoreCreateMutex();
    assert(lvgl_mux);
    xTaskCreate(example_lvgl_port_task, "LVGL", EXAMPLE_LVGL_TASK_STACK_SIZE, NULL, EXAMPLE_LVGL_TASK_PRIORITY, NULL);

    ESP_LOGI(TAG, "Display LVGL demos");
    // Lock the mutex due to the LVGL APIs are not thread-safe
    if (example_lvgl_lock(-1))
    {

        // lv_demo_widgets(); /* A widgets example */
        lv_demo_music(); /* A modern, smartphone-like music player demo. */
        // lv_demo_stress();       /* A stress test for LVGL. */
        // lv_demo_benchmark(); /* A demo to measure the performance of LVGL or to compare different settings. */

        // Release the mutex
        example_lvgl_unlock();
    }
}
`

这一段代码，单独运行，会提示

W (708) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`

moonlight729, 2025-06-17:
@mythbuster5 帮忙看下如何调整

suda-morris, 2025-06-18:
@moonlight729 
Please use the latest version of the TCA9554 driver, which is adapted to the new version of the I2C bus driver： https:components.espressif.com/components/espressif/esp_io_expander_tca9554/versions/2.0.1

moonlight729, 2025-06-18:
@suda-morris Hi,what about the ESP LCD Touch FT5x06 Controller driver,is there new version using the new i2c  structure?

mythbuster5, 2025-06-18:
新旧版本的I2C界限是明确的，只要你旧版的i2c 的某个函数被链接进来就一定会报错。你看到他调用了不一定会被链接。如果你是在idf 5.4版本上用，没有特殊的config调整，adf链接的是新驱动 https://github.com/espressif/esp-adf/blob/4200c64d4a78b9e9e2297818733aff21c72079a1/components/esp_peripherals/driver/i2c_bus/i2c_bus.h#L33

mythbuster5, 2025-06-23:
Issue closed.

closed 2025-06-23 by mythbuster5