espressif/esp-idf #16155: bt_avrc comile issue with cover art feature disabled

boborjan2, 2025-06-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Stack should compile with valid config options.

### What is the actual behavior?

Having avrcp configured but cover art support not brakes compilation at btc_avrc,c, line 1694, as variables are not guarded by "#if BTC_AV_CA_INCLUDED".

### Steps to reproduce.

CONFIG_BT_AVRCP_ENABLED=y
# CONFIG_BT_AVRCP_CT_COVER_ART_ENABLED is not set

### Build or installation Logs.

```plain
/home/v/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c: In function 'btc_avrc_get_profile_status':

/home/v/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:1695:22: error: 'btc_rc_cb_t' has no member named 'rc_cover_art_connected'

 1695 |         if (btc_rc_cb.rc_cover_art_connected) {

      |                      ^

make[4]: *** [esp-idf/bt/CMakeFiles/__idf_bt.dir/build.make:1993: esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/btc/profile/std/avrc/btc_avrc.c.obj] Error 1
```

esp-lrh, 2025-06-19:
Hi, @boborjan2 

This problem is caused by a recent change, we will fix it as soon as possible. You can the apply following changes temporarily:

```
diff --git a/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c b/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c
index 3ed8a02ad70..0e620097e18 100644
--- a/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c
+++ b/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c
@@ -1724,9 +1724,11 @@ void btc_avrc_get_profile_status(esp_avrc_profile_status_t *param)
         if (btc_avrc_ct_init_p()) {
             param->avrc_ct_inited = true;
         }
+#if BTC_AV_CA_INCLUDED
         if (btc_rc_cb.rc_cover_art_connected) {
             param->ct_cover_art_conn_num++;
         }
+#endif
     }
 }

```

Thanks for your feedback!

closed 2025-06-22 by espressif-bot

boborjan2, 2025-07-28:
This solution is nat part of v5.5 (while 5.4 does have it).