espressif/esp-idf #16159: ESP32-S2 enters low-power / undefined state after GND disconnection or slow power ramp-up

DiegoBittencourt-55, 2025-06-18:
**Answers checklist:**

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

**IDF version:**
ESP-IDF v5.1.2 and v5.4.1 (tested on both, same behavior)

**Espressif SoC revision:**
ESP32-S2FH4, chip revision v0.0 and v1.0 (both affected)

**Operating System used:**
Windows 11

**How did you build your project?**
Command line with idf.py and ESP-IDF VSCode extension

**If you are using Windows, please specify command line type:**
None

**Development Kits used:**
Custom board with ESP32-S2FH4 + external crystal
DevKit: ESP32-S2-MINI-1 (YD-ESP32-S2-mini 2022-v1.2 from VCC-GND.Studio)

**Power Supply used:**
Custom board: 12V -> 5V (SMPS) -> 3.3V (LDO) with 100uF + Espressif-recommended bypass caps. Or Direct 3.3V.
DevKit: tested via 5V (USB) and direct 3.3V injection (behavior more replicable with direct 3.3V)

**Expected behavior:**
After power restoration or GND reconnection, ESP32-S2 should boot and run user code reliably.

**Actual behavior:**
When the GND pin is disconnected and reconnected repeatedly, or if 3.3V decays slowly (such as with high capacitance), the ESP32-S2 enters a state where:

- No code is executed
- Power consumption stays around ~1mA (same as deep sleep or holding EN pin low)
- The chip only recovers after either:
- Pressing the EN (reset) button
- Completely powering off and re-powering after a few seconds

**Steps to reproduce:**
Flash simple LED blinking code on the custom board or ESP32-S2-MINI-1 DevKit:
void app_main(void) {
    // Blink LEDs on GPIOs 36, 37, 21 with 1s delay
}
Power the device with 3.3V from a regulated bench power supply.

Quickly disconnect and reconnect GND several times, or decrease the 3.3V supply slowly (i.e., simulating a large capacitor discharge by decreasing it to ~2.2V and increasing it back to 3.3V).

Note that eventually the ESP32-S2 stops executing code (no LED activity, no serial logs) and consumes 1~3mA.

Recovery is only possible via the EN button or a full power-down cycle.

Another, easier way to reproduce this issue is to force a reboot while the device is at low voltage:

Reduce the supply voltage to about 2.2V, press the EN button to reset the ESP32-S2, and then restore the voltage to 3.3V.

This reliably triggers the same issue.

**Observations:**

- Problem occurs with both chip revisions (v0.0 and v1.0).
- Occurs on custom boards and official ESP32-S2-MINI-1 DevKit.
- Easier to reproduce when powering via 3.3V directly than 5V via onboard regulator.
- Occurs regardless of CPU frequency (80/160/240MHz), and with both enabled/disabled brownout detector (voltage levels 3.19V, 2.67V, 2.44V, 2.56V).
- Occurs even with Bootloader config features enabled (factory reset trigger or test partition boot GPIO).
- Attempted mitigation via bootloader GPIO triggers had no effect.
- Oscilloscope shows that when large output capacitors are present on 3.3V rail, the power line decays slowly (linear discharge curve) during GND disconnection, which may be preventing internal brownout reset.
- Cannot reproduce same behavior with ESP32-S3FN8 or S3-WROOM-1 under same conditions.
- Log (looping boot when brownout is triggered):
- Attached above, showing successful boot, then brownout reset, then undefined behavior or reset loop. Eventually the chip stops responding completely.

**More information:**

- We have over 30,000 units manufactured with this design, and this problem represents a serious field reliability concern. We're looking for confirmation:
- Is this behavior expected or a known silicon limitation of ESP32-S2?
- Are there known design practices or software settings to prevent this lockup?
- Is it possible to force a software-level recovery from this state (e.g. via deep sleep or WDT)?
- Any recommended workaround applicable without hardware redesign?

We would appreciate guidance on this issue, especially because the problem does not occur on ESP32-S3 under similar scenarios.

**In summary:**
If the ESP32-S2 restarts or starts while its voltage is below the minimum value (e.g. around 2.2V) — a possible situation when there is a disconnection of the GND or partial shutdown —, the chip may enter an undefined state in which it does not execute the application code. This state is only reversed by pressing the EN button or by performing a complete power off cycle. This scenario occurs frequently in the field, because during maintenance of the equipment monitored by our product, it is common for the GND of the main system to be temporarily disconnected, directly affecting the behavior of the embedded ESP32-S2.

[logs.txt](https://github.com/user-attachments/files/20802150/logs.txt)

esp-wzh, 2025-06-30:
Hi, @DiegoBittencourt-55, does this problem happen on any of your esp32s2 chips? And can you post your sdkconfig?

I tried to reproduce this problem using IDF's [helloworld example](https://github.com/espressif/esp-idf/tree/v5.4.1/examples/get-started/hello_world), but it didn't work. I connected the chip to a regulated power supply and slowly reduced the voltage， I observed that the brownout detector works fine and the chip resets once the voltage drops below ~2.45V.

[image]

DiegoBittencourt-55, 2025-07-01:
Hi @esp-wzh,

I've attached the sdkconfig I used 

[sdkconfig.txt](https://github.com/user-attachments/files/20990400/sdkconfig.txt).

This issue is occurring on all of my ESP32-S2 chips.

I can confirm that the brownout detector does work in general, but in specific scenarios where it should trigger, it doesn’t. The issue is easy to reproduce by lowering the supply voltage below 2.2V (typically down to 1.8V). In that case, when I bring the voltage back to 3.3V, the chip does not boot unless I either press the RST button (EN pin) or completely power cycle the board.

Here’s a https://youtube.com/shorts/5mZwpjykWIg?feature=shared demonstrating this behavior using the ESP32-S2 Mini DevKit.

In the second video, https://youtube.com/shorts/8kftaIsmChQ?feature=shared, I replicate a similar scenario to what happens in my deployment: the GND connection of the board is temporarily disconnected during maintenance. When GND is reconnected, there's a momentary instability or noise on the GND line until it stabilizes. This resembles the case of hot-plugging a device by first connecting the positive wire to a battery, and only then the negative (GND).

Both the sdkconfig and the test code used are attached.

Let me know if any other information would be helpful.

I also attached the example code I used in the video.
[blink_example_main.txt](https://github.com/user-attachments/files/20990411/blink_example_main.txt)

boarchuz, 2025-07-01:
This would be expected for an ESP32, as documented in the datasheet (2.3.2) and countless forum posts, but I see that the ESP32S2 datasheet doesn't specifically cover this behaviour in its datasheet. Was there a change in hardware? Or should the S2 datasheet also explain this?

Specifically:
>When a battery is used as the power supply for the ESP32 series of chips and modules, a supply voltage supervisor is recommended, **so that a boot failure due to low voltage is avoided**. Users are recommended to **pull CHIP_PU low if the power supply for ESP32 is below 2.3 V**.

esp-wzh, 2025-07-08:
Wait... I just learned that this info was deleted on s2 because the SWD peripheral was added to esp32s2, which triggers a reset when the digital circuit does not work properly.

I tried to reproduce the problem with the example you provided. When VDD_3V3 dropped to ~1.8V, the log stopped, and then when VDD_3V3 rose to 3.3V, the chip was still stuck, but when the voltage continued to rise to ~3.4V, the chip triggered the power_on reset and returned to normal. I guess SWD worked. If this doesn't work for you, you can try turning off the SWD auto feed and feed the SWD by a CPU task to monitor the chips working status (See section 12.4 in [Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf)).