espressif/esp-idf #16193: Binary size bloating due to libstdc++ floating_to_chars std::to_chars and ryu looking up table

wuyuanyi135, 2025-06-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# Environment
OS: Windows 11
idf.py: v5.5-release
optimization flag: -Os

# Problem description
I am analyzing the size of the binary emitted by my C++ project and observed a huge increase in binary size. `idf.py size-files` indicates that an object file `floating_to_chars.o` took 12568 bytes of `.text` and 117059 bytes of data. [Map viewer](https://github.com/govind-mukundan/MapViewer) showed that the largest symbol was `(anonymous namespace)::ryu::POW10_SPLIT_2` and `(anonymous namespace)::ryu::POW10_SPLIT`. Together they took more than 100K of binary size.

[image]

The Ryu provides a lookup table for fixed-point conversion. By tracing upstream issues I found the [issue](https://github.com/llvm/llvm-project/issues/64180) discussing the binary size impact caused by Ryu. I believe such an optimization is even more critical for embedded system like esp32. 

Is there any advice about optimizing the C++ binary size when it involves these huge built-in objects?
# Attachment

Attaches the `.map` and `.elf` files for size analysis. 

[build.zip](https://github.com/user-attachments/files/20903972/build.zip)

atanisoft, 2025-06-25:
does this actually show up in the "bin" file which gets uploaded? The "elf" file itself isn't uploaded and may contain extra bits in some cases.

wuyuanyi135, 2025-06-25:
@igrr according to https://github.com/llvm/llvm-project/issues/64180 it might be feasible to use snprintf to override std::to_chars to to trade speed for space. I havent figured how to do that. My point is that a kconfig flag switching between implementations might be very helpful.

wuyuanyi135, 2025-06-26:
@igrr Thanks for pointing out the related source! I believe the symbol was pulled in by `fmt::format`. BTW, could the [iostream bloat](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/cplusplus.html#iostream) be also (partially) related to this table?

wuyuanyi135, 2025-06-30:
@igrr any chance the devs will take a look on the possible option (link wrapping) recently? I am not sure if there will be a simple solution. I will look into it these days

d4rkmen, 2025-10-15:
Hi, any working solution to get rid of this 128kb `floating_to_chars.o    |     128791` in ESP_IDF ? it appears even if not use floating point format

GoeBachmann, 2026-10-27:
Another solution is to use `fmt::format` instead of `std::format` or `std::to_chars`. The implementation in fmt 12.2.0 is much smaller on ESP than the libstdc++ implementation.