espressif/esp-idf #16195: USB Device: tud_hid_keyboard_report() skips keycode

dventurb, 2025-06-26:
Hello, 
I'm working on USB HID Keyboard using TinyUSB, and I'm encountering an error when sending keycodes in the array of 6 via tud_hid_keyboard_report(). 
I'm parsing an ASCII char by char, converted using a array. I send HID key reports in array of 6 and is flush when one of these conditions is true: 

1. The buffer reaches 6.
2. Its the last char in data

For the string `hapa.pt`, I expect it full string to be typed and it work when I send keycode individually, but when I send together in the array of 6 I get this `hap.t` .

```c
void read_line() {
  uint8_t *data = malloc(512);
  if(!data) return;

  int len = uart_read_bytes(UART_NUM_1, data, (512 - 1), pdMS_TO_TICKS(50));
  
  if(len > 0) {
    int counter = 0;
    uint8_t keycode[6] = {0}; 
    for (int i = 0; i < len; i++) {
      uint8_t entry = keymap[(uint8_t)data[i]];
      
      if(counter < 6) {
        keycode[counter] = entry;
        counter++;
      }

      if(counter == 6 || i + 1 == len) {

        tud_hid_keyboard_report(HID_ITF_PROTOCOL_KEYBOARD, 0, keycode);
        vTaskDelay(pdMS_TO_TICKS(20));
        
        tud_hid_keyboard_report(HID_ITF_PROTOCOL_KEYBOARD, 0, NULL);
        vTaskDelay(pdMS_TO_TICKS(20));

        counter = 0;
        memset(keycode, '\0', sizeof(keycode)); 
      }
    }
  }
  free(data);
  data = NULL;

  return;
}
``` 

I'm trying to understand if the problem is in my logic. Can someone help me understand why this happens? 
Thanks!

roma-jam, 2025-06-26:
Hi @dventurb, 

I can see, that you are reading data from the uart and then want to "type" the same data via keyboard. 

One important note regarding the HID reports protocol: it sends the keyscan values of the keys, that have been pressed. 
Basic boot report contain 6 key scan positions, which means you are able to tell the host that only 6 keys were pressed via 1 report transaction. 

If you have the specific symbols, which require modifiers (such as symbols that requires SHIFT being pressed), then I would suggest to do it with real typing emulation: one key - one report. 

So, if we have got example the data such as `hapa.pt` in reports it would be: 

```
// HID Report Data buffer, all values in HEX

00 00 0B 00 00 00 00 00  // 'h' key pressed 
00 00 00 00 00 00 00 00  // 'h' key released
00 00 04 00 00 00 00 00  // 'a' key pressed
00 00 00 00 00 00 00 00  // 'a' key released
00 00 13 00 00 00 00 00  // 'p' key pressed
00 00 00 00 00 00 00 00  // 'p' key released
00 00 37 00 00 00 00 00  // '.' key pressed
00 00 00 00 00 00 00 00  // '.' key released
00 00 13 00 00 00 00 00  // 'p' key pressed
00 00 00 00 00 00 00 00  // 'p' key released
00 00 17 00 00 00 00 00  // 't' key pressed
00 00 00 00 00 00 00 00  // 't' key released
```

If you need a modifier, you need to set it in the report: 

```
// HID Report Data buffer, all values in HEX
02 00 00 00 00 00 00 00  // Left Shift modifier pressed
00 00 00 00 00 00 00 00  // Left Shift modifier released
```

In this case you can emulate any key pressed event + pressed keys under the modifier (if the modifier key still pressed during the other key pressed event). 

HID Boot Report data structure could be found here: [ESP-USB Host HID Class Driver, hid_usage_keyboard.h](https://github.com/espressif/esp-usb/blob/master/host/class/hid/usb_host_hid/include/usb/hid_usage_keyboard.h#L272)

roma-jam, 2025-06-26:
Hi @dventurb, 

Great, glad that it helped! 

I‘ll close it as not planned as there are no problems with esp-idf here.

closed 2025-06-26 by roma-jam