espressif/esp-idf #16198: UART does not generate UART_WAKEUP events

doragasu, 2025-06-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 nodemcu like board

### Power Supply used.

USB

### What is the expected behavior?

The provided example, sets up the system, setups UART 0, configures light sleep for the UART to wakeup from light sleep on 3 edges, and enters light sleep.

When data is sent to UART 0, the system should wake up and a UART_WAKEUP event should be generated, causing the LED (connected to GPIO47) to toggle.

The example also uses GPIO48 to know when the device is in light sleep. This pin is set to 1 while active, and goes to 0 while in light sleep.

### What is the actual behavior?

But sending UART data does not wake the system. The LED on GPIO47 never toggles, while the LED on GPIO48 toggles briefly each 200 ms, but is not affected by UART activity (sending UART data does not appear to wake the system).

### Steps to reproduce.

1. Create a project for ESP32-S3 target. Put this in `sdkconfig.defaults` file, for logs to go to USB and to enable FreeRTOS tickless idle mode:

```
CONFIG_PM_ENABLE=y
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
```

2. Put this code in the main source file:

```C
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include <driver/uart.h>
#include <driver/gpio.h>
#include <esp_pm.h>
#include <esp_sleep.h>
#include <esp_log.h>

#include <soc/uart_reg.h>

// Map ESP_LOGE() macros to something shorter
#define LOGE(...) ESP_LOGE(__func__, __VA_ARGS__)
#define LOGD(...) ESP_LOGD(__func__, __VA_ARGS__)
#define LOGI(...) ESP_LOGI(__func__, __VA_ARGS__)
#define LOGW(...) ESP_LOGW(__func__, __VA_ARGS__)

// Uart device to use
#define UART_DEV UART_NUM_0
// Receive event queue size
#define EVENT_QUEUE_SIZE 16

// LED connected to GPIO47, toggles when an UART WAKEUP event occurs
#define LED_GPIO 47
// This pin is set to 1 while CPU is awake, and goes to 0 when in light sleep
#define AWAKE_GPIO 48

// Receive event queue
static QueueHandle_t q_evt = NULL;

static esp_err_t light_sleep_enable(void)
{
	const esp_pm_config_t pm_config = {
		.max_freq_mhz = 160,
		.min_freq_mhz = 40,
		.light_sleep_enable = true // enable auto light sleep
	};

	return  uart_set_wakeup_threshold(UART_DEV, 3) ||
		esp_sleep_enable_uart_wakeup(UART_DEV) ||
		esp_pm_configure(&pm_config);
}

static void led_toggle(void)
{
	static uint32_t on = 1;

	on ^= 1;

	gpio_set_level(LED_GPIO, on);
}

static esp_err_t io_setup(void)
{
	const gpio_config_t io_cfg = {
		.mode = GPIO_MODE_OUTPUT,
		.pin_bit_mask = BIT64(LED_GPIO) | BIT64(AWAKE_GPIO)
	};

	return  gpio_sleep_sel_dis(LED_GPIO) ||
		gpio_config(&io_cfg) ||
		gpio_set_level(LED_GPIO, 1) ||
		gpio_set_level(AWAKE_GPIO, 1);
}

static esp_err_t uart_setup(void)
{
	const uart_config_t uart_cfg = {
		.baud_rate = 115200,
		.data_bits = UART_DATA_8_BITS,
		.parity = UART_PARITY_DISABLE,
		.stop_bits = UART_STOP_BITS_1,
		.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
	        .source_clk = UART_SCLK_RTC
	};
	
	const uart_intr_config_t uart_intr = {
		// Interrupt on UART
		.intr_enable_mask = UART_WAKEUP_INT_RAW_M
	};

	esp_err_t err = uart_driver_install(UART_DEV, 1024, 0, EVENT_QUEUE_SIZE,
			&q_evt, 0);
	if (err) {
		LOGE("failed to install UART driver: %s (%d)", 
				esp_err_to_name(err), err);
		return err;
	}

	err = uart_param_config(UART_DEV, &uart_cfg) ||
		uart_intr_config(UART_DEV, &uart_intr) ||
		uart_enable_rx_intr(UART_DEV);
	if (err) {
		LOGE("UART configuration failed: %s (%d)",
				esp_err_to_name(err), err);
		return err;
	}

	uart_pattern_queue_reset(UART_DEV, EVENT_QUEUE_SIZE);
	uart_flush(UART_DEV);

	return ESP_OK;
}

static void recv_tsk(__attribute__((unused)) void *parameters)
{
	uart_event_t event;

	while(true) {
		if (xQueueReceive(q_evt, &event, pdMS_TO_TICKS(200))) {
			switch (event.type) {
			case UART_WAKEUP:
				led_toggle();
				break;

			default:
				LOGW("unhandled uart event: %d", event.type);
				uart_flush(UART_DEV);
				xQueueReset(q_evt);
				break;
			}
		}
	}
}

void app_main(void)
{
	ESP_ERROR_CHECK(io_setup());
	LOGI("Waiting 5 seconds for USB connection...");
	vTaskDelay(pdMS_TO_TICKS(5000));
	LOGI("Go!");
	ESP_ERROR_CHECK(light_sleep_enable());
	ESP_ERROR_CHECK(uart_setup());

	xTaskCreate(recv_tsk, "hrecv", 4096, NULL, tskIDLE_PRIORITY + 1, NULL);
}
```

3. Connect the ESP32-S3 USB interface to the PC, build and flash the project.
4. Connect the PC to the UART 0 interface of the ESP32-S3. Send data through the UART 0 while monitoring with an oscilloscope GPIO47 and GPIO48.
5. You will see GPIO47 never toggles and GPIO48 activity is not affected by the data sent through UART0, as shown in this example 1 second oscilloscope capture, that is always the same independently of the UART activity:

[image]

### Debug Logs.

```plain

```

nopnop2002, 2025-06-28:
Try official example.

https://github.com/espressif/esp-idf/blob/master/examples/system/light_sleep

doragasu, 2025-06-28:
Official example wakes on UART activity, **BUT** it also fails to generate the `UART_WAKEUP` events, [this switch case](https://github.com/espressif/esp-idf/blob/1bbf04cb4cf54d74c1fe21ed12dbf91eb7fb1019/examples/system/light_sleep/main/uart_wakeup.c#L89) is never reached.

doragasu, 2025-06-28:
Well, got it working with FreeRTOS tickless idle, then compared the differences with my example and it looks like it's just I was missing the `gpio_sleep_set_direction()` and `gpio_sleep_set_pull_mode()` configuration for the RX pin.

@nopnop2002 thanks a lot for help, I'm closing this issue now!

closed 2025-06-28 by doragasu