espressif/esp-idf #16200: WiFi MESH_ROOT crashes when Access Point switched off

michaelsimp, 2025-06-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ver 5.5.0

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Last year I was working on an ESP32 WiFi mesh problem where the ESP32 would crash sometimes when either the parent node disappeared or if MESH_ROOT, if the Access Point was switched off.  refer "WiFi Mesh unstable when parent offline (IDFGH-13875) #14720"

This was on IDF ver 5.3.0 and then 5.3.1 for the fix.
Espressif gave me a patch for this which successfully fixed both issues most of the time (not 100%).

I recently upgraded to IDF ver 5.5.0 and have found the WiFi Mesh to be a lot more stable which is great.

But I seem to have the problem back where my MESH_ROOT crashes whenever the WiFi Access Point is switched off.  It is not intermittent it crashes every time.  There is no problem if the parent of a MESH_NODE (non MESH_ROOT) is switched off.

I expect it not to crash.  MQTT and WiFi comms will obviously fail, but my application should continue to run.  When the Access Point is powered back on, I expect it to recover WiFi Mesh comms and then MQTT.  Again this works fine on my MESH_NODEs if I power off my MESH_ROOT.  All nodes running the same software.

### What is the actual behavior?

The MESH_ROOT crashes (locks up) as soon as the WiFi connection break is detected.

### Steps to reproduce.

1. On startup, a single ESP32 NODE establishes itself as MESH_ROOT connected to my WiFi Access Point
2. Power off the Access Point
3. Bang! - everything locks up - see logs below
...

### Debug Logs.

```plain
NOTE:  The last line in the log is not complete and stops at various places along this ESP_LOGW(). This is the most complete one I have caught.

I (85286) wifi:bcn_timeout,ap_probe_send_start
I (87796) wifi:ap_probe_send over, reset wifi status to disassoc
I (87796) wifi:state: run -> init (0xc800)
I (87796) wifi:pm stop, total sleep time: 0 us / 45167297 us

I (87796) wifi:<ba-del>idx:0, tid:0
I (87796) wifi:new:<7,0>, old:<7,1>, ap:<7,1>, sta:<7,1>, prof:7, snd_ch_cfg:0x0
E (17:27:13.551) transport_base: tcp_read error, errno=Software caused connection abort
I (17:27:13.553) aWifiMesh: <MESH_EVENT_PARENT_DISCONNECTED>reason:200
E (17:27:13.553) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read() error: errno=113
I (87806) mesh: [wifi]disconnected reason:200(beacon timeout), continuous:1/max:12, root, vote(,stopped)<><>
E (17:27:13.558) aMQTT: MQTT_EVENT_ERROR
E (17:27:13.559) aMQTT: MQTT5 return code is 0
E (17:27:13.560) aMQTT: Last error captured as transport's socket errno: 0x71
E (17:27:13.562) aMQTT: Last errno string (Software caused connection abort)
E (17:27:13.563) mqtt_client: mqtt_process_receive: mqtt_message_receive() returned -2
W (17:27:13.566) aMQTT: Broker disconnected
W (17:27:13.567) event: handler alread
```

zhangyanjiaoesp, 2025-06-27:
@michaelsimp 
The log above only shows that the Wi-Fi was disconnected; it does not indicate a crash.

zhangyanjiaoesp, 2025-06-30:
@michaelsimp can you provide a demo for me to test and reproduce this issue?

michaelsimp, 2025-07-01:
Hi Zhang
I noted the sys_evt task HWM was pretty low so I increased CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE from 2304 to 3000.
This seems to have fixed the problem.
If I print the tasks_info() on request from my console when the MESH_ROOT still has a connection to the AP, it reports sys_evt HWM 1036.  I also note that the console task is now status of "X" so X must mean its the active task.
When I power off the AP, the first report from the WiFi Mesh <MESH_EVENT_PARENT_DISCONNECTED> the sys_evt HWM drops to 884 and the 2nd report to 656. This seems a bit odd as it reports "event: handler already registered, overwriting", but is still seems to use another chunk of heap.  The HWM doesnt get any lower than this.
So my problem is fixed which is great thank you.  I am not sure why this problem does not occur with the example project ip_internal_network as it has the same original value for CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2304

One new observation/issue is that now that my MESH_ROOT has lost connection with the WiFi AP, calls to esp_mesh_get_type() keep returning that the node is MESH_ROOT which I don't think is correct.  I would expect it to drop to MESH_IDLE and Layer = -1.  What do you think?

Thanks so much for your help with this.

zhangyanjiaoesp, 2025-07-01:
@michaelsimp 
Glad to hear that your stuck issue has been resolved. 

> I am not sure why this problem does not occur with the example project ip_internal_network as it has the same original value for CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2304

Maybe you can check if there's any event registration logic elsewhere in your code?

> One new observation/issue is that now that my MESH_ROOT has lost connection with the WiFi AP, calls to esp_mesh_get_type() keep returning that the node is MESH_ROOT which I don't think is correct. I would expect it to drop to MESH_IDLE and Layer = -1. What do you think?

When the root node disconnect from the router, the reason code is 200, then the root node will try to reconnect to the router. So the mesh type is still MESH_ROOT

closed 2025-07-01 by michaelsimp