espressif/esp-idf #16205: [esp_http_client]: esp_http_client_perform() resets client-state on end of first chunk, breaking multi-chunk responses

bryghtlabs-richard, 2025-06-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2, v6.0-dev-774-g68b583f30c

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

With `is_async=true`, I expect this code to look until a full HTTP response is processed: `while(ESP_ERR_HTTP_EAGAIN == esp_http_client_perform(client))'

### What is the actual behavior?

In the case of an async client,  `while(ESP_ERR_HTTP_EAGAIN == esp_http_client_perform(client))' loops until the first chunk completes. In the case of multi-chunk responses, this is too early, breaking the stream and client.

Attempts to reuse the client connection fail because the server is still returning remaining chunks, but the HTTP client expects the next data from server to be HTTP headers to the next response, so it fails.

### Steps to reproduce.

1. Connect esp_http_client to a server that returns a line-delimited multi-chunk endless event stream ( Richard TODO: provide demo python server)
2. Run this: `while(ESP_ERR_HTTP_EAGAIN == esp_http_client_perform(client))`

If I can get this to reproduce in python I'll try to provide a stripped down example.

### Debug Logs.

```plain

```

### More Information.

I think the issue is in this code near the bottom of `esp_http_client_perform()`:

```
                http_dispatch_event(client, HTTP_EVENT_ON_FINISH, NULL, 0);
                http_dispatch_event_to_event_loop(HTTP_EVENT_ON_FINISH, &client, sizeof(esp_http_client_handle_t));

                client->response->buffer->raw_len = 0;
                if (!http_should_keep_alive(client->parser)) {
                    ESP_LOGD(TAG, "Close connection");
                    esp_http_client_close(client);
                } else {
                    if (client->state > HTTP_STATE_CONNECTED) {
                        client->state = HTTP_STATE_CONNECTED;
                        client->first_line_prepared = false;
                    }
                }
```

One thing I don't know: for endless chunked response stream, should HTTP_EVENT_ON_FINISH mean "chunk finished" or "response finished"? Currently they are the same since we can only process one chunk 🙃 .

Akshama2003, 2025-06-28:
Hi @bryghtlabs-richard 

🐞 Bug Summary
When using esp_http_client in asynchronous mode (is_async=true), the call to:

```
c

while (ESP_ERR_HTTP_EAGAIN == esp_http_client_perform(client)) { ... }
```
…exits after only the first chunk is received, rather than continuing until the entire HTTP response is processed.

This leads to:

Broken streams for multi-chunk or infinite chunked HTTP responses.

Reuse failure of the connection: client expects new headers, but server is still streaming chunks from previous response.

Misinterpretation of HTTP_EVENT_ON_FINISH in the context of chunked transfers.

✅ Expected Behavior
In asynchronous streaming, the client should:

Continue calling esp_http_client_perform() until all chunks are received (Transfer-Encoding: chunked) or the server closes the connection.

Only emit HTTP_EVENT_ON_FINISH after the entire response (not just a chunk) has been received.

❌ Actual Behavior
Async client exits esp_http_client_perform() loop immediately after one chunk.

Prematurely triggers HTTP_EVENT_ON_FINISH, implying the response is done.

This causes the client to reset its state to HTTP_STATE_CONNECTED, not expecting more data, resulting in protocol desync and reuse errors.

📍 Suspected Cause
You're right — the issue seems to be here:

```
c

http_dispatch_event(client, HTTP_EVENT_ON_FINISH, NULL, 0);
client->response->buffer->raw_len = 0;

if (!http_should_keep_alive(client->parser)) {
    esp_http_client_close(client);
} else {
    client->state = HTTP_STATE_CONNECTED;
}
```
This code immediately resets the state once any chunk ends, even though for chunked streams (like server-sent events or long-polling APIs), the HTTP response is not complete yet.

🔧 Suggested Fix / Design Change
1. Fix esp_http_client_perform() Logic
Ensure esp_http_client_perform() in async mode:

Loops through all chunks.

Only triggers HTTP_EVENT_ON_FINISH when the full HTTP response has been parsed (i.e. last-chunk 0\r\n\r\n seen or connection closed).

2. Introduce Distinction Between Chunk vs Response Completion
Consider introducing:

HTTP_EVENT_ON_CHUNK_FINISH for when a single chunk ends (optional).

Keep HTTP_EVENT_ON_FINISH exclusively for complete response.

This will allow better handling of:

Infinite chunked streams (e.g., Server-Sent Events).

Streaming APIs, where chunk =/= full response.

3. Document Async Behavior for Chunked Responses
Clearly document in ESP-IDF that in async mode:

Chunked responses are not yet fully supported for multi-chunk/streaming cases.

The workaround is to use sync mode, or maintain custom state tracking outside of esp_http_client.

🧪 Workaround for Users
Until fixed, for chunked streaming:

Avoid relying on HTTP_EVENT_ON_FINISH if using async mode.

Consider falling back to synchronous mode, or

Use custom TCP + esp_tls + manual chunk parsing (advanced users).

📝 Additional Notes
A stripped-down Python server reproducer would be very helpful for regression tests.

This behavior is especially problematic for IoT event streaming, server push, or SSE protocols where the connection is not meant to end after a chunk.

bryghtlabs-richard, 2025-07-01:
Also need to consider the zero-byte chunk, which should reset the client-state.

nileshkale123, 2025-08-01:
Hello @bryghtlabs-richard ,

I have tested using both the server you provided, which returns a limited number of chunks, and also with a modified version that streams chunks continuously with a delay between them.

Based on my observations:
- The state toggles to ESP_HTTP_STATE_ON_DATA_START
- All chunks are fetched sequentially as expected
- In case 1 (finite chunks), the HTTP_EVENT_ON_FINISH event is dispatched correctly
- In case 2 (infinite stream), the client continues to fetch chunks without dispatching HTTP_EVENT_ON_FINISH, which is expected

From what I see, this appears to be the correct behavior. Could you please clarify where exactly you're seeing an issue or unexpected behavior? I'd be happy to take a closer look.

bryghtlabs-richard, 2025-08-04:
Thanks @nileshkale123 , I misunderstood the timeout setting. I thought it meant, "wait up to X time for each esp_http_client_perform", but it seems to be, "fail the connection if no data within X time".

closed 2025-08-04 by bryghtlabs-richard

bryghtlabs-richard, 2025-08-13:
The mystery is solved. I had tried removing the timeout, but it defaults to 5 seconds. The server we connect to sends a blank line every 6 seconds, so it was still just a regular timeout. 🙃  Thanks again Nilesh.