espressif/esp-idf #16206: esp_http_client_perform() hangs after multiple HTTPS POST requests even with CONFIG_MBEDTLS_HARDWARE_MPI disabled

x-radio, 2025-06-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone,

I'm not sure if I should open a new issue. My problem seems similar to #8710 #9055 #9256 #9371

As far as I understood, back in 2022 the solution was to disable the CONFIG_MBEDTLS_HARDWARE_MPI option. I did that, but the problem remains.

My setup:
esp32s3 16/8, IDF 5.4.1, Windows 11, VSCode + IDF extension

What I'm doing:
I'm sending HTTPS POST requests using esp_http_client with esp_crt_bundle_attach.
This is a kind of stress test — requests are sent one by one with a 1-second delay.
After 10–300 requests (it varies each time), esp_http_client_perform() stops returning and hangs.
Naturally, the whole task gets stuck and the WDT triggers.

Later, I noticed in long logs that perform() actually does return every ~840–850 seconds.
I suspect this is related to the Maximum number of retransmissions of data segments and Maximum number of retransmissions of SYN segments.
I reduced those values to 4 and 1 (though oddly it only helped once!) and the 850 sec hang time dropped to ~18 sec.
Currently, those values are 4 and 4, but the delay is still 850 sec (maybe I missed something?).

What I’ve tried:

Both asynchronous mode (calling perform() from a state machine) and synchronous mode

Changed .timeout_ms from 2000 to 5000

Deleting and creating a new client for each request

Keeping the client between requests but calling esp_http_client_close()

Keeping the connection alive between requests

Disabled CONFIG_MBEDTLS_HARDWARE_MPI

Tried esp_wifi_disconnect() → esp_wifi_stop() → esp_wifi_deinit() → esp_netif_deinit() … some delay … then re-init everything — the task still doesn’t resume

Tried deleting the task and creating it again — the new task still hangs on perform()

What can I do?
Is there a way to fully reset the stack without doing a full system reboot?
If needed, I can provide code and configuration, no problem.
But maybe I missed something fundamental — so I’m asking for help.

Akshama2003, 2025-06-28:
Hi @x-radio 

This indeed seems similar to issues like #8710, #9055, and others related to TLS session hangs or stuck  **esp_http_client_perform()** behavior under stress.

✅ Confirmed Observations:
Your system: ESP32-S3, IDF 5.4.1, Windows 11, VSCode, https POST stress test

You disabled CONFIG_MBEDTLS_HARDWARE_MPI, as advised in older tickets.

You've tried both sync/async, resetting the client, WiFi deinit, task recreation, etc.

Logs show that esp_http_client_perform() unfreezes after ~850 seconds → may indicate TCP stack retry timeouts or TLS blocking on socket read.

✅ Root Cause (Likely):
esp_http_client_perform() uses blocking I/O internally, and when the socket gets stuck due to retransmissions or handshake timeouts (especially in poor network environments), it can hang despite timeout settings, especially during TLS operations.

Reducing TCP retransmissions (CONFIG_LWIP_TCP_MAXRTX and CONFIG_LWIP_TCP_SYNMAXRTX) might not fully reset TLS internal state or pending reads. Also, TLS recv timeouts in mbedTLS can override socket-level timeouts.

✅ Suggestions & Solution:
🔧 1. Manually Set Socket Timeout for TLS Layer
Use esp_http_client_set_transport_sockopt() (newer versions of ESP-IDF support it) or customize the transport_ssl layer to force TLS socket timeouts.

You can also patch or override mbedtls_ssl_set_bio() with a custom read callback that applies timeouts explicitly.

🧹 2. Force Close + Free All Sockets Between Requests
Make sure to:

```
c

esp_http_client_close(client);
esp_http_client_cleanup(client); // Frees resources
Optionally add a delay + check all netconns are cleaned.
```

🔁 3. Reinitialize TCP/IP Stack
If the issue still persists after socket cleanup, try:

```
c

esp_wifi_disconnect();
esp_wifi_stop();
esp_netif_deinit();
vTaskDelay(500 / portTICK_PERIOD_MS); // Let the deinit finish
esp_netif_init();
esp_event_loop_create_default();
esp_wifi_init();
```
This works only if you also deinit LWIP netifs — partial deinit may leave dead sockets.

🚀 4. Switch to Asynchronous esp_tls Layer
If you're comfortable digging deeper:

Build a custom esp_http_client using esp_tls_init() and sockets.

Then use non-blocking socket mode with select() timeout.

This gives full control and avoids TLS blocking forever.

🛠️ Temporary Workaround:
If reliability is critical:

Add a watchdog timer for your HTTP task.

If it detects a hang (no POST in X seconds), trigger esp_restart().
Not ideal, but better than a stuck production system.

🧪 Additional Debugging Tip:
Enable CONFIG_MBEDTLS_DEBUG and set esp_log_level_set("*", ESP_LOG_DEBUG) to get TLS-level logs. You may find which exact call is blocking (e.g. recv, handshake, etc.).

✅ Final Notes
This issue is not isolated, and is often rooted in mbedTLS blocking behavior + LWIP retries.

We're flagging this internally for improvement in documentation and robust handling of broken peer links in esp_http_client.

x-radio, 2025-07-01:
Hello @Akshama2003. Thanks for your answer. But it looks like it have been made with some AI bot. I couldn't find any notes about esp_http_client_set_transport_sockopt(). Im sorry if i mistake...

nileshkale123, 2025-07-16:
Hello @bryghtlabs-richard .

Yes, please allow us some time. We will be back on this very soon.

x-radio, 2025-07-16:
Hi! It turns out it was my mistake. Due to task priority inheritance during concurrent resource access, the priority of the main task was changing from 1 to 5. The main task had a `while(1) { ...; taskYIELD(); }` loop, which was blocking other tasks, including the one with the HTTP client.

Switching to `vTaskDelay(1)` resolved the issue.