espressif/esp-idf #16219: Decoding core dump fails on MacOS Apple Silicon

dzungpv, 2025-06-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDF version:** 
https://github.com/espressif/esp-idf/commit/858a988d6eb90f54661abe282c523879c6ad0116

**Board**
ESP32DEVKIT 

**Operating System used:**
MacOS 15.5 (24F74) M2 Max

**How did you build your project?**
Command line with idf.py

**Sample project:** 
https://github.com/MarkEvens/Download-core-dump

I run this command to decode the coredump file after run the sample project, make it crash and get the dump file, I have tried both file get with http download and esptool read-flash, and it the same

```
espcoredump.py info_corefile --core core.bin --core-format raw build/core_dump.elf
```

Error logs:
```
===============================================================
==================== ESP32 CORE DUMP START ====================
The ROM ELF file won't load automatically since it was not found for the provided chip type.
Traceback (most recent call last):
  File "/Users/vietdzung/esp/esp-idf-v5.4.x/components/espcoredump/espcoredump.py", line 76, in <module>
    main()
  File "/Users/vietdzung/esp/esp-idf-v5.4.x/components/espcoredump/espcoredump.py", line 61, in main
    temp_core_files = espcoredump.info_corefile()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/coredump.py", line 573, in info_corefile
    self.print_crashed_task_info(marker)
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/coredump.py", line 341, in print_crashed_task_info
    task_name = self.gdb_esp.get_freertos_task_name(marker)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/gdb.py", line 109, in get_freertos_task_name
    val = self.get_tcb_variable(tcb_addr, 'pcTaskName')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/gdb.py", line 91, in get_tcb_variable
    val = self._gdbmi_data_evaluate_expression('(char*)((TCB_t *)0x%x)->%s' % (tcb_addr, variable))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/gdb.py", line 85, in _gdbmi_data_evaluate_expression
    return self._gdbmi_run_cmd_get_one_response("-data-evaluate-expression \"%s\"" % expr,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/gdb.py", line 80, in _gdbmi_run_cmd_get_one_response
    return self._gdbmi_run_cmd_get_responses(cmd, resp_message, resp_type, response_delay_sec=response_delay_sec,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/gdb.py", line 59, in _gdbmi_run_cmd_get_responses
    self.p.write(cmd, read_response=False)
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/pygdbmi/gdbcontroller.py", line 125, in write
    return self.io_manager.write(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vietdzung/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/pygdbmi/IoManager.py", line 276, in write
    self.stdin.write(mi_cmd_to_write_nl.encode())  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe

```

It seem cause by `xtensa-esp32-elf-gdb` on MacOS so  I try get it version with:
```
xtensa-esp32-elf-gdb --version
```
The it show the result with warning:
```
dyld[51112]: symbol '__ZNKSt3__119bad_expected_accessIvE4whatEv' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/IsolatedCoreAudioClient.framework/Versions/A/IsolatedCoreAudioClient is being set to 0xBAD4007.
dyld[51112]: symbol '__ZTINSt3__119bad_expected_accessIvEE' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/IsolatedCoreAudioClient.framework/Versions/A/IsolatedCoreAudioClient is being set to 0xBAD4007.
dyld[51112]: symbol '__ZTINSt3__13pmr15memory_resourceE' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/caulk.framework/Versions/A/caulk is being set to 0xBAD4007.
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
The last I try with help of AI, run GDB directly:
`ensa-esp32-elf-gdb -c core.bin build/core_dump.elf -ex "bt"`

And it failed too with the logs:
```
dyld[60974]: symbol '__ZNKSt3__119bad_expected_accessIvE4whatEv' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/IsolatedCoreAudioClient.framework/Versions/A/IsolatedCoreAudioClient is being set to 0xBAD4007.
dyld[60974]: symbol '__ZTINSt3__119bad_expected_accessIvEE' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/IsolatedCoreAudioClient.framework/Versions/A/IsolatedCoreAudioClient is being set to 0xBAD4007.
dyld[60974]: symbol '__ZTINSt3__13pmr15memory_resourceE' missing from root that overrides /usr/lib/libc++.1.dylib. Use of that symbol in /System/Library/PrivateFrameworks/caulk.framework/Versions/A/caulk is being set to 0xBAD4007.
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin21.1 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/core_dump.elf...

Fatal signal: Segmentation fault: 11
----- Backtrace -----
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>.

zsh: segmentation fault  xtensa-esp32-elf-gdb -c core.bin build/core_dump.elf -ex "bt"
```

dzungpv, 2025-06-29:
It cause by xtensa-esp32-elf-gdb  14.2_20240403. It is not compatible with the MacOs 15.4, I don't know how to update the xtensa-esp32-elf-gdb tool so I do below trick:
1, Download version 16.2_20250324 here https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v16.2_20250324
2, Backup current 14.2_20240403 in ..espressif/tools/xtensa-esp-elf-gdb
3, Rename 16.2_20250324 to 2_20240403

And voila, it work.

Lapshin, 2025-07-01:
@dzungpv , @erhankur , @gerekon , we already have MR in our internal repo to upgrade GDB to the latest for branches 5.1..5.4

But I don't know when it will be merged and released

Lapshin, 2025-07-02:
@dzungpv, required changes are already done. They blocked by our IDF release process, which I personally don't fully understand.

Creating additional PRs wouldn't help here

The next 5.4.x will be released 2025/10/08 according to the roadmap https://github.com/espressif/esp-idf/blob/master/ROADMAP.md#esp-idf-bugfix-releases

Since you already found a solution, I guess this issue not blocking you

NevynUK, 2025-08-16:
+1 on this issue, I have this problem on a Mac M1 and an M2 machine.

I am using IDF 5.5.

NevynUK, 2025-08-16:
I think I have resolved this on one of my machines.

At the start I had:
* python 3.11 and 3.13 installed using brew
* python 3.12 installed using conda
* python 3.12 installed in a pyenv

It looks like the conda installation was taking preference.

My solution:
* Removed conda
* use pyenv to ensure that python 3.13.5 was installed an active
* Close down the terminals and re-open
* Remove the tools from the .espressif folder
* Re-run the install script for the tools
* Run the export script

I can now attach to a board and debug once more.

shengt25, 2025-10-20:
@NevynUK Thanks, I had the similar problem and solved by disablling conda before installing the ESP-IDF. Though not for core dump decoding, mine was for regular GDB debugging using `xtensa-esp32s3-elf-gdb`

My symptoms is, when running `xtensa-esp32s3-elf-gdb ./blink.elf`, it ends up immediate `Segmentation fault`. No dyld warnings for me just direct crash when loading any `elf` file. 

I didn't remove conda but just disabled it, the full steps are:
1. Delete existing `.espressif` folder
2. Disable conda by commenting out the conda init block in `~/.zshrc`, or if you are using bash then `.bashrc`:
```bash
# >>> conda initialize >>>
# !! Contents within this block are managed by 'conda init' !!
# ... (comment out all lines)
# <<< conda initialize <
```
3. Open a new terminal
4. Run `./install.sh` to reinstall ESP-IDF
5. Verify that `xtensa-esp32s3-elf-gdb` works properly when loading `elf` files
6. Uncomment conda init in `~/.zshrc` and restart terminal

Disable conda temporarily before ESP-IDF installation, then re-enable it afterwards. It can remain active during normal GDB usage. It seems that conda interfere with `install.sh` script somehow, but the exact reason is unclear yet.

P.S. Works with macOS 26.