espressif/esp-idf #16224: Can't build esp-arduino custom library

cniedzi, 2025-06-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm trying to compile custom arduino-esp32 library on Ubuntu. 
I have an issue during executing the command `ubuntu@ubuntu:~/esp32-arduino-lib-builder$ ./build.sh -b menuconfig -t esp32s3`.
I'm getting the famous "rich module" error:
```
Python 3.12.3
"python3" has been detected
error: Unable to import the rich module: No module named 'rich'. Please execute the install script.

ERROR: Activation script failed
Command '['/home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python', '/home/ubuntu/esp32-arduino-lib-builder/esp-idf/tools/export_utils/activate_venv.py', '--export', '--shell', 'bash']' returned non-zero exit status 1.
To view detailed debug information, set ESP_IDF_EXPORT_DEBUG=1 and run the export script again.
* Building for esp32s3
idf.py -DIDF_TARGET="esp32s3" -DSDKCONFIG_DEFAULTS="configs/defconfig.common;configs/defconfig.esp32s3;configs/defconfig.debug_default;configs/defconfig.esp_sr" menuconfig
./build.sh: line 172: idf.py: command not found
```

When I'm trying to upgrade python, it is saying that rich is already installed:
```
ubuntu@ubuntu:~/esp32-arduino-lib-builder$ pip3 install rich
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: rich in /home/ubuntu/.local/lib/python3.12/site-packages (14.0.0)
Requirement already satisfied: markdown-it-py>=2.2.0 in /usr/lib/python3/dist-packages (from rich) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in /usr/lib/python3/dist-packages (from rich) (2.17.2)
Requirement already satisfied: mdurl~=0.1 in /usr/lib/python3/dist-packages (from markdown-it-py>=2.2.0->rich) (0.1.2)
```

What should I do to be able to build and run menuconfig?

peterdragun, 2025-07-01:
Hi @cniedzi, I'm sorry for the issues. This looks to be the same issue as reported here: https://github.com/espressif/esp-idf-kconfig/pull/16

For now, as a workaround, you are installing `rich` to a different environment. Try running `/home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python -m pip install rich` - this should install it to the correct environment and fix this issue.

This can also be resolved by re-running the esp-idf installation script.

peterdragun, 2025-07-01:
> Hi, thank you for your reply. I'm affraid the suggested workaround doesn't work for me :)

Please run the command as I send it - don't change the directory, and then run Python because this will still use the system Python instead of the one from the environment. Just copy and paste the whole command.

`/home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python -m pip install rich`

cniedzi, 2025-07-01:
> /home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python -m pip install rich

ubuntu@ubuntu:~/esp32-arduino-lib-builder$ /home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python -m pip install rich
/home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python: No module named pip
ubuntu@ubuntu:~/esp32-arduino-lib-builder$

dobairoland, 2025-07-02:
@cniedzi I suspect that the issue is with your environment, so next versions won't fix it. My suggestions will help you to clean it up.

cniedzi, 2025-07-02:
> [@cniedzi](https://github.com/cniedzi) I suspect that the issue is with your environment, so next versions won't fix it. My suggestions will help you to clean it up.

I precisely followed this guide: https://github.com/domenk/esp32-arduino-lib-builder-windows.

cniedzi, 2025-07-02:
I will try to collect and submit the diagnostic information about my environment using the tool you provided.

nopnop2002, 2025-07-03:
>I precisely followed this guide: https://github.com/domenk/esp32-arduino-lib-builder-windows.

This document contains the following statement:
```
ESP32 build script does not seem to support virtual environments at this time
```

But you are using a virtual environment.
````
Command '['/home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python', '/home/ubuntu/esp32-arduino-lib-builder/esp-idf/tools/export_utils/activate_venv.py', '--export', '--shell', 'bash']' returned non-zero exit status 1.
To view detailed debug information, set ESP_IDF_EXPORT_DEBUG=1 and run the export script again.
* Building for esp32s3
```

Follow the documentation to remove the virtual environment.

cniedzi, 2025-07-03:
> You can check if you are in the virtual environment with the following command:
> 
> ```
> $ which python3
> /home/nop/.espressif/python_env/idf5.4_py3.8_env/bin/python3
> ```
> 
> If you get this result, you are in the virtual environment. ESP32 Arduino Lib Builder can't work.
> 
> ```
> $ which python3
> /usr/bin/python3
> ```
> 
> If you get this result, you are in the real environment. ESP32 Arduino Lib Builder can work.

ubuntu@ubuntu:~/esp32-arduino-lib-builder$ which python3
/usr/bin/python3

cniedzi, 2025-07-03:
> I think this is not an ESP-IDF issue, but an esp32-arduino-lib-builder issue.
> 
> Both are EspressIF products, but probably have different maintainers.
> 
> This is more appropriate.
> 
> https://github.com/espressif/esp32-arduino-lib-builder/issues

Thx. I'm closing this issue and will rise there.

closed 2025-07-03 by cniedzi