espressif/esp-idf #16226: Infinite loop with event <MESH_EVENT_PARENT_DISCONNECTED>

hangyugit, 2025-07-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1 - Commit 4c2820d

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROOM32U

### Power Supply used.

USB

### What is the expected behavior?

If you send the MAC addresses of child nodes to the root node using MQTT, the addresses are added to the beacon signal and sent, and other nodes see the order of the addresses and connect to the parent node. Based on the ip internal network example

### What is the actual behavior?

When i trying to connect to the current root node, the following error occurs in an infinite loop:
`I (214403) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, non-root, vote(,)<><>
I (214403) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201`

But sometimes things go well.

[main.zip](https://github.com/user-attachments/files/20991082/main.zip)

[main.zip](https://github.com/user-attachments/files/20991083/main.zip)

https://github.com/espressif/esp-idf/issues/9955
I found a similar issue in the link above, but I think it's a different case because I have self_organized disabled

### Steps to reproduce.

1. Step Turn on the root node
2. Step Send mac address to root node via mqttx app
3. Step Turn on child node

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6276
load:0x40078000,len:15748
[... 208 lines cut ...]
[0;32mI (12303) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, non-root, vote(,)<><>[0m
[0;32mI (12303) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (12433) mesh: [wifi]disconnected reason:201(), continuous:5/max:12, non-root, vote(,)<><>[0m
[0;32mI (12433) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (12553) mesh: [wifi]disconnected reason:201(), continuous:6/max:12, non-root, vote(,)<><>[0m
[0;32mI (12553) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (12683) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, non-root, vote(,)<><>[0m
[0;32mI (12683) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (12803) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, non-root, vote(,)<><>[0m
[0;32mI (12803) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (12933) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, non-root, vote(,)<><>[0m
[0;32mI (12933) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (13053) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, non-root, vote(,)<><>[0m
[0;32mI (13053) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (13183) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, non-root, vote(,)<><>[0m
[0;32mI (13183) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (13303) mesh: [wifi]disconnected reason:201(), continuous:5/max:12, non-root, vote(,)<><>[0m
[0;32mI (13303) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (13433) mesh: [wifi]disconnected reason:201(), continuous:6/max:12, non-root, vote(,)<><>[0m
[0;32mI (13433) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (13553) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, non-root, vote(,)<><>[0m
[0;32mI (13553) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
[0;32mI (13683) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, non-root, vote(,)<><>[0m
[0;32mI (13683) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201[0m
```

zhangyanjiaoesp, 2025-07-02:
@hangyugit 
we need to check whether the root node is working correctly when the child node connect fail. So you need to **capture both the root node's and the child node's wireless packets at the same time**, like the following:

[image]

hangyugit, 2025-07-05:
@zhangyanjiaoesp
Sorry for the delay, I just found the problem. The unknown device keeps trying to connect to the root node, making the root node busy and causing the connection to fail. Is there a way to make the ESP32 accept connection requests only from the ESP32?

[image]

The unknown device here is HONGAIPRECIS.

hangyugit, 2025-07-05:
@zhangyanjiaoesp 
If there is a way to resolve the phenomenon where connection attempts from other external devices interfere with the connection 
between ESP-MESH nodes, please let me know.

zhangyanjiaoesp, 2025-07-07:
@hangyugit 
you can call `esp_mesh_set_config(mesh_cfg_t *config)` to set the `config->mesh_ap.nonmesh_max_connection` to `0`.

zhangyanjiaoesp, 2025-08-05:
@hangyugit 
`the root node needs to be connected to an external router` is the STA's behavior, but the `config->mesh_ap.nonmesh_max_connection` is the config for softAP, it means the SoftAP doesn't allow external no mensh STA to connect, so set `config->mesh_ap.nonmesh_max_connection` to 0 won't impact the STA connecting to router.

closed 2025-08-05 by hangyugit