espressif/esp-idf #16230: ESP_LCD: Implement display brightness

sandynomad, 2025-07-01:
### Is your feature request related to a problem?

I'm using an AMOLED display that supports display brightness.

I know the command (0x51) works as when the initialisation data is modified the brightness changes on init.

The command to alter brightness is defined in "esp_lcp_panel_commands.h" (LCD_CMD_WRDISBV), however through the official IDF API for an LCD screen I cannot see how this command can be sent.

atanisoft, 2025-07-01:
https://github.com/espressif/esp-idf/tree/master/components/esp_lcd will have any relevant code that would live in that library built on your system from code in that tree (actual path will depend on which branch you are using).

However, you would likely send the commands using public APIs such as:
```
esp_lcd_panel_io_tx_param(panelIO_, panel_init_cmds[cmd].cmd,
            panel_init_cmds[cmd].data, panel_init_cmds[cmd].data_count & 0x1F);
```
Which I use internally for passing a number of panel init commands that are not defined for ST7789 (any version).

sandynomad, 2025-07-02:
Hi @atanisoft - I did implement a kludge, something similar to you I suspect.

However I do believe this feature request for an extension to the official ESP LCD API is still warranted.

Given the command is already defined in the header file, it would be beneficial to other developers if it could be used naturally.