espressif/esp-idf #16231: I2C on esp32-p4 fails to correctly read values leaving some buffer untouched on SHT3x sensor

stkw0, 2025-07-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.4.2

### Espressif SoC revision.

ESP32-P4NRW32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

 ESP32-P4-DevKit	 	

### Power Supply used.

USB

### What is the expected behavior?

All data from the SHT3x is read correctly (temperature, humidity and crc).

### What is the actual behavior?

Only the first three bytes are read (the temperature and the CRC) but are placed in the last buffers. There is no error reported nor any timeout but the three last bytes are missing (details below).

### Steps to reproduce.

Execute an operations like the next:

```cpp
        uint8_t addressw = (0x44 << 1) | 1;
        uint8_t temperature[2] = {};
        uint8_t humidity[2] = {};
        uint8_t crc[2] = {};

        i2c_operation_job_t i2c_ops2[] = {
            {.command = I2C_MASTER_CMD_START},
            {.command = I2C_MASTER_CMD_WRITE,
             .write = {.ack_check = true, .data = (uint8_t*)&addressw, .total_bytes = 1}},
            {.command = I2C_MASTER_CMD_READ,
             .read = {.ack_value = I2C_ACK_VAL, .data = (uint8_t*)temperature, .total_bytes = 2}}, // This must be nack.
            {.command = I2C_MASTER_CMD_READ,
             .read = {.ack_value = I2C_ACK_VAL, .data = (uint8_t*)&crc[0], .total_bytes = 1}},
            {.command = I2C_MASTER_CMD_READ,
             .read = {.ack_value = I2C_ACK_VAL, .data = (uint8_t*)humidity, .total_bytes = 2}}, // This must be nack.
            {.command = I2C_MASTER_CMD_READ,
             .read = {.ack_value = I2C_NACK_VAL, .data = (uint8_t*)&crc[1], .total_bytes = 1}},
            {.command = I2C_MASTER_CMD_STOP},
        };

        ESP_ERROR_CHECK(i2c_master_execute_defined_operations(
            dev_handle, i2c_ops2, sizeof(i2c_ops2) / sizeof(i2c_operation_job_t), -1));
```

Temperature and crc[0] will contain 0x00. But if the value is initialized to any other value, like 0xff, after executing the operation the values will be untouched.
Humidity and crc[1] will contain the first three bytes. The calcualted and read CRC matches, so the humidity seems to be working properly.

### Debug Logs.

```plain

```

### Diagnostic report archive.

The  ESP32-P4-DevKit	appears in the system not as a ttyUSBN but as a ttyACMN, which don't seems to be compatible with esp-dif-diag.

### More Information.

Example output of the bytes:
 temperature[0] = 0
 temperature[1] = 0
 crc[0] = 0
 humidity[0] = 0x6d
 humidity[1] = 0xdc
 crc[1] = 0xed

When reading the actual transmited data with a logic analyzer we can see that the 6 bytes are present, but the esp32p4 only read half of it, discarding the last half:

[image]

mythbuster5, 2025-07-02:
Thanks for report, Will debug for this issue.

stkw0, 2025-07-03:
Thank you very much for the fast reply.

I just tested the provided patch and seems to be working!
I made a for loop doing consecutive reads and the CRC of both, temperature and humidity, matches the expected CRC plus the values are inside the expected range.

closed 2025-07-24 by espressif-bot

mythbuster5, 2025-09-16:
Or give more details for `some cases`? Or is it easy to reproduce?

stkw0, 2025-09-16:
> [@stkw0](https://github.com/stkw0) I saw it reports a software timeout. So what will happen if you give a longer timeout value?

I put a timeout of 5 seconds. Removing the patch works. 

At first I thought it was a problem with the MCP23017 component and opened an issue here https://github.com/espressif/esp-iot-solution/issues/568#issuecomment-3252924319 where I attached some photos of what I saw with the oscilloscope. I think it should be easy to reproduce by applying this patch and trying to run the test_app of mcp23017 component.

stkw0, 2025-09-16:
Yes, the P4