espressif/esp-idf #16233: SDMMC + ESP-Hosted via SDIO do not work together since c2b8ea0 (IDFGH-15592) (EHM-68)

johnboiles, 2025-07-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-4513-g1bbf04cb4c

### Espressif SoC revision.

ESP32-P4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4-Function-EV-Board

### Power Supply used.

USB

### What is the expected behavior?

In commit f9765d0316 and earlier it was possible to configure ESP-Hosted and SDMMC both using SDIO.

### What is the actual behavior?

As far as I can tell it is no longer possible to use ESP-Hosted and SDMMC with SDIO at the same time.

After commits 402bf0ce58 ~ c2b8ea07d0 `sd_host_claim_controller` fails with message `no available sd host controller`.

```
E (385) SD_HOST: sd_host_create_sdmmc_controller(82): no available sd host controller
E (395) sdmmc_periph: sdmmc_host_init(84): failed to create new SD controller
E (395) vfs_fat_sdmmc: host init failed (0x105).
```

I believe this is related to the SDMMC controller [hardcoding](https://github.com/espressif/esp-idf/commit/c2b8ea07d0e8ace82fd0f05207d61072e81a4587#r160272293) `SDMMC_LL_HOST_CTLR_NUMS` as `1U`. 

It also may be related to this [comment in sd_host_sdmmc.c](https://github.com/espressif/esp-idf/blob/1bbf04cb4cf54d74c1fe21ed12dbf91eb7fb1019/components/esp_driver_sdmmc/src/sd_host_sdmmc.c#L110).

```
    //use `ETS_SDIO_HOST_INTR_SOURCE` for now, replace with `sdmmc_periph.c` then. IDF-12585
    ESP_GOTO_ON_ERROR(esp_intr_alloc(ETS_SDIO_HOST_INTR_SOURCE, 0, &sd_host_isr, ctlr, &ctlr->intr_handle), err, TAG, "failed to install interrupt service");
```

### Steps to reproduce.

I'm attaching a minimal repro project for the ESP32-P4-Function-EV-Board.

* Checkout esp-idf f9765d0316
* `idf.py build flash monitor`
* Observe no errors
* Checkout esp-idf c2b8ea07d0 or later (esp-idf master as of this issue's filing 1bbf04cb4c)
* `idf.py build flash monitor`
* Observe sdmmc failure

[... 155 lines cut ...]
--- 0x4000bc84: transport_drv_sta_tx at /Users/johnboiles/Developer/repos/sindarin-inc/repro_sdmmc_esp_hosted_crash/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:216
--- 0x4001aec6: esp_wifi_remote_channel_rx at /Users/johnboiles/Developer/repos/sindarin-inc/repro_sdmmc_esp_hosted_crash/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

I (347) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000bbcc] Rx[0x4001aec6]
--- 0x4000bbcc: transport_drv_ap_tx at /Users/johnboiles/Developer/repos/sindarin-inc/repro_sdmmc_esp_hosted_crash/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:246
--- 0x4001aec6: esp_wifi_remote_channel_rx at /Users/johnboiles/Developer/repos/sindarin-inc/repro_sdmmc_esp_hosted_crash/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_n
Repro case for sdmmc + esp_hosted
et.c:19

I (355) main_task: Started on CPU0
I (365) main_task: Calling app_main()
I (365) repro_sdmmc_esp_hosted_crash: Initializing SD card
I (365) repro_sdmmc_esp_hosted_crash: Using SDMMC peripheral
W (375) ldo: The voltage value 0 is out of the recommended range [500, 2700]
I (375) repro_sdmmc_esp_hosted_crash: Mounting filesystem
E (385) SD_HOST: sd_host_create_sdmmc_controller(82): no available sd host controller
E (395) sdmmc_periph: sdmmc_host_init(84): failed to create new SD controller
E (395) vfs_fat_sdmmc: host init failed (0x105).
E (405) repro_sdmmc_esp_hosted_crash: Failed to initialize the card (ESP_ERR_NOT_FOUND). Make sure SD card lines have pull-up resistors in place.
I (415) main_task: Returned from app_main()
```

### Diagnostic report archive.

[idf-diag-303e2f1c-bc43-486d-a074-c6ed2bed0c67.zip](https://github.com/user-attachments/files/20972354/idf-diag-303e2f1c-bc43-486d-a074-c6ed2bed0c67.zip)

servetcoskun, 2025-08-31:
Is there a ‘fix’ we can use until this is resolved? I am on 5.4 trying to host sd card files on my network using the sdmmc example and esp-hosted-mcu library with the esp32p4 function evaluation board.

mantriyogesh, 2025-09-02:
@SohKamYung-Espressif wrt https://github.com/espressif/esp-idf/issues/16233#issuecomment-3025818961, does this issue gets reproduced on 5.4 also?

@servetcoskun would it be possible to state the exact `esp-idf` commit that you use?

mantriyogesh, 2025-09-03:
@SohKamYung-Espressif PTAL

mantriyogesh, 2025-09-03:
@JohnnyZig Can you please state your idf commit and version in use?

SohKamYung-Espressif, 2025-09-03:
This is what the repo test project does:

```
void app_main(void)
{
[...]
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.slot = SDMMC_HOST_SLOT_0;
[...]
```

Which is basically the same as what @JohnnyZig does in https://github.com/espressif/esp-idf/issues/16233#issuecomment-3245692592 but without modifying ESP-IDF source code.

The repo test project works for release/v5.5 and release/v5.4, but not for master.

BTW, on master, the file `components/esp_driver_sdmmc/include/driver/sdmmc_default_configs.h` has been moved to `components/esp_driver_sdmmc/legacy/include/driver/sdmmc_default_configs.h`

https://github.com/espressif/esp-idf/blob/2044fba6e71422446986f9ae0909b1ab67e57815/components/esp_driver_sdmmc/legacy/include/driver/sdmmc_default_configs.h

ghpille, 2025-10-15:
I was wondering, with ESP-IDF v6.0-dev-2034-g48a40b0001-dirty, is it possible to use SDMMC and ESP-Hosted in the same project?
I've been hit with 
```
E (1723) SD_HOST: sd_host_create_sdmmc_controller(82): no available sd host controller
E (1733) sdmmc_periph: sdmmc_host_init(84): failed to create new SD controller
```
since I added WiFi to my project.  Any workaround?  Alternatives for ESP-Hosted? (I'm on ESP32-P4-NANO, so with the ESP32-C6).

johnboiles, 2025-10-16:
AFAIK this hasn't been fixed. It may be possible to revert the SDMMC changes from earlier in the year but I haven't attempted that.

ghpille, 2025-10-16:
So the problem is that sdmmc_host_init is called when initializing the card reader as well as when starting ESP-Hosted.  The second one finds that the controller was already initialized.  @johnboiles  Is that the end of the story?  Is that "controller" (I don't even know if it is HW or SW) only useable by one task?  My intention was to handle the network in a separate task   - does that matter?
In my configuration, when I assigned Hosted to slot 0, menuconfig assigned the same pin numbers as those in use by the card reader.  I can understand that is a no go.  I tried to change them, but I couldn't, I had to choose a number between 44 and 44, eg.

SohKamYung-Espressif, 2025-10-16:
Hi, @ghpille 

> In my configuration, when I assigned Hosted to slot 0, menuconfig assigned the same pin numbers as those in use by the card reader. I can understand that is a no go. I tried to change them, but I couldn't, I had to choose a number between 44 and 44, eg

This is the expected behaviour of configuring ESP-Hosted to use SDMMC Slot 0. On the ESP32-P4, the GPIO pins for SDMMC Slot 0 are fixed and routed to the SD Card slot on the P4 dev board.

The GPIO pins for SDMMC Slot 1, however, can be changed.

In the menuconfig for ESP-Hosted, if you go to `Component config` -> `ESP-Hosted` -> `Configure GPIOs for Development Board` there are two options:

- `ESP32-P4-Function-EV-Board with on-board C6 or C5/C2 on EV board`, fixes the GPIOs so that the P4 can communicate using SDIO with the on-board C6 on the P4 + C6 dev board.
- `No development board` allows you to choose any GPIO for the SDMMC Slot 1 signals.

I hope that clears up the GPIO pin usage for SDMMC Slot 0 and Slot 1.

mantriyogesh, 2025-11-11:
Hello @johnboiles ,

> I suspect the P4 is a lower priority for their team than their other chips.

Sorry for the delay, the issue involved multiple teams involved to finally get the solution, so got delayed a bit. We have all chips as priority. 

@johnboiles , @jackw163 , no need to go to older idf just to do this.
We got this working fine for IDF master as well as older IDF versions.

We are also adding up example in esp-hosted-mcu repo itself, so that it would be easy for user to use hosted + sd card access for chips like ESP32-P4 and ESP32-S3.