espressif/esp-idf #16234: BluFi (Nimble): Add possibillity to send Manufacturer Specific Data in advertisement

heikothole, 2025-07-02:
### Is your feature request related to a problem?

I want to add "Manufacturer Specific Data" to the bluetooth advertisment.
But in esp_blufi_adv_start() (esp-idf-v5.4\components\bt\common\btc\profile\esp\blufi\nimble_host\esp_blufi.c) the fields struct is filled by the function. It is not possible to add "Manufacturer Specific Data".

### Describe the solution you'd like.

A parameter in esp_blufi_adv_start() which could contain "Manufacturer Specific Data"
e.g. 
esp_blufi_adv_start( uint8_t *data, size_t length )

### Describe alternatives you've considered.

I tried to write my own function which first stopped the advertisments (ble_gap_adv_stop()), then filled own fields (ble_gap_adv_set_fields()) and restarted advertisments (ble_gap_adv_start()).
But then, BluFi does not work, because esp_blufi_gap_event() isn't called. Or I cannot pass the pointer to the function because it is static.

rahult-github, 2025-07-02:
Hi @heikothole  ,

As seen [here](https://github.com/espressif/esp-nimble/blob/0e3d2d39bb7dad01a70b6e810014c43cea235d24/nimble/host/include/host/ble_hs_adv.h#L154), mfg_data is part of ble_hs_adv_fields.    So, you need to add manufacturing data information in that field to be shown up on remote side as part of adv data.  However, do confirm that total adv data is not beyond 31 bytes, as mandated by BLE spec. 

Below is just  a sample code for reference. 

`
 char manu_data[4] = {'M','A','N', 'U'}; 
 fields.mfg_data = (uint8_t *)manu_data;
 fields.mfg_data_len = strlen(manu_data) ;

`

heikothole, 2025-07-02:
Hi @rahult-github ,

thank you for your feedback.
I know that mfg_data is a part of ble_hs_adv_fields.
But esp_blufi.c is part of the ESP-IDF. And I don't want to modify code inside the ESP-IDF. Then I would have to make the modification again with every update.

heikothole, 2025-07-11:
Hi  @rahult-github 

> yes, in the patch i provided, this change is present.

Oh yes, I had overlooked that.

> Yes, will perform some more internal testing and plan to merge this .

Thanks a lot.

heikothole, 2025-07-29:
Hi @rahult-github ,

do you know when the change will be included in an official release?

KeysPAN0114, 2025-08-21:
Hello, I want to implement the same function using esp32c2, I tried to port the code to my project, referencing the header definition of the ported code, I use #include "btc_blufi_prf.h" in my project, and then the compile error occurs, saying XXX/main/networking/wifi/blufi_init.c:18:
G:/XXX/ESPIDF/IDF/master/esp-idf/components/bt/common/btc/profile/esp/blufi/include/blufi_int.h:10:10: fatal error: btc/btc_task.h: No such file or directory
10 | #include "btc/btc_task.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Missing "btc/btc_task.h" file name found in the following component(s):