espressif/esp-idf #16239: [esp_http_client]: underlying transport errors are not captured

bryghtlabs-richard, 2025-07-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-774-g68b583f30c

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-s3-wroom-1 n8r8

### Power Supply used.

USB

### What is the expected behavior?

Transport errors resulting in HTTP_EVENT_DISCONNECTED should bubble up through the protocol stack 

### What is the actual behavior?

Instead, `esp_tls_get_and_clear_last_error()` and errno sometimes both return zero, so the disconnect is then nearly undebuggable.

### Steps to reproduce.

1. Setup a Linux-based router
2. Setup iptables to TCP-RST periodically. I used `iptables -F; iptables -A FORWARD -p tcp --dport 443 -m statis
tic --mode nth --every 11 --packet 1 -j REJECT --reject-with tcp-reset`
3. Monitor console output of examples/protocols/esp_http_client - may need to run it a few times to drop TCP mid-request

Often mid-request TCP connection loss looks like this:

```
E (48961) transport_base: poll_write select error 104, errno = Connection reset by peer, fd = 55
W (48961) transport_base: Poll timeout or error, errno=Success, fd=55, timeout_ms=5000
E (48971) HTTP_CLIENT: Error write request
I (48971) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
^ esp_tls_get_and_clear_last_error() returns zero - no way to get 104 error code.
I (48981) HTTP_CLIENT: errno: 0x0
^ I also printed errno - also zero - no way to get 104 error code.
E (48981) HTTP_CLIENT: HTTP request failed: ESP_ERR_HTTP_WRITE_DATA
E (32856) HTTP_CLIENT: HTTP request failed: ESP_ERR_HTTP_WRITE_DATA
```

### Debug Logs.

```plain
I (551) example_connect: Waiting for IP(s)
I (3041) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (3041) wifi:state: init -> auth (0xb0)
I (3051) wifi:state: auth -> assoc (0x0)
[... 167 lines cut ...]
I (48511) HTTP_CLIENT: HTTP partial download =>
I (48651) esp-x509-crt-bundle: Certificate validated
I (48961) HTTP_CLIENT: HTTP Status = 206, content_length = 69
E (48961) transport_base: poll_write select error 104, errno = Connection reset by peer, fd = 55
W (48961) transport_base: Poll timeout or error, errno=Success, fd=55, timeout_ms=5000
E (48971) HTTP_CLIENT: Error write request
I (48971) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (48981) HTTP_CLIENT: errno: 0x0
E (48981) HTTP_CLIENT: HTTP request failed: ESP_ERR_HTTP_WRITE_DATA
I (49061) esp-x509-crt-bundle: Certificate validated
E (49351) esp-tls-mbedtls: read error :-0x0050
E (49351) transport_base: esp_tls_conn_read error, errno=Connection reset by peer
E (49351) HTTP_CLIENT: HTTP request failed: ESP_ERR_HTTP_FETCH_HEADER
I (49351) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (49361) HTTP_CLIENT: errno: 0x68
I (49361) HTTP_CLIENT: Finish http example
```

### Diagnostic report archive.

[idf-diag-a52f1a5b-0208-4f32-9721-659575abe72f.zip](https://github.com/user-attachments/files/21020031/idf-diag-a52f1a5b-0208-4f32-9721-659575abe72f.zip)

### More Information.

Also impacts 5.1.2, 5.1.6, 5.5

nileshkale123, 2025-08-13:
Hello @bryghtlabs-richard ,

I have tested and verified the issue on my end based on your suggested reproduction steps.

**Setup**:
1. Modified iptables to send a TCP reset for every 5th packet using the command: `sudo iptables -A OUTPUT -p tcp --dport 443 -m statistic --mode nth --every 5 --packet 1 -j REJECT --reject-with tcp-reset`
2. Executed `esp_http_client_example.c`’s `http_partial_download()` test in a loop (20 iterations) to cover all possible scenarios.
3. Added `errno` print just after read/write error in `esp_tls_mbedtls.c`

**Observation**:
I consistently observed `errno = 104` (**Connection reset by peer**) whenever the corresponding error was triggered.

**read error logs**
`E (1292067324) esp-tls-mbedtls: read error :-0x0050
E (1292067324) esp-tls-mbedtls: in esp_mbedtls_read errno: 104 (Connection reset by peer)
E (1292067324) transport_base: esp_tls_conn_read error, errno=Connection reset by peer`

**write error logs**
`E (1292067186) esp-tls-mbedtls: write error :-0x0050
E (1292067186) esp-tls-mbedtls: in esp_mbedtls_write errno: 104 (Connection reset by peer)
E (1292067186) transport_base: esp_tls_conn_write error, errno=Connection reset by peer`

**Recommendation**:
I suggest you try this approach on your side to re-verify the behavior. I believe it will address your query.

bryghtlabs-richard, 2025-08-13:
>Observation: I consistently observed errno = 104 (Connection reset by peer) whenever the corresponding error was triggered.

I see that in the logs, but I think I want esp_tls_get_and_clear_last_error() to return the error. I'll attach my stripped down example code.

When I reset all TCP with ` iptables -A FORWARD -p tcp --dport 443 -j REJECT --reject-with tcp-reset`, then attempt to connect:

```
I (7435) HTTP_CLIENT: HTTPS async requests =>
I (7445) main_task: Returned from app_main()
I (7465) wifi:<ba-add>idx:0 (ifx:0, c8:d3:a3:5a:33:c7), tid:0, ssn:1, winSize:64
E (7485) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x004E
I (7485) esp-tls-mbedtls: Certificate verified.
^^^Separately, this is a very strange thing to say when it could not download the certificate.
E (7485) HTTP_CLIENT: Connection failed
I (7485) HTTP_CLIENT: HTTP_EVENT_ERROR
E (7495) HTTP_CLIENT: Error perform http request ESP_ERR_HTTP_CONNECT
I (7495) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (7505) HTTP_CLIENT: esp_tls_get_and_clear_last_error() returned ESP_OK
^^^Why did it not return the underlying cause of the connection error?
I (7505) HTTP_CLIENT: Finish http example
I (11415) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fd0a:3515:e5b1:0000:3e84:27ff:fec7:127c, type: ESP_IP6_ADDR_IS_UNIQUE_LOCAL
```

When I connect to HTTPS server and begin download, then once download begins kill all TCP with ` iptables -A FORWARD -p tcp --dport 443 -j REJECT --reject-with tcp-reset`:

```
I (18765) HTTP_CLIENT: HTTP_EVENT_ON_DATA, len=512
I (18775) HTTP_CLIENT: HTTP_EVENT_ON_DATA, len=512
I (18785) HTTP_CLIENT: HTTP_EVENT_ON_DATA, len=512
E (18795) esp-tls-mbedtls: read error :-0x0050
E (18795) transport_base: esp_tls_conn_read error, errno=Connection reset by peer
I (18795) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
^^^Why did it send finish-event? This download did not finish.
I (18795) HTTP_CLIENT: HTTPS Status = 200, content_length = 3901456384
I (18805) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (18805) HTTP_CLIENT: esp_tls_get_and_clear_last_error() returned ESP_OK
^^^Why did it not return the underlying cause of the connection error?
I (18815) HTTP_CLIENT: Finish http example
```

I also tried adding https://github.com/espressif/esp-idf/pull/16194 , but spurious HTTP_EVENT_ON_FINISH still occurs and `esp_tls_get_and_clear_last_error()` still returns ESP_OK when TCP-RST in the middle of a response download. Should I make another ticket for spurious HTTP_EVENT_ON_FINISH ?

bryghtlabs-richard, 2025-08-13:
Attach example client with long-running HTTPS download [esp_http_client_example.c.txt](https://github.com/user-attachments/files/21757102/esp_http_client_example.c.txt)

nileshkale123, 2025-08-19:
Hello @bryghtlabs-richard ,

I have reverified once again. 

Setup:
1. Target - linux
2. Blocking all packets using `sudo iptables -A OUTPUT -p tcp -m tcp --dport 443 -j REJECT --reject-with tcp-reset
`
3. Executing subtask `http_partial_download()` in `esp_http_client_example.c`

Observation:
1. With this, I am **not** getting any log for 'certificate verified'.

NOTE:

> Also, as we will remove the confusing and unnecessary log of [certificate verification](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/esp-tls/esp_tls_mbedtls.c#L475), because this API got called at [line](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/esp-tls/esp_tls_mbedtls.c#L311) to check if there is an error in certificate verification, as per the comment. Also, I have checked the Wireshark trace, and as per this, I was able to see the client hello and server hello phase completed successfully, which meant the client has a certificate at its end. I request you to confirm if this is the same happening at your end.

Also, I am attaching patch to remove this unnecessary print. Please verify if you still getting these logs even with these changes.

[patch_print_remove.txt](https://github.com/user-attachments/files/21852650/patch_print_remove.txt)

**If the issue is still persists, please tell me your testing setup information in detail.**

nileshkale123, 2025-08-27:
Thanks @bryghtlabs-richard ,

We have raised an MR internally. We will try to get it merged and backported soon.

closed 2025-09-26 by espressif-bot