espressif/esp-idf #16242: msc_example_main.c, callback dont use xQueueSendFromISR()

savenlid, 2025-07-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

As I understand the call-back is executing under ISR status.
If I am wrong I will of course feel ashamed for posting this issue.

closed 2025-07-03 by tore-espressif

savenlid, 2025-07-04:
unrelated question.

I start with usb-host-msc
I want to see if I have a usb-memory stick after boot.

If no memory stick I want to tear down the host completely, and start a NCM device instead.

I do manage to do this "almost".

The MSC task I can tear down.
But the host-task is waiting blocked for:
"        usb_host_lib_handle_events(portMAX_DELAY, &event_flags);  "

and if no usb-memory stick was ever connected no action will happen here.
it "hangs"

Can I provoke some activity from my program to create a "dummy" event so it continues checks a flag and can be torn-down ?

// tear-down
    ESP_ERROR_CHECK( msc_host_uninstall() );
    vTaskDelay(10); // Give clients some time to uninstall
    ESP_ERROR_CHECK(usb_host_uninstall());
    vTaskDelete(NULL);

ps
hmm, maybe I can just change portMAX_DELAY to a second or so and then it will "loop" ?
ds

Thanks
/Nicholas

savenlid, 2025-07-04:
I did a timeout and was able to do my tear-down

but the 
    ESP_ERROR_CHECK(usb_host_uninstall());

crash

on a ESP32-P4

I (30788) USB_MSC: MSC queue received
I (30788) USB_MSC: MSC do
I (30788) USB_MSC: MSC Idle
I (30788) USB_MSC: MSC Idle no work todo
I (30788) USB_MSC: MSC do
I (30788) USB_MSC: MSC disconnect
I (30888) OTA: OTA USB first boot flag is cleared
I (30888) USB_MSC: USB msc task delete
I (31838) USB_MSC: USB-HOST-MSC Uninstall
I (32038) USB_MSC: USB-HOST uninstall
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4001cdd2
--- 0x4001cdd2: USB_host_task at C:/workspace/batteries-chargers/esp32/ch1_p4/main/usb_msc.c:686 (discriminator 1)

file: "./main/usb_msc.c" line 686
func: USB_host_task
expression: usb_host_uninstall()

abort() was called at PC 0x4ff074f5 on core 0
--- 0x4ff074f5: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.4.1-2/components/esp_system/esp_err.c:49

--- Stack dump detected
Core  0 register dump:
MEPC    : 0x4ff00dd2  RA      : 0x4ff07500  SP      : 0x4ff2f3c0  GP      : 0x4ff12680
--- 0x4ff00dd2: panic_abort at C:/Espressif/frameworks/esp-idf-v5.4.1-2/components/esp_system/panic.c:468
--- 0x4ff07500: __ubsan_include at C:/Espressif/frameworks/esp-idf-v5.4.1-2/components/esp_system/ubsan.c:311

TP      : 0x4ff2f4c0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x4ff2f424  A0      : 0x4ff2f3ec  A1      : 0x4ff2f422
A2      : 0x00000000  A3      : 0x4ff2f419  A4      : 0x00000001  A5      : 0x4ff1b000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000

--- Warning: Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'C:\\workspace\\batteries-chargers\\esp32\\ch1_p4\\build\\blink.elf', '-ex', 'target remote | "C:\\Espressif\\python_env\\idf5.4_py3.11_env\\Scripts\\python.exe" -m esp_idf_panic_decoder --target esp32p4 "C:\\Users\\toonsav\\AppData\\Local\\Temp\\tmp8y12_p3m"', '-ex', 'bt']' returned non-zero exit status 1.
--- Warning: b"\r\nwarning: could not convert 'main' from the host encoding (CP1252) to UTF-32.\r\nThis normally should not happen, please file a bug report.\r\nerror starting child process 'C:\\Espressif\\python_env\\idf5.4_py3.11_env\\Scripts\\python.exe -m esp_idf_panic_decoder --target esp32p4 C:\\Users\\toonsav\\AppData\\Local\\Temp\\tmp8y12_p3m': CreateProcess: No such file or directory\r\nNo stack.\r\n"

Stack memory:
4ff2f3c0: 0x400bdecc 0x4001cdd6 0x4ff2f420 0x4ff10aa4 0x00000000 0x00000000 0x00000000 0x4ff15bf4
4ff2f3e0: 0x4ff2f424 0x4ff15c10 0x4ff2f420 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4ff2f400: 0x43502074 0x34783020 0x37306666 0x20356634 0x63206e6f 0x2065726f 0x00000030 0x00000000
4ff2f420: 0x00000030 0x30666634 0x35663437 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff074f8
4ff2f440: 0x400bddd8 0x00000000 0x00000000 0x4001cdd6 0x00000000 0x00000000 0x00000000 0x00000001
4ff2f460: 0x00000005 0x00001000 0x00000000 0x4001ca34 0x00000000 0x00000000 0x00000002 0x00000000
4ff2f480: 0x00000000 0x00000000 0x00000000 0x4ff082ca 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f4a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff2f4c0: 0x00000000 0x4ff2e4e4 0x4ff2e4e4 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5
4ff2f4e0: 0xa5a5a5a5 0x00000160 0x4ff2f280 0x00000c47 0x4ff16874 0x4ff16874 0x4ff2f4e8 0x4ff1686c
4ff2f500: 0x00000017 0x4ff2f6a4 0x4ff2f6a4 0x4ff2f4e8 0x00000000 0x00000002 0x4ff2e4e4 0x20627375
4ff2f520: 0x74736f68 0x73617420 0x0000006b 0x7fffffff 0x4ff2f4e0 0x00000002 0x00000000 0x00000000
4ff2f540: 0x00000000 0x00000000 0x4ff1b500 0x4ff1b568 0x4ff1b5d0 0x00000000 0x00000000 0x00000001
4ff2f560: 0x00000000 0x00000000 0x00000000 0x40007b9e 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f5a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f5c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f5e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f600: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x868ff0ef
4ff2f640: 0x44854409 0x4ff2f4e0 0x00000028 0x00000000 0x00000001 0x00000000 0x00000000 0x4ff2f658
4ff2f660: 0x4ff2f678 0x4ff2f6d0 0x4ff2f73c 0x4ff2f8f0 0x00000000 0x00000054 0x4ff2f678 0x4ff2f678
4ff2f680: 0x4ff2f678 0x4ff2f678 0x00000000 0x4ff2f690 0xffffffff 0x4ff2f690 0x4ff2f690 0x00000000
4ff2f6a0: 0x4ff2f6a4 0xffffffff 0x4ff2f6a4 0x4ff2f6a4 0x00000001 0x00000001 0x00000000 0x46b2c600
4ff2f6c0: 0x00000000 0xb33fffff 0x00000000 0x00000054 0x00000000 0x4ff2f6d0 0x00000000 0x00000000
4ff2f6e0: 0x00000000 0x4ff2f6e8 0xffffffff 0x4ff2f6e8 0x4ff2f6e8 0x00000000 0x4ff2f6fc 0xffffffff
[... cut ...] 0xb33fffff
4ff2f720: 0x00000000 0x00000010 0x00000000 0x4ff2f73c 0x00000000 0x00000001 0x0000001c 0x00000001
4ff2f740: 0x00000000 0x00000001 0x00000001 0x00000000 0x00000000 0x00000000 0x0000000c 0x4ff2f76c
4ff2f760: 0x4ff2f84c 0x4ff2f754 0x0000004c 0x4ff2f7bc 0x4ff2fa00 0x00000000 0x4ff2f774 0x00000000
4ff2f780: 0x4ff2f77c 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000
4ff2f7a0: 0x00000001 0x00000000 0x40045160 0x00000000 0x4ff2f7f4 0x00000000 0x00000034 0x50000000

roma-jam, 2025-07-04:
Hi @savenlid, 

to unblock the client, you can use the `usb_host_client_unblock()` and after deregestring the client you will get the `USB_HOST_LIB_EVENT_FLAGS_NO_CLIENTS` event.  On this event you can mark all devices as free with `usb_host_device_free_all()` and then wait for the `USB_HOST_LIB_EVENT_FLAGS_ALL_FREE` event. 

Please, refer to the [usb_host_lib example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/usb_host_lib/main/usb_host_lib_main.c#L126), to check how the USB Host Library could be closed with or without connected device.

savenlid, 2025-07-04:
I realize I don't read properly what you write.
Sorry.
I will read/implement/try.

savenlid, 2025-07-04:
I fail with this,

Will continue later.
But this can consume hours.

savenlid, 2025-07-04:
I did decipher this.

doing this beforehand as part of your tear-down is they key
inside msc-task, last thing you do.
    ESP_ERROR_CHECK( msc_host_uninstall() );
    vTaskDelay(pdMS_TO_TICKS(100));  // give some time for host task to crunch

Then you get the event the host task waits for:

I (30788) USB_MSC: MSC shutdown  //   (vfs shutdown,  MSC device remove is ever used) 
I (30888) USB_MSC: USB-HOST-MSC Uninstall
I (30898) USB_MSC: USB msc task delete  // killing the MSC task

// now the host task gets the event,
I (30898) USB_MSC: No clients remaining, freeing all devices
I (31098) USB_MSC: USB-HOST uninstall
I (31098) USB_MSC: USB-HOST task delete

In the  msc example "msc_host_uninstall" is done inside host-task
after the event-loop
did not fly, will not fly.