espressif/esp-idf #16243: ESP-IDF upgrade 5.3 to 5.4 breaks deep sleep wake from ext0 GPIO on ESP32

dvosully, 2025-07-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2-2-

### Espressif SoC revision.

ESP32-DOWD-V3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

I expected the ESP32 to wake on ext0 GPIO interrupt when RTC Clock config is set to use 'Internal 8.5MHz oscillator, divided by 256' after upgrade to IDF v5.4 latest

### What is the actual behavior?

The ESP32 does not wake on ext0 GPIO interrupt when RTC Clock config is set to use 'Internal 8.5MHz oscillator, divided by 256' after upgrade to IDF v5.4 latest. If the RTC Clock config is set to 'Internal 150 kHz RC oscillator' the ESP32 does wake on GPIO interrupt.

### Steps to reproduce.

I have not been able to reproduce this using the deep_sleep example program using the sdkconfig from my program. In both RTC configurations the example program wakes on GPIO interrupt using my custom board and GPIO_34. My application only wakes on GPIO interrupt if RTC Clock is set to RC oscillator.

### Debug Logs.

```plain

```

### More Information.

I have applied the patch suggested by @esp-wzh in https://github.com/espressif/esp-idf/issues/15403#issuecomment-2705401332 to report the sleep flags passed to esp_sleep_start, and they are different between the deep_sleep example program and my application.

The example deep sleep utility reports the following sleep flags when it goes to sleep (in 8MD256 RTC Clock configuration): `sleep_flags:0x000400e1`
The example deep sleep utility reports the following sleep flags when it goes to sleep (in RC RTC Clock configuration): `sleep_flags:0x000401e1`
My program reports the following sleep flags when it goes to sleep (in both RTC Clock configurations): `sleep_flags:0x000401e1`

The bit that is different is RTC_SLEEP_PD_INT_8M (bit 8 on the ESP32), I think indicating that the 8M oscillator is getting powered down while it is configured as the RTC clock source in my application.

esp-wzh, 2025-07-03:
Hi, thanks for the info, yes you are right, I think the INT_8M clock is powered down during sleep, which is the direct cause of the problem.

However, the IDF startup code has already processed this logic when selecting the slow clock source. If the INT_8M source is selected, this flag will not be set, see:
https://github.com/espressif/esp-idf/blob/v5.4.2/components/esp_hw_support/port/esp32/rtc_clk.c#L290
https://github.com/espressif/esp-idf/blob/v5.4.2/components/esp_hw_support/sleep_modes.c#L2510

I can't reproduce this problem in a simple example, can you describe the scenario in which you encounter the problem in more detail? Does the problem only occur after OTA or reset several times?

dvosully, 2025-07-03:
Thanks for the quick response. I have also not been able to reproduce this in a simple example.

The problem occurs both after OTA and after external hardware reset several times.

When I trace the lines you showed above:
https://github.com/espressif/esp-idf/blob/v5.4.2/components/esp_hw_support/port/esp32/rtc_clk.c#L290 got executed in startup as expected
https://github.com/espressif/esp-idf/blob/v5.4.2/components/esp_hw_support/sleep_modes.c#L2510 did NOT get executed at shutdown.
https://github.com/espressif/esp-idf/blob/v5.4.2/components/esp_hw_support/sleep_modes.c#L2519 did NOT get executed at shutdown.
When I trace sleep_flags at the start vs end of get_sleep_flags(), i get
```
get_sleep_flags()start:0x000001e1
get_sleep_flags()end:0x000401e1
```

Any advice you can give to help me track it down is welcome. I'll consider trying to find the IDF commit that introduced the bug soon, and disabling parts of my application.

dvosully, 2025-07-04:
I've applied that patch and the printf is only called to turn `ESP_SLEEP_RTC_USE_RC_FAST_MODE` on once at powerup from external reset, it is never turned off. So I started looking for memory corruption.

I placed a new RTC_FAST_ATTR variable (statically initialised and only ever read) immediately before the s_sleep_sub_mode_ref_cnt declaration and initialised it with data. On powerup it contained the expected data, but on poweroff, it had been clobbered to zeros.

Looking at the memory use of my application with idf.py size-files, the only file with RTC_FAST memory is sleep_modes.c and I never knowingly use it at runtime.

I've never yet successfully got the debugger going, but I might need to try again to find where it's coming from...

Once again open to suggestions, but it seems I'll need to pick through the memory map to get a better idea of the culprit.

dvosully, 2025-07-08:
I've narrowed the cause and can supply a minimal reproducer.
The fault occurs when esp_deep_sleep_start() is called from a task running on core 1 instead of core 0.

To reproduce:
Build the deep_sleep example for ESP32 and change [this](https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/examples/system/deep_sleep/main/deep_sleep_example_main.c#L171) line to `xTaskCreatePinnedToCore(deep_sleep_task, "deep_sleep_task", 4096, NULL, 6, NULL, 1);`
https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/examples/system/deep_sleep/main/deep_sleep_example_main.c#L171

esp-wzh, 2025-07-08:
Sorry, I still can't reproduce it, can you upload the complete code and sdkconfig? @dvosully

closed 2025-07-24 by espressif-bot

tpbedford-oomdata, 2025-08-05:
> 
> This is indeed a bug in esp-idf, introduced by [commit](https://github.com/espressif/esp-idf/commit/eab98765ad33bbb6604612915398702bd7b524ba). The direct cause of the problem is that esp32's Core1 can not access RTC_FAST memory, but some key variables that identify sleep mode are stored in it.
> 
> On v5.4.2, you can use this [patch](https://github.com/user-attachments/files/21119069/fix-IDFGH-15614-v5.4.2.patch.zip) to bypass the problem. I will soon fix this bug in master and backport it to v5.4.
> 
> I apologize for the inconvenience this bug has caused, and thank you for reporting it. [@dvosully](https://github.com/dvosully)

Is this patch a complete fix, @esp-wzh ? I've applied it, but RTC_SLOW_ATTR mem is getting reset back to 0 on WDT and soft reset. RTC_NOINIT_ATTR is preserved for me in these resets, but on first power-on the contents of RTC_NOINIT_ATTR s_sleep_sub_mode_ref_cnt then becomes garbage.

Is there a variable attribute that instructs the esp32 to preserve memory during wdt and soft reset (and sleeps), but also initializes to 0 on cold boot?

tpbedford-oomdata, 2025-08-05:
> This is expected behavior. After a reset, the software should reinitialize submode when it is restarted.

Fair, though after a WDT or soft reset s_sleep_sub_mode_ref_cnt is cleared to 0 and select_rtc_slow_clk(SLOW_CLK_8MD256) does nothing - it detects the rtc clk is already 8MD256 so does not set it, and s_sleep_sub_mode_ref_cnt remains at 0

- hard reboot: rtc clk is not set, so select_rtc_slow_clk(SLOW_CLK_8MD256) sets s_sleep_sub_mode_ref_cnt, and sleep works OK
- soft/wdt reboot: rtc clk is set, so select_rtc_slow_clk(SLOW_CLK_8MD256) does nothing, but s_sleep_sub_mode_ref_cnt is cleared to 0, and sleep mode does not work

Further, after a soft reset, calling rtc_clk_slow_src_set(SOC_RTC_SLOW_CLK_SRC_RC_SLOW) causes a crash at assert(s_sleep_sub_mode_ref_cnt[mode] >= 0); because it tries to decrement below 0

I'm seeing clarity on whether this is comprehensively fixed in a later release (I'm using 5.4.1) because the commit that introduced s_sleep_sub_mode_ref_cnt evidently has numerous issues.

esp-wzh, 2025-08-05:
> > This is expected behavior. After a reset, the software should reinitialize submode when it is restarted.
> 
> Fair, though after a WDT or soft reset s_sleep_sub_mode_ref_cnt is cleared to 0 and select_rtc_slow_clk(SLOW_CLK_8MD256) does nothing - it detects the rtc clk is already 8MD256 so does not call esp_sleep_sub_mode_config(...), and s_sleep_sub_mode_ref_cnt remains at 0
> 
>     * hard reboot: rtc clk is not set, so select_rtc_slow_clk(SLOW_CLK_8MD256) sets s_sleep_sub_mode_ref_cnt, and sleep works OK
> 
>     * soft/wdt reboot: rtc clk is set, so select_rtc_slow_clk(SLOW_CLK_8MD256) does nothing, but s_sleep_sub_mode_ref_cnt is cleared to 0, and sleep mode does not work
> 
> 
> Further, after a soft reset, calling rtc_clk_slow_src_set(SOC_RTC_SLOW_CLK_SRC_RC_SLOW) causes a crash inside at assert(s_sleep_sub_mode_ref_cnt[mode] >= 0); because it tries to decrement below 0
> 
> I'm seeing clarity on whether this is comprehensively fixed in a later release (I'm using 5.4.1) because the commit that introduced s_sleep_sub_mode_ref_cnt evidently has numerous issues.

Oh, this is indeed a bug, sorry for that, I will fix it in the next release.  You can temporarily use the following patch to bypass this problem:
```diff
diff --git a/components/esp_hw_support/port/esp32/rtc_clk.c b/components/esp_hw_support/port/esp32/rtc_clk.c
index 63ac7ce3a14..95ef744a5f0 100644
--- a/components/esp_hw_support/port/esp32/rtc_clk.c
+++ b/components/esp_hw_support/port/esp32/rtc_clk.c
@@ -286,7 +286,7 @@ void rtc_clk_slow_src_set(soc_rtc_slow_clk_src_t clk_src)
 #ifndef BOOTLOADER_BUILD
     soc_rtc_slow_clk_src_t clk_src_before_switch = clk_ll_rtc_slow_get_src();
     // Keep the RTC8M_CLK on in sleep if RTC clock is rc_fast_d256.
-    if (clk_src == SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256 && clk_src_before_switch != SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256) {       // Switch to RC_FAST_D256
+    if ((clk_src == SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256) && (esp_sleep_sub_mode_dump_config(NULL)[ESP_SLEEP_RTC_USE_RC_FAST_MODE] == 0)) { // Switch to RC_FAST_D256
         esp_sleep_sub_mode_config(ESP_SLEEP_RTC_USE_RC_FAST_MODE, true);
     } else if (clk_src != SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256 && clk_src_before_switch == SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256) { // Switch away from RC_FAST_D256
         esp_sleep_sub_mode_config(ESP_SLEEP_RTC_USE_RC_FAST_MODE, false);
```

tpbedford-oomdata, 2025-08-05:
@esp-wzh That latest patch tests well for me, thanks!