espressif/esp-idf #16251: esp_read_mac doesn't work with ESP32-P4

cniedzi, 2025-07-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 framework-espidf @ 3.50402.0 (5.4.2)

### Espressif SoC revision.

ESP32-P4 (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare ESP32-P4-Module-DEV-KIT

### Power Supply used.

USB

### What is the expected behavior?

The following code:
```
#include "esp_mac.h"
esp_err_t ret = esp_read_mac(baseMac, ESP_MAC_WIFI_STA);
```
should return ESP_OK and valid Mac address for WiFi.

### What is the actual behavior?

The following code:
```
#include "esp_mac.h"
esp_err_t ret = esp_read_mac(baseMac, ESP_MAC_WIFI_STA);
```
returns value of 262
```
E (8907) system_api: 0 mac type is incorrect (not found)
```

### Steps to reproduce.

Run code in Arduino IDE or VS Code (Platformio)

```
#include "esp_mac.h"

esp_err_t ret = esp_read_mac(baseMac, ESP_MAC_WIFI_STA);
```

### Debug Logs.

```plain

```

nopnop2002, 2025-07-03:
The esp32-p4 does not have WiFi functionality.

cniedzi, 2025-07-03:
> The esp32-p4 does not have WiFi functionality.

it utilizes ESP32-C6 for WiFi operations.

bryghtlabs-richard, 2025-07-03:
Would the MAC address be tied to the NIC then?

chegewara, 2025-07-03:
> > The esp32-p4 does not have WiFi functionality.
> 
> it utilizes ESP32-C6 for WiFi operations.

Its one option, but this can be any esp32 series and you have different API to read it. 
On P4 you can only read ETH mac.