espressif/esp-idf #16609: spi_lcd_touch_example does no build due to ASM errors.

sailingsnoopy, 2025-07-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The example project builds successfully.

### What is the actual behavior?

The build fails due to not found opcodes.

### Steps to reproduce.

1. Create a new project from the example project https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/spi_lcd_touch
2. Build the project.

### Build or installation Logs.

```plain
[591/1729] Performing build step for 'bootloader'
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/Users/sailingsnoopy/Documents/Projekte/Sonstiges/ESP32 Sandbox/Motor_Driver/lcd_example/spi_lcd_touch/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_security esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/bootloader C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/bootloader_support C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/efuse C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_app_format C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_bootloader_format C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_common C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_security C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_system C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esptool_py C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/hal C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/log C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/bootloader/subproject/main C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/newlib C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/partition_table C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/soc C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/spi_flash C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/xtensa
-- Configuring done (40.7s)
-- Generating done (1.7s)
-- Build files have been written to: C:/Users/sailingsnoopy/Documents/Projekte/Sonstiges/ESP32 Sandbox/Motor_Driver/lcd_example/spi_lcd_touch/build/bootloader
[1/2] C:\WINDOWS\system32\cmd.exe /C "cd /D "C:\Users\sailingsnoopy\Documents\Projekte\Sonstiges\ESP32 Sandbox\Motor_Driver\lcd_example\spi_lcd_touch\build\bootloader\esp-idf\esptool_py" && C:\Users\sailingsnoopy\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 "C:/Users/sailingsnoopy/Documents/Projekte/Sonstiges/ESP32 Sandbox/Motor_Driver/lcd_example/spi_lcd_touch/build/bootloader/bootloader.bin""
Bootloader binary size 0x51c0 bytes. 0x2e40 bytes (36%) free.
[834/1729] Building ASM object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xtensa_vectors.S.obj
FAILED: esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xtensa_vectors.S.obj 
ccache C:\Users\sailingsnoopy\.espressif\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.4.2\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I"C:/Users/sailingsnoopy/Documents/Projekte/Sonstiges/ESP32 Sandbox/Motor_Driver/lcd_example/spi_lcd_touch/build/config" -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/xtensa/esp32s3/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/xtensa/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/xtensa/deprecated_include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/newlib/platform_include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/config/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/config/include/freertos -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/config/xtensa/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/freertos/esp_additions/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/include/soc -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/dma/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/ldo/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/debug_probe/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_hw_support/port/esp32s3/include 
-IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/heap/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/heap/tlsf -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/log/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/soc/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/soc/esp32s3 -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/soc/esp32s3/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/soc/esp32s3/register -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/hal/platform_port/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/hal/esp32s3/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/hal/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_rom/esp32s3 -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_common/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_system/include -IC:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/esp_system/port/soc -IC:/Users/sailingsnoop [... cut ...]
C:/Users/sailingsnoopy/esp/v5.4.2/esp-idf/components/hal/esp32s3/include/hal/dedic_gpio_cpu_ll.h:28: Error: unknown opcode or format name 'ee.get_gpio_in'
[847/1729] Building C object esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/arg_dstr.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Users\sailingsnoopy\.espressif\tools\ninja\1.12.1\ninja.EXE" terminated with exit code: 1.
```

### More Information.

My assumption is, there is something wrong with the lvgl config, but I am not familiar enough with the IDE to tell for sure.

Alvin1Zhang, 2025-07-06:
Thanks for reporting.

sailingsnoopy, 2025-07-07:
I was thinking the same, but dismissed the idea after trying with a clean installation. Language is set to English, so I don't expect any issues there.

sailingsnoopy, 2025-07-07:
After another time reinstalling the IDE it works. There seems to be soome instability, but it is solved for now.

closed 2025-07-07 by sailingsnoopy