espressif/esp-idf #16729: [Naming Conflict] ESP-IDF cdc_acm_init in cdc_acm.h conflicts with function in CherryUSB cdc_acm_template.c, causing reboot loop

MDLZCOOL, 2025-07-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

When using CherryUSB cdc_acm_template, esp32s3 should work properly.

### What is the actual behavior?

Endless reboot loop.

### Steps to reproduce.

1. add latest CherryUSB  to project
2. set necessary settings in CherryUSB menuconfig
3. in app_main, call `cdc_acm_init(0, ESP_USBD_BASE);` and in endless loop, call `cdc_acm_data_send_with_dtr_test(0);`
4. compile, flash and monitor

### Build or installation Logs.

```plain
run `idf.py monitor -p COM24`, and I got this:

 bash
I (26) boot: ESP-IDF v5.4.2 2nd stage bootloader
I (26) boot: compile time Jul  6 2025 22:01:14
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (29) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (80) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a660h ( 42592) map
I (95) esp_image: segment 1: paddr=0001a688 vaddr=3fc92400 size=03324h ( 13092) load
I (98) esp_image: segment 2: paddr=0001d9b4 vaddr=40374000 size=02664h (  9828) load
I (105) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17b8ch ( 97164) map
I (127) esp_image: segment 4: paddr=00037bb4 vaddr=40376664 size=0bcdch ( 48348) load
I (138) esp_image: segment 5: paddr=00043898 vaddr=600fe000 size=0001ch (    28) load
I (144) boot: Loaded app from partition at offset 0x10000
I (144) boot: Disabling RNG early entropy source...
I (156) cpu_start: Multicore app
I (165) cpu_start: Pro cpu start user code
I (165) cpu_start: cpu freq: 160000000 Hz
I (165) app_init: Application information:
I (165) app_init: Project name:     hello_world
I (169) app_init: App version:      1
I (173) app_init: Compile time:     Jul  6 2025 22:19:34
I (178) app_init: ELF file SHA256:  0e8cc4c01...
I (182) app_init: ESP-IDF:          v5.4.2
I (186) efuse_init: Min chip rev:     v0.0
I (190) efuse_init: Max chip rev:     v0.99
I (194) efuse_init: Chip rev:         v0.2
I (198) heap_init: Initializing. RAM available for dynamic allocation:
I (204) heap_init: At 3FC95FF8 len 00053718 (333 KiB): RAM
I (209) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (214) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (219) heap_init: At 600FE01C len 00001FCC (7 KiB): RTCRAM
I (226) spi_flash: detected chip: generic
I (228) spi_flash: flash io: dio
W (231) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (244) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (250) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (257) main_task: Started on CPU0
I (267) main_task: Calling app_main()
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400570e8  PS      : 0x00060130  A0      : 0x8005117d  A1      : 0x3fc98890
--- 0x400570e8: memset in ROM

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x60080000  A5      : 0x00000000
A6      : 0x3fc95e9c  A7      : 0x06008000  A8      : 0x8200cf78  A9      : 0x3fc98510
A10     : 0x00000026  A11     : 0x00000000  A12     : 0xfffffff7  A13     : 0x3c024243
A14     : 0x00000015  A15     : 0x3fc98720  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x06007fff
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM

Backtrace: 0x400570e5:0x3fc98890 0x4005117a:0x3fc988a0 0x42009b04:0x3fc988c0 0x42017513:0x3fc988e0 0x4037a4c5:0x3fc98910
--- 0x400570e5: memset in ROM
--- 0x4005117a: cdc_acm_init in ROM
--- 0x42009b04: app_main at D:/program-projects/esp-idf-proj/hello_world/main/hello_world_main.c:21
--- 0x42017513: main_task at C:/Espressif/frameworks/esp-idf-v5.4.2/components/freertos/app_startup.c:208
--- 0x4037a4c5: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.4.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### Diagnostic report archive.

[diag-f885393a-8b0d-4ae0-96f8-1c0f984a0b82.zip](https://github.com/user-attachments/files/21089584/diag-f885393a-8b0d-4ae0-96f8-1c0f984a0b82.zip)

igi540, 2025-07-07:
The reboot loop is due to a `StoreProhibited` exception caused by a NULL pointer dereference in `cdc_acm_init()`. This usually happens when the USB peripheral or stack isn't initialized properly before calling CherryUSB functions.

Make sure CherryUSB settings in `menuconfig` are correctly set for ESP32-S3 with internal PHY.

Add error handling for `cdc_acm_init()` to catch issues early.

igrr, 2025-07-07:
I think the point is that CherryUSB's cdc_acm_init function is not linked at all, instead the linker picks up this ROM function with the same name:

https://github.com/espressif/esp-idf/blob/5b11d5b26a8bf151fc6bac400158859eedd413bc/components/esp_rom/esp32s3/ld/esp32s3.rom.ld#L678

A solution could be to rename the `cdc_acm_*` functions in the ROM linker script e.g. to `rom_cdc_acm_*` and update the way they are called from IDF accordingly. Then linking CherryUSB functions will be possible without a naming conflict.

peter-marcisovsky, 2025-07-07:
By following the cherryusb [example](https://github.com/CherryUSB/cherryusb_esp32/blob/main/examples/device/main/main.c) for esp32, I managed to reproduce the issue. The only solution I could find was to rename the function.

Renaming it in the cherryusb, would probably mean to break the public API.
So it would make more sense to rename it in esp-idf. 

We will discuss and let you know. We wanted to support the cherryusb sometimes in future anyway.