espressif/esp-idf #16748: ESP32S3 LCD 2 Frame Buffers with Bounce Buffer

mattmatt206, 2025-07-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When using 2 frame buffers (or 3) with bounce buffer enabled, only the first frame buffer is used. This completely removes the point of 2 frame buffers as it always draws to the current frame buffer, so visible redrawing is seen

suda-morris, 2025-07-08:
https://github.com/espressif/esp-idf/blob/5b11d5b26a8bf151fc6bac400158859eedd413bc/components/esp_lcd/rgb/esp_lcd_panel_rgb.c#L664

This will update the cur_fb_index.

---

https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/rgb/esp_lcd_panel_rgb.c#L840

This will update the bb_fb_index to the cur_fb_index.

---

So "only the first frame buffer is used" is not true maybe?

mattmatt206, 2025-07-09:
That's true if using the same buffers as the draw and lcd frame buffer. However, if using software rotate in the [esp_lvgl_port](https://github.com/espressif/esp-bsp/tree/76cc90336b34955fc76b510557b837e963b6a9e9/components/esp_lvgl_port), its required to copy the partial draw buffer into one of the frame buffers as you are unable to use full/direct with software rotate. And when it has to copy it only draws to the current index which is the same as the lcd frame buffer. see
https://github.com/espressif/esp-idf/blob/5b11d5b26a8bf151fc6bac400158859eedd413bc/components/esp_lcd/rgb/esp_lcd_panel_rgb.c#L638

maybe there should be some method to change the index?