espressif/esp-idf #16767: ESP32S3adc参考设置问题，导致adc读取转换成电压数值不对

beautifulpeer, 2025-07-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. 在新版本的ESP-IDFV5.4.2中，并没有ADC Attenuation的表格，在https://www.espressif.com.cn/sites/default/files/documentation/esp32-s3_datasheet_cn.pdf 的5.5ADC特性中有，其ADC_ATTEN_DB_2_5测量范围是0mV-1100mV

2.在esp-idf 4.4.6版本中 ADC_ATTEN_DB_2_5 测量范围是0mV-1250mV

3.我使用手里的硬件，通过参考电压计算实际电压值时，我使用1100mV计算出的数值不对，相差很大，而使用1250mV计算出的数值相对准确

4.可我使用的ESP-IDF版本是5.4.2呀，这是为什么，我应该如何选择，是使用1250还是1100，这个有什么区别？
@esp-cjh

beautifulpeer, 2025-07-11:
使用oneshot_read的历程测试（含校准），计算得出的结果也和实际的不符，电压2700mV，测量得出结果为2597mV

larryli, 2025-07-11:
https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/peripherals/adc_calibration.html

> 在 ESP32-S3 中，模数转换器 (ADC) 比较输入的模拟电压和参考电压，以确定每一位数字输出结果。ESP32-S3 设计的 ADC 参考电压为 1100 mV。然而，不同芯片的真实参考电压可能会略有变化，范围在 1000 mV 到 1200 mV 之间。

esp-cjh, 2025-07-11:
@weiyuhannnn 这个是 ADC 的问题，请分配给对应的同事进行支持。

beautifulpeer, 2025-07-14:
@weiyuhannnn

beautifulpeer, 2025-07-15:
1.oneshot_read历程中就是使用的校准后的，使用的衰减为ADC_ATTEN_DB_12，其对应的测量范围是 0 ~ 2900mV 总误差：–50至50 mV，而实际测试结果就是上面的“使用oneshot_read的历程测试（含校准），计算得出的结果也和实际的不符，电压2700mV，测量得出结果为2597mV”

[image]

引用“oneshot_read例成测试不准问题，请提供更多信息” 
这个直接使用的测试历程，只是修改了ADC通道，代码如下
@Bruce297 

/*
 * SPDX-FileCopyrightText: 2022-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/soc_caps.h"
#include "esp_log.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"

const static char *TAG = "EXAMPLE";

/*---------------------------------------------------------------
        ADC General Macros
---------------------------------------------------------------*/
//ADC1 Channels
#if CONFIG_IDF_TARGET_ESP32
#define EXAMPLE_ADC1_CHAN0          ADC_CHANNEL_4
#define EXAMPLE_ADC1_CHAN1          ADC_CHANNEL_5
#else
#define EXAMPLE_ADC1_CHAN0          ADC_CHANNEL_4
#define EXAMPLE_ADC1_CHAN1          ADC_CHANNEL_3
#endif

#if (SOC_ADC_PERIPH_NUM >= 2) && !CONFIG_IDF_TARGET_ESP32C3
/**
 * On ESP32C3, ADC2 is no longer supported, due to its HW limitation.
 * Search for errata on espressif website for more details.
 */
#define EXAMPLE_USE_ADC2            1
#endif

#if EXAMPLE_USE_ADC2
//ADC2 Channels
#if CONFIG_IDF_TARGET_ESP32
#define EXAMPLE_ADC2_CHAN0          ADC_CHANNEL_0
#else
#define EXAMPLE_ADC2_CHAN0          ADC_CHANNEL_0
#endif
#endif  //#if EXAMPLE_USE_ADC2

#define EXAMPLE_ADC_ATTEN           ADC_ATTEN_DB_12

static int adc_raw[2][10];
static int voltage[2][10];
static bool example_adc_calibration_init(adc_unit_t unit, adc_channel_t channel, adc_atten_t atten, adc_cali_handle_t *out_handle);
static void example_adc_calibration_deinit(adc_cali_handle_t handle);

void app_main(void)
{
    //-------------ADC1 Init---------------//
    adc_oneshot_unit_handle_t adc1_handle;
    adc_oneshot_unit_init_cfg_t init_config1 = {
        .unit_id = ADC_UNIT_1,
    };
    ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));
[... 115 lines cut ...]
#endif

    *out_handle = handle;
    if (ret == ESP_OK) {
        ESP_LOGI(TAG, "Calibration Success");
    } else if (ret == ESP_ERR_NOT_SUPPORTED || !calibrated) {
        ESP_LOGW(TAG, "eFuse not burnt, skip software calibration");
    } else {
        ESP_LOGE(TAG, "Invalid arg or no memory");
    }

    return calibrated;
}

static void example_adc_calibration_deinit(adc_cali_handle_t handle)
{
#if ADC_CALI_SCHEME_CURVE_FITTING_SUPPORTED
    ESP_LOGI(TAG, "deregister %s calibration scheme", "Curve Fitting");
    ESP_ERROR_CHECK(adc_cali_delete_scheme_curve_fitting(handle));

#elif ADC_CALI_SCHEME_LINE_FITTING_SUPPORTED
    ESP_LOGI(TAG, "deregister %s calibration scheme", "Line Fitting");
    ESP_ERROR_CHECK(adc_cali_delete_scheme_line_fitting(handle));
#endif
}

beautifulpeer, 2025-07-22:
@espressif-bot