espressif/esp-idf #16769: ESP32C2  Nimble gatt server, indicate data failed because Out of memory.

ArvinHou, 2025-07-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF Version: 5.4.2
Target:  esp32c2

I am using the example of nimble gatt server.I made some modifications.
I can connect to the device normally, subscribe, and send and receive data.
But after I sent the data 4 times through the device, the code shows out of  memory.
ESP32c2 can receive data, but cannot send data.
When sending data, a notify event was received. The "status=6" means out of memory.
But the log shows "ram available:  16728"
When I disconnect  and reconnect bluetooth.I can send the data 4 more times.

```
NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=16 status=6 is_indication=1
```

detail log is bellow:
```

I (76305) NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=16 status=7 is_indication=1
I (76306) NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=16 status=7 is_indication=1
I (76312) NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=16 status=7 is_indication=1
I (76320) NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=16 status=7 is_indication=1
I (76329) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=8 reason=2 prevn=0 curn=0 previ=1 curi=0
I (76339) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=8
I (76345) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=16 reason=2 prevn=0 curn=0 previ=1 curi=0
I (76355) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=16
I (76362) NimBLE_GATT_Server: disconnected from peer; reason=531
I (76369) NimBLE: GAP procedure initiated: advertise; 
I (76373) NimBLE: disc_mode=2
I (76375) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=816
I (76385) NimBLE: 

I (76387) NimBLE_GATT_Server: advertising started!

I (81624) NimBLE_GATT_Server: connection established; status=0
I (81624) NimBLE_GATT_Server: connection handle: 1
I (81624) NimBLE_GATT_Server: device id address: type=0, value=BA:F8:9B:76:EF:D0
I (81630) NimBLE_GATT_Server: peer id address: type=1, value=63:3B:16:7D:9A:77
I (81637) NimBLE_GATT_Server: conn_itvl=24, conn_latency=0, supervision_timeout=72, encrypted=0, authenticated=0, bonded=0

I (81648) NimBLE: GAP procedure initiated: 
I (81652) NimBLE: connection parameter update; conn_handle=1 itvl_min=24 itvl_max=24 latency=3 supervision_timeout=72 min_ce_len=0 max_ce_len=0
I (81664) NimBLE: 

I (81744) NimBLE_GATT_Server: mtu update event; conn_handle=1 cid=4 mtu=512

I (81924) NimBLE_GATT_Server: connection updated; status=0
I (81924) NimBLE_GATT_Server: connection handle: 1
I (81924) NimBLE_GATT_Server: device id address: type=0, value=BA:F8:9B:76:EF:D0
I (81930) NimBLE_GATT_Server: peer id address: type=1, value=63:3B:16:7D:9A:77
I (81937) NimBLE_GATT_Server: conn_itvl=24, conn_latency=3, supervision_timeout=72, encrypted=0, authenticated=0, bonded=0

I (81993) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=8 reason=1 prevn=0 curn=0 previ=0 curi=1
I (81993) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=8

I (84093) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=16 reason=1 prevn=0 curn=0 previ=0 curi=1
I (84093) NimBLE_GATT_Server: subscribe event; conn_handle=1 attr_handle=16

I (100444) NimBLE_GATT_Server: characteristic write; conn_handle=1 attr_handle=19
0 100,096: ble recv:
 123456
0 100,096: ble send:
[... 144 lines cut ...]
            
            ble_gatts_indicate(indicate_chr_conn_handle,
                           indicate_chr_val_handle);
            

            printf("ram available:  %ld\r\n",esp_get_free_heap_size());
            return rc == 0 ? 0 : BLE_ATT_ERR_INSUFFICIENT_RES;
        }
        break;
        
    /* Unknown event */
    default:
        break;
    }

    return BLE_ATT_ERR_UNLIKELY;
}
```

In  `BLE_GATT_ACCESS_OP_WRITE_CHR`  event, receive data and process it. using `ble_gatts_indicate` notify device to read data. Cache data into an array `tmp_tx`.
In  `BLE_GATT_ACCESS_OP_READ_CHR` event,using `os_mbuf_append` send data. send data cached in  `tmp_tx`.The `tmp_tx` is a  global array.

I also tried `os_mbuf_get` ,but not work.

Who can give me some advice，Thanks very much!

Singh-Sumeet, 2025-07-11:
Hi @ArvinHou ,
I will suggest two things:
1. In NimBLE the maximum GATT procedures that can be queued happen simultaneously is set to 4, can you try increasing it to any value other than 4 to see if the behavior changes. This is not a solution, it will however help to narrow down the issue. You can update the config named `CONFIG_BT_NIMBLE_GATT_MAX_PROCS` using idf.py menuconfig > Component config > Bluetooth > NimBLE Options > Maximum number of GATT client procedures.
2. Note that the Bluetooth spec mandates that no more than 1 GATT procedures may run at once on a channel, so it is always recommended to start the next procedure after the previous one finishes (instead of looping and sending a bunch of requests).
Since you have not provided the client side code/logs, and I can see that the client is performing a bunch of writes, I will suggest that you should ensure that subsequent writes happen after the previous write is finished. 
In NimBLE this can be accoplished by performing the next procedure in the previous one's callback. 
In the client code, you may have used `ble_gattc_write(conn_handle, attr_handle, om, cb, cb_arg)`. Note the 4th parameter, namely `cb`. This takes a function pointer and invokes that function after the client has received the write response of the respective request.
So you may start the next write procedure, in the client side, within the callback (as provided in `cb`) of the previous procedure.

Meanwhile, I will reproduce the issue on my end, and it will be helpful if you can provide the client side code/logs.

ArvinHou, 2025-07-13:
@Singh-Sumeet  Thanks for your reply！I have tested it.When I change  CONFIG_BT_NIMBLE_GATT_MAX_PROCS to 6 ,I can receive data 6 times.

I added my code based on the example, and everything was normal for sending and receiving.
In my program, I adjusted some settings to reduce memory usage.
Is there something wrong with the sdkconfig  that caused it？
The sdkconfig file is bellow:

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.4.2 Project Configuration
#
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_AHB_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_CONSISTS_OF_ONE_KEY_BLOCK=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_ECC_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_ASSIST_DEBUG_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_RNG_SUPPORTED=y
CONFIG_SOC_LIGHT_SLEEP_SUPPORTED=y
CONFIG_SOC_DEEP_SLEEP_SUPPORTED=y
CONFIG_SOC_LP_PERIPH_SHARE_INTERRUPT=y
CONFIG_SOC_PM_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_IIR_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=1
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=5
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=1
CONFIG_SOC_ADC_PATT_LEN_MAX=8
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_IIR_FILTER_NUM=2
CONFIG_SOC_ADC_DIGI_MONITOR_NUM=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=83333
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=611
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=12
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_ADC_CALIBRATION_V1_SUPPORTED=y
CONFIG_SOC_ADC_SELF_HW_CALI_SUPPORTED=y
CONFIG_SOC_ADC_SHARED_POWER=y
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED=y
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_CPU_CORES_NUM=1
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FLEXIBLE_INTC=y
[... 2039 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```
 the code example is here:

[NimBLE_GATT_Server.zip](https://github.com/user-attachments/files/21205546/NimBLE_GATT_Server.zip)

Singh-Sumeet, 2025-08-11:
Hi @ArvinHou ,
Thank you for pointing this issue out. The sdkconfig was helpful.
I am working on a permanent fix for this issue, but as a temporary fix, please add the following to your sdkconfig or enable these from menuconfig:

1. CONFIG_BT_NIMBLE_ROLE_CENTRAL=y
2. CONFIG_BT_NIMBLE_GATT_CLIENT=y

This should fix your issue, however it will increase the flash size.
I will let you know once the permanent fix is up.