espressif/esp-idf #16771: SDMMC and WiFi on SDIO combinly not working on ESP32-P4

nbenhur, 2025-07-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP32-P4 Function EV Board. I am using ESP-IDF Version 5.4.1. on ubuntu OS. I have tested SDMMC card and WiFi examples separately on the P4 board and are working fine. In my project i wanted to use SDMMC and Wifi both. When i use both of them i am getting error. sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x108
H_SDIO_DRV: failed to read registers.

[image]

rrtandler, 2025-07-14:
Hi  @nbenhur,

Could you provide more broader context - post code snippets showing ho the two examples were joined into the single project ?

JoaoFelipeBinho, 2025-08-02:
Hi @nbenhur

I'm facing the same issue. When I mound a SD card after initialize [ESP-P4/C6 dev kit](https://www.waveshare.com/esp32-p4-nano.htm) wifi connection I got the following errors

```
I (13513) SD_Mount: Initializing SD card                                                                                                                                                                           
I (13513) SD_Mount: Using SDMMC peripheral                                                                                                                                                                         
I (13523) SD_Mount: Using SDMMC peripheral                                                                                                                                                                         
W (13523) ldo: The voltage value 0 is out of the recommended range [500, 2700]                                                                                                                                     
I (13533) SD_Mount: Mounting filesystem                                                                                                                                                                            
I (13533) sdmmc_periph: sdmmc_host_init: SDMMC host already initialized, skipping init flow                                                                                                                        
I (13543) gpio: GPIO[43]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0                                                                                                                    
I (13543) gpio: GPIO[44]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0                                                                                                                    
I (13553) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0                                                                                                                    
I (13573) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0                                                                                                                    
E (13583) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x107                                                                                                                                            
E (13583) H_SDIO_DRV: failed to read registers                                                                                                                                                                     
E (13583) sdmmc_io: sdmmc_io_rw_extended: sdmmc_send_cmd returned 0x107                                                                                                                                            
E (13593) H_SDIO_DRV: failed to read registers                                                                                                                                                                     
I (13583) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0                                                                                                                    
I (13603) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0                                                                                                                    
I (13673) gpio: GPIO[42]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0                                                                                                                    
I (13683) SD_Mount: SD Filesystem mounted                                                                                                                                                                          
Name: SD64G                                                                                                                                                                                                        
Type: SDHC                                                                                                                                                                                                         
Speed: 40.00 MHz (limit: 40.00 MHz)                                                                                                                                                                                
Size: 59640MB                                                                                                                                                                                                      
CSD: ver=2, sector_size=512, capacity=122142720 read_bl_len=9                                                                                                                                                      
SSR: bus_width=4                                                                                                                                                                                                   
I (13703) FAT_SaveData: Data saved to file: /sdcard/2769643974.txt                                                                                                                                                 
[... cut ...]
E (13733) H_SDIO_DRV: sdio_write_task: 1: Failed to send data: 263 54 54                                 
E (13743) H_SDIO_DRV: Unrecoverable host sdio state, reset host mcu                                      
I (13753) os_wrapper_esp: Restarting host           
ESP-ROM:esp32p4-eco2-20240710                       
Build:Jul 10 2024                                   
rst:0xc (SW_CPU_RESET),boot:0x30f (SPI_FAST_FLASH_BOOT) 
```

When I take off SD card mount steps this errors don't happen and I can send packages over WiFi without problmes

JoaoFelipeBinho, 2025-08-02:
I've found the issue for my case. Maybe it can be useful for you @nbenhur.

I’ve noticed that SDMMC and SDIO were using the same physical SD hardware (both on slot 1). I tried to change the C6 communication (via menuconfig) to use slot 0 but encountered errors, so I ended up setting the SDMMC slot to 0.

On my SD host configuration it got like this:
```
sdmmc_host_t host = SDMMC_HOST_DEFAULT();
host.slot = SDMMC_HOST_SLOT_0;
```

azzanrauf, 2025-08-11:
Hi @JoaoFelipeBinho 
I am facing the similar kind of problem here. Can you please give me some details of your project so i can replicate and find the bug in my project